nextcloud/nextcloudpi #1781: Updating to NC 26 - "This version of Nextcloud requires at least PHP 8.0"

chronohart, 2023-05-27:
Attempting to update from NC 25 to NC 26 through the cli, the process aborts and reverts back to NC 25 with the message "This version of Nextcloud requires at least PHP 8.0<br/>You are currently running 7.4.33. Please update your PHP version.Abort".

**Output from nc-update-nextcloud**

Running nc-update-nextcloud
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Current   Nextcloud version 25.0.3.2
Available Nextcloud version 26.0.1
Download Nextcloud 26.0.1...
Back up current instance...
Cannot load Zend OPcache - it was already loaded
check free space...
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20230527_1685197459.tar.gz generated
Cannot load Zend OPcache - it was already loaded
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20230527_1685197459-25.0.3.2.tar.gz'...
Install Nextcloud 26.0.1...
Fix permissions...
Upgrade...
Cannot load Zend OPcache - it was already loaded
This version of Nextcloud requires at least PHP 8.0<br/>You are currently running 7.4.33. Please update your PHP version.Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20230527_1685197459-25.0.3.2.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20230527_1685197459-25.0.3.2.tar.gz...
restore files...
restore database...
Cannot load Zend OPcache - it was already loaded
No datadir found in backup
Cannot load Zend OPcache - it was already loaded
Maintenance mode disabled
Cannot load Zend OPcache - it was already loaded
Starting scan for user 1 out of 4 (chronohart)
Starting scan for user 2 out of 4 (guest)
Starting scan for user 3 out of 4 (jgturbin)
Starting scan for user 4 out of 4 (ncp)
+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 152     | 2311  | 00:00:03     |
+---------+-------+--------------+
Installing template 'php/opcache.ini.sh'...
Cannot load Zend OPcache - it was already loaded
System config value tempdirectory set to string /media/myCloudDrive/ncdata/tmp
Cannot load Zend OPcache - it was already loaded
System config value logfile set to string /media/myCloudDrive/ncdata/nextcloud.log
Cannot load Zend OPcache - it was already loaded
System config value trusted_domains => 11 set to string 74.97.29.229
Cannot load Zend OPcache - it was already loaded
System config value trusted_domains => 1 set to string 192.168.1.10
Cannot load Zend OPcache - it was already loaded
System config value trusted_domains => 14 set to string nextcloudpi
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
[... 183 lines cut ...]
```

{"reqId":"p5xZMTbpuGYW6DBXA72Z","level":3,"time":"2023-05-27T15:05:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":"Failed sending activity email to user \"chronohart\"","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"Swift_TransportException","Message":"Failed to authenticate on SMTP server with username \"chronohart@gmail.com\" using 1 possible authenticators. Authenticator LOGIN returned Expected response code 235 but got code \"535\", with message \"535-5.7.8 Username and Password not accepted. Learn more at\r\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials k16-20020a0cd690000000b006166a48357asm2042654qvi.60 - gsmtp\r\n\".","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/EsmtpTransport.php","line":371,"function":"afterEhlo","class":"Swift_Transport_Esmtp_AuthHandler","type":"->"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/AbstractSmtpTransport.php","line":148,"function":"doHeloCommand","class":"Swift_Transport_EsmtpTransport","type":"->"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Mailer.php","line":65,"function":"start","class":"Swift_Transport_AbstractSmtpTransport","type":"->"},{"file":"/var/www/nextcloud/lib/private/Mail/Mailer.php","line":191,"function":"send","class":"Swift_Mailer","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/MailQueueHandler.php","line":415,"function":"send","class":"OC\\Mail\\Mailer","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/MailQueueHandler.php","line":165,"function":"sendEmailToUser","class":"OCA\\Activity\\MailQueueHandler","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/BackgroundJob/EmailNotification.php","line":61,"function":"sendEmails","class":"OCA\\Activity\\MailQueueHandler","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/Job.php","line":54,"function":"run","class":"OCA\\Activity\\BackgroundJob\\EmailNotification","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":60,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":152,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/Esmtp/AuthHandler.php","Line":191,"message":"Failed sending activity email to user \"{user}\"","CustomMessage":"Failed sending activity email to user \"chronohart\""}}
{"reqId":"p5xZMTbpuGYW6DBXA72Z","level":2,"time":"2023-05-27T15:05:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":"Failed sending activity email to user 'chronohart'.","userAgent":"--","version":"25.0.3.2","data":{"app":"activity"}}
{"reqId":"p5xZMTbpuGYW6DBXA72Z","level":3,"time":"2023-05-27T15:05:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":"Failed sending activity email to user \"jgturbin\"","userAgent":"--","version":"25.0.3.2","exception":{"Exception":"Swift_TransportException","Message":"Failed to authenticate on SMTP server with username \"chronohart@gmail.com\" using 1 possible authenticators. Authenticator LOGIN returned Expected response code 235 but got code \"535\", with message \"535-5.7.8 Username and Password not accepted. Learn more at\r\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials w16-20020a05622a191000b003ef13aa5b0bsm2177018qtc.82 - gsmtp\r\n\".","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/EsmtpTransport.php","line":371,"function":"afterEhlo","class":"Swift_Transport_Esmtp_AuthHandler","type":"->"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/AbstractSmtpTransport.php","line":148,"function":"doHeloCommand","class":"Swift_Transport_EsmtpTransport","type":"->"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Mailer.php","line":65,"function":"start","class":"Swift_Transport_AbstractSmtpTransport","type":"->"},{"file":"/var/www/nextcloud/lib/private/Mail/Mailer.php","line":191,"function":"send","class":"Swift_Mailer","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/MailQueueHandler.php","line":415,"function":"send","class":"OC\\Mail\\Mailer","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/MailQueueHandler.php","line":165,"function":"sendEmailToUser","class":"OCA\\Activity\\MailQueueHandler","type":"->"},{"file":"/var/www/nextcloud/apps/activity/lib/BackgroundJob/EmailNotification.php","line":61,"function":"sendEmails","class":"OCA\\Activity\\MailQueueHandler","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/Job.php","line":54,"function":"run","class":"OCA\\Activity\\BackgroundJob\\EmailNotification","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":60,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":152,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/ [... cut ...] 620a101900b007463509f94asm1968522qkj.55 - gsmtp\r\n\".","exception":{},"CustomMessage":"Failed to authenticate on SMTP server with username \"chronohart@gmail.com\" using 1 possible authenticators. Authenticator LOGIN returned Expected response code 235 but got code \"535\", with message \"535-5.7.8 Username and Password not accepted. Learn more at\r\n535 5.7.8  https://support.google.com/mail/?p=BadCredentials z25-20020a05620a101900b007463509f94asm1968522qkj.55 - gsmtp\r\n\"."}}

```
</details>

mayermart, 2023-05-31:
I may have had a similar issue. Check your ncp.cfg file and try set the line for php version from 8.1 to 7.4.
After that i think i ran ncp-update which then updated php to 8.1.

chronohart, 2023-05-31:
Thanks for the reponse @mayermart . Can you point me to where I can find the ncp.cfg file? I'm still very much learning my way around nextcloudpi and I'm having trouble figuring that out.

chronohart, 2023-06-01:
Found the solution! Long story short, I had to update specifically to NC 25.0.6 through the terminal before trying to update to NC latest (26.0.1). Apparently going to 25.0.6 is the way to get PHP correctly updated.

Credit goes to: https://help.nextcloud.com/t/i-have-latest-nextcloudpi-version-but-dated-php-version/161951. Not sure why I couldn't find that post until now.

monolyf, 2023-06-16:
I too was getting the same PHP version error

Updating to version 25.0.7 via the terminal triggered PHP to update to version 8.x and then i was successfully able to update to NC 26

TiPi-Miou-Miou, 2023-06-28:
Hi.

Happy to see I'm not the only one in this loop hole !
I only do ncp updates via web page.
I'll try the way indicated by @chronohart 
Fingers cross...

githubtefo, 2023-07-10:
What should we do the ones that skipped 5.0.26 and are already in 5.0.28 with PHP7.4?

mwip, 2023-07-19:
I, too, face this issue now. Are there any solutions by now. AFAIK, Debian 11 which OP and I are running does not ship PHP8. Will NCP bring its own PHP?

All input is appreciated.

FadeFx, 2023-07-23:
@chronohart according to your logs you suffer from an old issue as well, there is a duplicate symlink to Zend opcache... see details hrere: https://github.com/nextcloud/nextcloudpi/issues/416#issuecomment-366183822

chronohart, 2023-08-01:
> @chronohart according to your logs you suffer from an old issue as well, there is a duplicate symlink to Zend opcache... see details hrere: [#416 (comment)](https://github.com/nextcloud/nextcloudpi/issues/416#issuecomment-366183822)

Thanks for spotting that! I'll look into it.

mcsitter, 2023-09-05:
> I may have had a similar issue. Check your ncp.cfg file and try set the line for php version from 8.1 to 7.4. After that i think i ran ncp-update which then updated php to 8.1.

Changing the line did not change the behavior of ncp-update-nc for me. The ncp-update-nc script checks if the PHPVER (set in library.sh which reads from ncp.cfg file) is more recent than 8.1. Even though 8.1 is not installed, the ncp.cfg indicates so and does so again after every attempt to upgrade nc. The "Upgrading PHP..." line never gets echoed as  you can see in the attached log. I had upgraded from 24.something to 25.0.6 (which installed 25.0.6.1). The attempt to update to 26.0.3 failed. An attempt to upgrade to 25.0.7 did not fix the issue and I still cant install 26.0.3.

I feel like this is a bug in the update script. The linked forum post indicated a solution with an updated script, which apparently put the Upgrading of PHP at the top of the script. Maybe this is a potential fix?

I am on the current ncp version v1.52.4, which runs on by Raspberry Pi.

[log.txt](https://github.com/nextcloud/nextcloudpi/files/12517614/log.txt)

githubtefo, 2024-02-18:
Hi there - we are still stuck in NC 25 and support is an unsupported version.
Please, we need a **safe** way to update to PHP8 and therefore NC26 the users that skipped 5.0.28 without noticing it was the one updating PHP.
Thank you for your help.