nextcloud/nextcloudpi #1783: Open port check fails because a url is not accesseble

robgerrit, 2023-06-12:
The " System Info" page reports ports 80 and 443 are closed. This, while other tools (including https://www.yougetsignal.com/tools/open-ports/) indicate that the ports are open. A small investigation shows that the site https://checkreferencedport.zendai.net.eu.org/ referenced in ncp-diag in the browser returns this response: " Error conncecting to checkport.zendai.net.eu.org. PR_CONNECT_RESET_ERROR. 

Output of `sudo ncp-report` -->
[ncp-report230612.txt](https://github.com/nextcloud/nextcloudpi/files/11718615/ncp-report230612.txt)

ZendaiOwl, 2023-06-12:
Thank you for raising the issue and letting us know

I noticed the issue as well a little earlier and fixed it, it works again for me at least when testing, so it should be resolved.

Can you try again and see if it works for you now?

ZendaiOwl, 2023-06-12:
Great! I'm glad to hear it's working again for you as well

Since the issue seems to be fully resolved I'm closing this issue

closed 2023-06-12 by ZendaiOwl