nextcloud/nextcloudpi #1786: Backup too slow

bassman56, 2023-06-19:
**Description**  

The current daily backup now takes several hours to complete. The nextcloud is in maintenance mode all the time and not accessible. Why not create a snapshot (I use btrfs) and backup the snapshot instead? then we can go out of maintenance mode within a few seconds?

**Example**  
#
no change in interface. maybe an additional indicator whether backup is running in the background or not.

FadeFx, 2023-07-22:
Backup seems to work normally for me,  do you have data included in your backup and where do you store your backups (backup location)?

bassman56, 2023-07-22:
Hi,

I put all data into the backup, so the backup has about 350GB. The backup goes to a locally connected USB drive, capable of 50..100Mb/s So the whole backup blocks the nextcloud for about 2 hours in maintenance mode, with tendency to increase with more data on the nextcloud in the future.

 

My question: can you change the backup procedure to first make a snapshot (my data is on btrfs), then go out of maintenance and do the backup of the snapshot in the background?

Best Regards Christian

FadeFx, 2023-07-22:
You can BC-snapshot-sync to the other HDD and do the backup to that path and exclude data. I think nc-snapshot-sync does not use maintainance mode, should not make sense anyway...