nextcloud/nextcloudpi #1787: NCP Docker [v1.51.4]: Cannot configure nc-backup-auto due to missing parameter field BACKUPHOUR

florianhenig, 2023-06-25:
Setting up nc-backup-auto exits with the error message **ERROR: 'BACKUPHOUR' must be a number between 0 and 24, was: '$BACKUPHOUR'**.

I was able to narrow down the issue to a missing parameter line for this new introduced parameter both within **ncp-config** and **ncp-web**.

A quick workaround for me was setting `BACKUPHOUR` hardcoded inside the script [`nextcloudpi/bin/ncp/BACKUPS/nc-backup-auto.sh`](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup-auto.sh) after line 50.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.51.4
NextcloudPi image    NextcloudPi_docker_05-01-23
OS                   Debian GNU/Linux 11. 5.15.84-v8+ (aarch64)
automount            no
USB devices          sda sdb sdc 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      264G/916G
rootfs usage         10G/29G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               158days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "5ea7818fb822",
            "11": "88.76.93.112",
            "1": "172.17.0.2",
            "14": "5ea7818fb822",
            "6": "192.168.188.99",
            "15": "henig.duckdns.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.3.2",
        "overwrite.cli.url": "https:\/\/5ea7818fb822\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/data\/nextcloud\/data\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false,
        "twofactor_enforced": "true",
        "twofactor_enforced_groups": [],
        "twofactor_enforced_excluded_groups": [],
        "htaccess.RewriteBase": "\/",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "ssl",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "465",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "logfile": "\/data\/nextcloud\/data\/nextcloud.log",
        "data-fingerprint": "3bc47199a1220c0630c43df7c7541864"
    }
}

```
</details>

pjaydev, 2023-07-02:
same problem here. 

@florianhenig Thanks a lot for the workaround! I had to set BACKUPHOUR right _before_ line 50.

theCalcaholic, 2023-07-17:
Should be fixed in v1.52.1

closed 2023-07-17 by theCalcaholic