nextcloud/nextcloudpi #1790: Nextcloud 28 support

sunjam, 2023-06-28:
For internal testing and notes. Do not expect support!!!!!! Please do report testing issues here.
[Release Schedule](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule)
Dates TBD

- https://github.com/nextcloud/server/issues/36437
  - "Drop support with Nextcloud 29 if v28 still supported it"

[Nextcloud 28 Milestones](https://github.com/nextcloud/server/milestone/224)

sunjam, 2023-08-06:
It seems that the numbering scheme is changing and instead of 28 the next major release cycle will be for 27.1
Feature freeze 8/1 and final release 9/16 for the conference, see maintenance release schedule above.

REAPERSbattlecry, 2023-12-13:
Was curious to test NC 28 because of the Nextcloud announcement. Update from NC 27.1.3 went smooth and without issues. Seems like everything is running well.

schoetju, 2024-04-01:
NC 28 has been made available with https://github.com/nextcloud/nextcloudpi/pull/1877 .
Issue can be changed to "closed" now.

closed 2024-04-02 by sunjam