nextcloud/nextcloudpi #1794: 503 Service Unavailable Error

OfficialMuffin, 2023-07-16:
My NextcloudPi server has suddenly stopped working. I'm thinking it may have been a power cut that happened about a week ago.
I have checked the output of `/var/log/php8.1-fpm.log` and every time I try accessing the local IP of the server, these lines are printed out each time the page is refreshed:

```
[16-Jul-2023 14:53:15] WARNING: [pool www] child 21600 exited on signal 11 (SIGSEGV) after 1239.565783 seconds from start
[16-Jul-2023 14:53:15] NOTICE: [pool www] child 25093 started
```

Here is the output of `ncp-report`:

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.52.0
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.0.img
OS                   Debian GNU/Linux 11. 5.15.61-v8+ (aarch64)
automount            yes
USB devices          sda sdb 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      500G/917G
rootfs usage         500G/917G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    26.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               39min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "next",
            "1": "192.168.1.130",
            "2": "***REMOVED SENSITIVE VALUE***",
            "14": "next",
            "11": "***REMOVED SENSITIVE VALUE***"
        },
[... 100 lines cut ...]
```

{"reqId":"5f4j8rim7rHLkCSfqt5y","level":3,"time":"2023-07-14T19:18:38+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"dns_get_record(): A temporary server error occurred. at /var/www/nextcloud/lib/private/Http/Client/DnsPinMiddleware.php#84","userAgent":"--","version":"26.0.1.1","data":{"app":"PHP"}}
{"reqId":"5f4j8rim7rHLkCSfqt5y","level":3,"time":"2023-07-14T19:18:48+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"dns_get_record(): A temporary server error occurred. at /var/www/nextcloud/lib/private/Http/Client/DnsPinMiddleware.php#84","userAgent":"--","version":"26.0.1.1","data":{"app":"PHP"}}
{"reqId":"5f4j8rim7rHLkCSfqt5y","level":3,"time":"2023-07-14T19:18:58+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"dns_get_record(): A temporary server error occurred. at /var/www/nextcloud/lib/private/Http/Client/DnsPinMiddleware.php#84","userAgent":"--","version":"26.0.1.1","data":{"app":"PHP"}}
{"reqId":"5f4j8rim7rHLkCSfqt5y","level":2,"time":"2023-07-14T19:19:08+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 6: Could not resolve host: apps.nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://apps.nextcloud.com/api/v1/apps.json","userAgent":"--","version":"26.0.1.1","data":{"app":"appstoreFetcher"}}
{"reqId":"5f4j8rim7rHLkCSfqt5y","level":2,"time":"2023-07-14T19:19:20+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"5f4j8rim7rHLkCSfqt5y","level":2,"time":"2023-07-14T19:19:30+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"OZHO2Hk4Eh33NSa7lYcg","level":2,"time":"2023-07-14T20:20:12+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"OZHO2Hk4Eh33NSa7lYcg","level":2,"time":"2023-07-14T20:20:22+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"PoQSVydB3s1dU6j1IfIj","level":2,"time":"2023-07-14T21:25:24+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"PoQSVydB3s1dU6j1IfIj","level":2,"time":"2023-07-14T21:25:24+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"EhtepijmDhtVuNB7FOmm","level":2,"time":"2023-07-15T10:40:15+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"EhtepijmDhtVuNB7FOmm","level":2,"time":"2023-07-15T10:40:25+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"khfhYsLG1MLwW1PmN0ia","level":2,"time":"2023-07-16T12:20:46+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"khfhYsLG1MLwW1PmN0ia","level":2,"time":"2023-07-16T12:20:56+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"https://nextcloud.com/blog/feed read error : cURL error 6: Could not resolve host: nextcloud.com (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://nextcloud.com/blog/feed","userAgent":"--","version":"26.0.1.1","data":{"app":"news"}}
{"reqId":"sPCG2jy1nqMgtXLxrXJq","level":4,"time":"2023-07-16T13:19:25+00:00","remoteAddr":"","user":"--","app":"nextcloudpi","method":"","url":"--","message":"Error during app loading: read error on connection to /var/run/redis/redis.sock:0","userAgent":"--","version":"26.0.1.1","exception":{"Exception":"RedisException","Message":"read error on connection to /var/run/redis/redis.sock:0","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":56,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":217,"function": [... cut ...] le.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":56,"message":"Could not boot workflowengine: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Could not boot workflowengine: Redis server /var/run/redis/redis.sock:0 went away"}}

```
</details>

theCalcaholic, 2023-07-16:
You seem to have 2 problems:
1. Your redis service is down (check `sudo systemctl restart redis` and `systemctl status redis`)
2. Your notify_push service is not installed correctly (but that shouldn't prevent NC from working overall)

Please refer to the nextcloud forum for further support requests, though. There are many people over there who can help you. :)

closed 2023-07-16 by theCalcaholic

OfficialMuffin, 2023-07-18:
A system reboot should have fixed the redis issue, should it not?

OfficialMuffin, 2023-07-18:
Installing NextcloudPi `v1.52.1` and updating Nextcloud to `26.0.3` has fixed it. It may have overwritten some corrupted files.