nextcloud/nextcloudpi #1807: nextcloud reported as down (but it's up & running)

pkra, 2023-07-30:
I've got myself into a weird position and wanted to see if I can recover form it. After trying (unsuccessfully) to upgrade a rasberry pi to nc v26 a few months ago, I'm currently on ncp v1.52.3 and nextcloud 25.0.7.

In this state, the web UI for ncp gives me `Nextcloud check: error` -- but nextcloud is running just fine.

I had hoped to upgrade myself out of this but keep failing. While NCP updates just fine, updating nextcloud fails for me. Via the web UI, I just get a "red light" without any information in the output box. Via ssh (e.g,. `$ ncp-update-nc 25.0.9`) I get `Nextcloud is currently down`.

I'm wondering if there's a strategy to try to recover from this weird state.

FadeFx, 2023-07-31:
I remember having this behaviour when using different php versions:
https://help.nextcloud.com/t/nextcloud-cron-suddenly-is-giving-an-error/165494

pkra, 2023-07-31:
Thanks @FadeFx ! I had not found that thread before. That sounds about right. I'll try that and continue there if necessary.

closed 2023-07-31 by pkra