nextcloud/nextcloudpi #1811: Nextcloud 29 Support

sunjam, 2023-08-06:
For internal testing and notes. Do not expect support!!!!!! Please do report testing issues here.
[Release Schedule](https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule) -- TBD
[Nextcloud 29 Milestones](https://github.com/nextcloud/server/milestone/234)

- https://github.com/nextcloud/server/issues/36437
  - "Drop support with Nextcloud 29 if v28 still supported it"
- #1810 
- #1790

Schniefel, 2024-08-21:
Any news on this?

theCalcaholic, 2024-08-21:
@Schniefel Expect NC 29 support over the course of this week

Schniefel, 2024-09-01:
I assume, that didn't work as expected. Wouldn't it be better to directly focus on Nextcloud 30?

sunjam, 2024-09-02:
Updates are version by version, so it isn't technically feasible to skip 29.  It'll get there.

theCalcaholic, 2024-09-04:
@Schniefel Unfortunately no, because I got Covid... :(
Once we have super for NC 29, NC 30 will be supported shortly after.
The reason it takes so much work off the included PHP update

Schniefel, 2024-09-04:
Sorry to hear that. I hope you get well soon. :/

danni-storm, 2024-09-05:
@theCalcaholic Take time for yourself, this is a volunteer project and we can wait (even if we don't want to 😉). Get well first!

theCalcaholic, 2024-09-07:
Nextcloud 29 support has been added with NCP v1.55.0

closed 2024-09-07 by theCalcaholic