nextcloud/nextcloudpi #1816: Nextcloudpi reports ports 80 and 443 closed, but they are open

dsmic, 2023-08-18:
The reason seems to be, that the service on

curl --silent --max-time 4 --ipv4 https://checkport.zendai.net.eu.org/check

is not responding.

From a comment in ncp-diag I understand, the service is set up by:

https://github.com/ZendaiOwl

and it is probably down…

Detlef

P.S. output of ncp-diag

bash ncp-diag
NextcloudPi version|v1.52.4
OS|Raspbian GNU/Linux 11. 5.10.103-v8+ (aarch64)
automount|no
USB devices|sda
datadir|/mnt/nextcloud/data
data in SD|no
data filesystem|ext2/ext3
data disk usage|166G/872G
rootfs usage|24G/59G
swapfile|/var/swap
dbdir|/var/lib/mysql
Nextcloud check|error
HTTPD service|up
PHP service|up
MariaDB service|up
Redis service|up
HPB service|down
Postfix service|up
Internet check|ok
Public IPv4|87.170.184.156
Public IPv6|2003:ce:bf1f:7ab8:78f4:cd1f:8b4a:b47d
Port 80|closed
Port 443|closed
IP|192.168.2.3
Gateway|192.168.2.1
Interface|eth0
Certificates|physik.selfhost.eu
NAT loopback|yes
Uptime|16:45

theCalcaholic, 2023-08-23:
@ZendaiOwl Can you check the service?

ZendaiOwl, 2023-08-26:
@theCalcaholic It should be working again, I restarted the app when you pinged me and I have re-deployed it now with some slight changes in the configuration. 

Hopefully it'll keep running without becoming unresponsive after some time, if it happens again I'll destroy the app and create a new one.

dsmic, 2023-08-26:
seems ok,

but should be a good idea if it would not report closed, but „not availablel? I think I remember that something like that was in the code, but it did not work as expected?

ZendaiOwl, 2023-08-28:
The N/A in the code is if a public IP is unavailable from the system it's being run on, not if the application checking the port-forwarding is unavailable

n0rs3, 2026-09-15:
Closing this issue since the checkport service is working now.

closed 2026-09-15 by n0rs3