nextcloud/nextcloudpi #1818: Error when upgrading nextcloud: table nextcloud.oc_jobs doesn't exist

janvlug, 2023-08-21:
I have migrated an old database to nextcloudpi. The old database had no prefixes in the database tables, so I set in the config file dbtableprefix to ''.

But when trying to upgrade nextcloud, I get this:

[image]

table nextcloud.oc_jobs doesn't exist

**System information**

```

NextcloudPi version  v1.52.4
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.3.img
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/nextcloudpi_data/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      1.3T/5.3T
rootfs usage         4.0G/59G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    25.0.9.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               3days

```

janvlug, 2023-08-22:
This error also happens when trying to upgrade from 25.0.9 to 25.0.10:

[ nc-update-nextcloud ] (Tue Aug 22 09:33:53 BST 2023)
Current Nextcloud version 25.0.9.2
Available Nextcloud version 25.0.10
ERROR 1146 (42S02) at line 1: Table 'nextcloud.oc_jobs' doesn't exist

janvlug, 2023-08-22:
There is a 'jobs' table in my nextcloud database, but no 'oc_jobs' database:

MariaDB [nextcloud]> show tables like 'jobs';
+----------------------------+
| Tables_in_nextcloud (jobs) |
+----------------------------+
| jobs                       |
+----------------------------+
1 row in set (0.002 sec)

janvlug, 2023-08-26:
I tried changing the line with the `oc_` prefix removed, but now the update failed on downloading:

[image]

But maybe this is a totally independent issue.

janvlug, 2024-02-25:
I ran again into this issue. In the `/usr/local/bin/ncp-update-nc` script is this fragment:
```
# make sure that cron.php is not running and there are no pending jobs
# https://github.com/nextcloud/server/issues/10949
pgrep -cf cron.php &>/dev/null && { pkill -f cron.php; sleep 3; }
pgrep -cf cron.php &>/dev/null && { echo "cron.php running. Abort"; exit 1; }
mysql nextcloud <<<"UPDATE oc_jobs SET reserved_at=0;"
```
The issue with my Nextcloudpi installation is that it is a migration from a previous installation. My installation does not use the **`oc_`** prefix in the database, so this line fails:
```
mysql nextcloud <<<"UPDATE oc_jobs SET reserved_at=0;"
```
Note however the reference to the **closed** bug `10949` [[WORKAROUND] NC14 RC2: occ upgrade hangs on running cronjob it states](https://github.com/nextcloud/server/issues/10949) in the above fragment of the script.

One of the comments in bug `10949` states:
"We removed this via https://github.com/nextcloud/server/pull/12188"

That merge requests seems to fix bug `9992`: [Cronjob blocking update (forever)](https://github.com/nextcloud/server/issues/9992)

Is my interpretation correct that the part of the script that I quoted above can therefore be removed? If that is indeed the case, removing that part would also fix this issue.

janvlug, 2024-02-25:
Another option to solve this issue would be to get the `dbtableprefix` from the config file: `www/nextcloud/config/config.php`, and use that in the line:
```
mysql nextcloud <<<"UPDATE oc_jobs SET reserved_at=0;"
```

I would have to figure out how to get the value of a php config file in a `/bin/bash` script.

janvlug, 2024-02-25:
This is a command to get the value of `dbtableprefix` in a script:
```
php -r 'include("/var/www/nextcloud/config/config.php"); print ( $CONFIG['"'dbtableprefix'"'] );'
```

janvlug, 2024-03-31:
I created a pull request to solve the database prefix. But maybe setting the whole SQL command that I improved is not necessary any more, as described in: https://github.com/nextcloud/nextcloudpi/issues/1818#issuecomment-1962895946.

theCalcaholic, 2024-05-04:
@janvlug Thank you for the PR! I adjusted it a bit, so that the dynamically detected DB prefix is used everywhere and not just in ncp-update-nc.
It's now in devel and being tested

rustyb78, 2024-08-22:
I'm getting this error with 1.54.3.

theCalcaholic, 2024-09-24:
@rustyb78 Please create a new issue for this. The original issue is fixed, so we need to debug your case separataely

closed 2024-09-24 by theCalcaholic