nextcloud/nextcloudpi #1822: Secure web sockets wss still called at the old domain

bozicm, 2023-08-26:
I've noticed in my developer console that after using a different (public domain, e.g. mycloud.example.com), the JS worker is trying to connect to the default domain from setup. The error I'm receiving:

    Firefox can’t establish a connection to the server at wss://nextcloudpi/push/ws. notifications-vendors-node_modules_nextcloud_initial-state_dist_index_esm_js-node_modules_nextcloud_moment_-4efa53.js:2:54060

I guess there should be a dynamic change to a different `trusted_domains` for wss protocol. But I guess that must be fixed in config.php somewhere else than that directive?

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.52.1
NextcloudPi image    NextCloudPi_RaspberryPi_v1.52.0.img
OS                   Debian GNU/Linux 11. 5.15.61-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      185G/1.9T
rootfs usage         5.7G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    26.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
Uptime               36days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "10": "localhost",
            "9": "mycloud.example.com"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "26.0.1.1",
        "overwrite.cli.url": "https:\/\/mycloud.example.com\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/myCloudDrive\/ncdata\/data\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": 1024,
        "preview_max_y": 1024,
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/media\/myCloudDrive\/ncdata\/data\/nextcloud.log",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "loglevel": "2",
        "log_type": "file",
        "htaccess.RewriteBase": "\/",
        "memories.exiftool": "\/var\/www\/nextcloud\/apps\/memories\/exiftool-bin\/exiftool-aarch64-glibc",
        "memories.vod.path": "\/var\/www\/nextcloud\/apps\/memories\/exiftool-bin\/go-vod-aarch64",
        "memories.gis_type": 1,
        "memories.vod.disable": false,
        "memories.vod.ffprobe": "\/usr\/bin\/ffprobe",
        "memories.vod.ffmpeg": "\/usr\/bin\/ffmpeg",
        "mail_sendmailmode": "smtp",
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpauth": 1,
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sat Aug 26 00:00:00.767978 2023] [ssl:warn] [pid 612087:tid 548130223168] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Aug 26 00:00:00.773054 2023] [mpm_event:notice] [pid 612087:tid 548130223168] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Sat Aug 26 00:00:00.773094 2023] [core:notice] [pid 612087:tid 548130223168] AH00094: Command line: '/usr/sbin/apache2'
[Sat Aug 26 14:55:18.403174 2023] [mpm_event:notice] [pid 612087:tid 548130223168] AH00492: caught SIGWINCH, shutting down gracefully
[Sat Aug 26 14:55:18.529829 2023] [ssl:warn] [pid 2179534:tid 547895940160] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Aug 26 14:55:18.558657 2023] [ssl:warn] [pid 2179538:tid 547895940160] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Aug 26 14:55:18.563592 2023] [mpm_event:notice] [pid 2179538:tid 547895940160] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Sat Aug 26 14:55:18.563817 2023] [core:notice] [pid 2179538:tid 547895940160] AH00094: Command line: '/usr/sbin/apache2'
[Sat Aug 26 14:59:37.180332 2023] [mpm_event:notice] [pid 2179538:tid 547895940160] AH00492: caught SIGWINCH, shutting down gracefully
[Sat Aug 26 14:59:37.288084 2023] [ssl:warn] [pid 2179720:tid 548272579648] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Aug 26 14:59:37.312269 2023] [ssl:warn] [pid 2179722:tid 548272579648] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Aug 26 14:59:37.321590 2023] [mpm_event:notice] [pid 2179722:tid 548272579648] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Sat Aug 26 14:59:37.321737 2023] [core:notice] [pid 2179722:tid 548272579648] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

bozicm, 2023-08-26:
Ok, I figured it out by reading the https://github.com/nextcloud/notify_push

Just running `ncc notify_push:setup` with the config listed above did the trick. But I guess there could be some automation or warning for that?

thelner, 2025-01-14:
I had the same issue, `ncc notify_push:setup` after updating `config/config.php` did the trick.
My build is also nextcloudpi. Wondering if this config not getting updated issue is related to their update/config scrips.