nextcloud/nextcloudpi #1827: Latest Version Update

luke6464, 2023-09-12:
Latest Version Update
Hi, I updated the Nextcloudpi container from Portainer by redownloading and recreating the image of the latest version of Nextcloudpi and the update occurred correctly from the previous version (1.51.3) but now the panel shows 1.51.6 as the latest version.
I tried to re-download and recreate the last image again from Portainer but the panel update does not go beyond version 1.51.6.
How is it possible?
Where am I wrong?
I also attach the screenshot of the system information from the panel.
Despite the Nextcloud check error everything works correctly!
What caused that error?
How to correct it?
Thank you
[image]

pkra, 2023-09-12:
This sounds a lot like what I had been experiencing at https://github.com/nextcloud/nextcloudpi/issues/1807. Maybe the linked thread on that issues can help you, too.

luke6464, 2023-09-12:
> This sounds a lot like what I had been experiencing at #1807. Maybe the linked thread on that issues can help you, too.

I checked what was described in that post regarding the PHP version but everything seems fine as you can see from the attached screenshot... the PHP version is 8.2.
Other ideas?
Thank you
[image]

luke6464, 2023-09-13:
> Latest Version Update
> Hi, I updated the Nextcloudpi container from Portainer by redownloading and recreating the image of the latest version of Nextcloudpi and the update occurred correctly from the [...]

Guys, does anyone have any thoughts on this issue? ...
I can't update the image to the latest version beyond 1.51.6.
Thank you

getwet, 2023-09-15:
Hi luke6464,
I had this problem this morning. I was wondering that my backups was not running since 09/12.
Backup setting shows an error:
An unhandled exception has been thrown:
Error: Interface "OCA\Files_Versions\Versions\INeedSyncVersionBackend" not found in /var/www/nextcloud/apps/groupfolders/lib/Versions/VersionsBackend.php:46
Stack trace: ...
And on Nextcloud I saw some app updates for groupfolders and more. After the updates everything was good and backups running again. Additionally on Nextcloudpi panel everything was green again.
I don´t know why it is connected to each other, but this was my fix.
Hopefully it will do for you the same.

paeddi, 2023-09-15:
> Hi luke6464, I had this problem this morning. I was wondering that my backups was not running since 09/12. Backup setting shows an error: An unhandled exception has been thrown: Error: Interface "OCA\Files_Versions\Versions\INeedSyncVersionBackend" not found in /var/www/nextcloud/apps/groupfolders/lib/Versions/VersionsBackend.php:46 Stack trace: ... And on Nextcloud I saw some app updates for groupfolders and more. After the updates everything was good and backups running again. Additionally on Nextcloudpi panel everything was green again. I don´t know why it is connected to each other, but this was my fix. Hopefully it will do for you the same. But finally my Panel shows: NextcloudPi version | v1.52.4

That fixed it for me too.
Thank you!

getwet, 2023-09-15:
> > Hi luke6464, ...
> 
> 
> 
> That fixed it for me too.
> 
> Thank you!

With pleasure 😊👍

getwet, 2023-09-15:
You're wellcome 😊👍

getwet, 2023-09-15:
@luke6464 
I'm not much familiar with docker, but is it normal that your ports are closed?

luke6464, 2023-09-20:
Thanks guys,
I solved it by changing the php version from 8.2 to 8.1 nel container di Nextcloud.