nextcloud/nextcloudpi #1829: NextCloudPi GUI suddenly looks a mess

luke6464, 2023-09-20:
Hi,
I updated the NextcloudPi version to the latest 25.0.9.2 (Panel v.1.51.6) and after a few days the GUI was completely corrupt (see attached screenshots).
I saw another topic similar to the following link where it was solved with the “clear-op-cache” command from the control panel but unfortunately in my case it didn’t work.

[https://help.nextcloud.com/t/nextcloudpi-web-gui-suddenly-looks-a-mess/156208](url)

I also performed a system reboot and cleared the browser cache but the problem remains.

Any other ideas to solve the problem?

Thank you

Allego anche il file di Logs del pannello ...
[ncp_report_logs.txt](https://github.com/nextcloud/nextcloudpi/files/12673993/ncp_report_logs.txt)

Screenshots
-----------------------
[image]
[image]
[image]

manankanchu, 2023-10-12:
Facing the same issue since several weeks .... web interface of NC renders useless in the mean time, cannot even scroll ...

manankanchu, 2023-10-16:
> Facing the same issue since several weeks .... web interface of NC renders useless in the mean time, cannot even scroll ... seems like no CSS loaded but that may be a secondary effect

Could fix it:

- log into Pi via SSH
- sudo ncp-config
- go to "tools"
- execute "clear-php-opcache"