nextcloud/nextcloudpi #1832: Raspberry Pi 5 support

sunjam, 2023-09-29:
Raspberry Pi 5 has been announced.  Launches [end of October](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/).  fyi, for planning purposes and image support.

christoph-bittmann, 2023-10-30:
For Rapberry Pi 5 the new debian bookworm based raspbian os is needed [1]. Official an upgrade is not possible / recommended. But the boys from the alternative dietpi distribution has released an upgrade script [2]. Maybe it can adopted for the nextcloudpi project as well?

* [1] https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/
* [2] https://dietpi.com/blog/?p=2809#upgrade

theCalcaholic, 2023-10-30:
@christoph-bittmann We won't be able to support debian bookworm for at least the next 3 versions of NCP. In particular, I want to get rid of the dependency of the custom PHP repository before doing an OS upgrade, because that combination is very error prone and has caused many issues in the past.
In the next two versions, support for docker containers within NCP will be introduced and thereafter, individual services will be moved to containers. That will reduce the risk of system upgrades significantly, as individual services won't be affected by them.

Once that's working, we'll move to bookworm

darguez, 2023-12-27:
Hey, good job keeping ncp up and running! Any estimated date to support Raspberry Pi 5? For those of us who already have a board, it would be interesting to know if it is worth waiting or trying other alternatives in the meantime.

sunjam, 2023-12-27:
Try testing it and letting us know.  Most people do not have one.

darguez, 2023-12-27:
> Try testing it and letting us know. Most people do not have one.

I've already done it, after installing Raspberry Pi OS 64bit I ran as root:

`# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`

and I get the following message:
```
Installing NextCloudPi...
ERROR: distro not supported:
Debian GNU/Linux 12 \n \l
```

theCalcaholic, 2023-12-27:
@darguez If all goes well with the next releases, I can see NCP supporting bookworm (and therefore the RPi5) in 2024-03

godalming123, 2023-12-29:
I know that this applies to the script, but does it also apply to the raspberry pi image in the releases. I've used raspberry pi imager on the img file that was created by extracting the zip, enabled USB boot for 25 watt power supplies, but my pi (connected with thernet) doesn't show up on the WiFi router after booting, and I can't connect a display because I don't have a micro hdmi cable yet.

theCalcaholic, 2024-01-02:
@godalming123 I have no way to test this, unfortunately, because I don't own an RPi 5. However, there might be a way to get support for the RPi 5 early than I was hoping via Armbian (instead of Raspbian). I'm still exploring that option though, so no promises.

godalming123, 2024-01-07:
@theCalcaholic now that my micro-HDMI to HDMI cable has arrived I can just use ubuntu with the nextcloud snap.

Lyndonsource, 2024-01-08:
Fresh install of Armbian I ran as root:

# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash

same error : ERROR: distro not supported

Lyndonsource, 2024-01-08:
Fresh install of Armbian I ran as root:

#: curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash

same error : ERROR: distro not supported

sunjam, 2024-01-08:
That is because Bookworm support is in development.
https://github.com/nextcloud/nextcloudpi/issues/1730

Gepamal, 2024-03-02:
On a fresh raspberry Pi 5 with the full 64 bit version, I just ran:

`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash` 

 I got: 
`Getting build code...
Cloning in '/tmp/nextcloudpi.sJmta8/nextcloudpi'...`
`Failed to connect to github.com port 443 after 136053 ms: Couldn't connect to server`

I guess we'll have to wait a bit :)

blueskyfeeling, 2024-03-09:
In the meantime, is there an outlook when NextcloudPI can be installed on the Raspberry Pi5?

Markistobud, 2024-04-11:
Hi, 

I've just try instal the .img on my raspy5 and i've got the same error message.

Do you know when nextcloudpi will be working on raspy 5 ?

Thanks for you work !

sunjam, 2024-04-12:
Look at the original links in this issue.

isketerol64, 2024-04-29:
I can assist with testing on a bona fide Raspberry Pi 5 this week.  Is there a test suite to be executed, or do you want ad hoc testing to be completed?

sunjam, 2024-04-29:
There is an image in the previous reply you can test. ncp-report is built in.

theCalcaholic, 2024-05-02:
Support for the Raspberry Pi 5 has now finally landed with v1.54.0

closed 2024-05-02 by theCalcaholic

teledyn, 2024-05-04:
I've been running it for two days now, upgrade was smooth, no obvious
issues (but pacing the rug waiting for my fave app devs to catch up to
Nextcloud 29 😅

theCalcaholic, 2024-05-04:
@teledyn NC 29 will also probably be the next milestone for NCP (including PHP 8.3) :)