nextcloud/nextcloudpi #1834: Link to Helper Script Wrong

rocket-plane, 2023-09-30:
Hi,

on the website [https://nextcloudpi.com/](https://nextcloudpi.com/) under LXD/LXC it says, that I should call
`bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/ct/nextcloudpi-v5.sh)"`
for proxmox installation. But that link results in a 404: [https://github.com/tteck/Proxmox/raw/main/ct/nextcloudpi-v5.sh](https://github.com/tteck/Proxmox/raw/main/ct/nextcloudpi-v5.sh)
It would be nice if this would be fixed.

theCalcaholic, 2023-10-01:
Thank you for the heads-up. That's outdated, you can now just use the official LXC container from https://github.com/nextcloud/nextcloudpi/releases to install NCP on Proxmox

theCalcaholic, 2024-11-08:
I fixed the link and added the documentation for the LXC image installation method

closed 2024-11-08 by theCalcaholic