nextcloud/nextcloudpi #1840: nextcloudpi icon for the app hits a CORS exception

teledyn, 2023-10-18:
uBlockOrigin blocks https://ownyourbits.com/category/nextcloudpi/ so I couldn't look there.

https://github.com/nextcloud/nextcloudpi/wiki/FAQ makes no mention of CORS

the documentation page for the nextcloudpi app is a 404, but the link to report leads to here, and I could find no mention of the app called nextcloudpi, this appears to be about the distro nextcloudpi (which I am enjoying). So I don't even know what that app would do if it worked, however it appears by default on the dashboard, and when clicked it displays a text page with only a CORS error, the user is then locked out and sometimes the server is left in a locked state.

considering the /settings/apps/installed/nextcloudpi shows this app as version 0.0.1 I'm hoping I can just remove it.

f8ttyc8t, 2023-11-11:
Same here, results in {"message":"CORS requires basic auth"}
Also https://apps.nextcloud.com/apps/nextcloudpi results in "not found" response page.

Otherwise: nextcloudpi is a real gem, thanks for sharing it!

Wrong-Code, 2024-10-25:
Having noticed that this issue is still open with no solution in sight, I'll add my 2 cents (IMHO)

The main problem is that the application portal is not using basic authentication, while the NCP admin panel needs/pretends basic authentication, hence the CORS message. As someone has noticed elsewhere, by reloading the failed page the admin panel should start, asking for the user/password and using indeed the basic authentication.

There is however another subtle catch here: if you access Nextcloud via a reverse-proxy address, which is quite typical, your reverse-proxy configuration *must* also include a rewrite for the 4443 port, because this is exactly what the default NextcloudPi app does: adding :4443 to whatever host URL you have used to access Nextcloud. This may work, or not, and it didn't in my case exactly because there wasn't an entry in my Caddyfile dealing with the 4443 port.

Given the above, I'd say that the NextcloudPi app is not worth the hassle. You can obtain the same result by having a bookmark in your browser pointing directly to the NextcloudPi admin panel URL. I ended up disabling the NextcloudPi app.

N0m4n904, 2026-09-04:
I am facing the same problem on a new installation. unfortunately I cannot enable SSH without logging into the pi. What is the default password of the local user? I already tried pi/raspberry like it was in the past, but this did not work