nextcloud/nextcloudpi #1844: Add Orange Pi 5 Plus board

dedsecproyt, 2023-10-28:
**Description**  
I need clear image NextCloudPI for Orange Pi 5 Plus

**Example**  
Orange Pi 5 Plus more batter then RPI-5. Orange PI has two 2.5GB ethernet for faster transfer data. more powerfull cpu for hard work or many users. has a lot of ram (up to 32GB). has M.2 SSD and M.2 WIFI slot for easy upgrade. have external 16GB/32GB/64GB/128GB/256GB eMMC module

PeterGamma, 2024-09-05:
Since Corona we don t trust in Rasbperry Pi anymore. The Orange Pi 5 and the Orange Pi zero 2 w are our alternatives. And we miss a NextCloud image for these SBCs.

theCalcaholic, 2024-09-06:
There is now a hardware survey for NCP: https://cryptpad.fr/form/#/2/form/view/M7tS0U81qlmc4HsQzrJist5HudbCxBfaN6Zz3HL-tz0/

But yes, Orange Pi 5 support is likely to happen :)

PeterGamma, 2024-09-06:
This is good news. The Orange Pi 5 is one of the best Rasbperry Pi alternatives so far according to my own little experience. And I would like to avoid to have a stock of Raspberry Pi's in my screed to be save during the next Corona crisis. Support of the Orange Pi 5 would solve this problem.

theCalcaholic, 2024-09-07:
The Orange Pi 5 and 5 plus are now officially supported as of v1.55.0

closed 2024-09-07 by theCalcaholic

PeterGamma, 2024-09-09:
Thank you, theCalcaholic, I am very happy to hear about that.