nextcloud/nextcloudpi #1849: Deprecate Docker information from readme and other documentation

sunjam, 2023-11-18:
Documentation needs to be refined to not encourage docker usage as it has been deprecated.

theCalcaholic, 2023-11-20:
Agreed

ZendaiOwl, 2023-11-20:
Removed the Docker information #1850

sunjam, 2023-11-22:
Well, still worth linking to documentation on why there is no longer Docker support, yes?

> Removed the Docker information from the README #1851
> 
> EDIT: Made a new PR without the trailing commits

Looks good, but we do need to still include at least linkage to why Docker support was removed.  Also, I want to not 25 is reaching EOL, which means it is worth reiterating that Docker support ends with it on the ncp end.

ZendaiOwl, 2023-11-22:
Alright, I've got a meeting with my handler in university in 50 minutes for my graduation exam so I can change it after that

closed 2024-03-30 by theCalcaholic