nextcloud/nextcloudpi #1853: Reversproxy

markuskraus83, 2023-11-30:
Die Reverse-Proxy-Header-Konfiguration ist fehlerhaft oder du greifst auf Nextcloud über einen vertrauenswürdigen Proxy zu. Ist dies nicht der Fall, dann besteht ein Sicherheitsproblem, das einem Angreifer erlaubt, die IP-Adresse, die für Nextcloud sichtbar ist, auszuspähen.

Ich habe kein Reversproxy

Aktuelle Version Nextcloudpi 27.1.4
Wie kann ich den Fehler  beheben?

[image]

markuskraus83, 2023-12-03:
Ist wieder von alleine Verschwunden???

ADadlDudlDa, 2023-12-20:
Ich habe das identische Problem mit Version NextCloudPi 27.1.3

markuskraus83, 2023-12-24:
> Ich habe das identische Problem mit Version NextCloudPi 27.1.3

Jetzt ist auch wieder bei mir da? Sehr kommisch

joshtrichards, 2025-02-04:
If still experiencing this, please post on the help forum for troubleshooting assistance:

https://help.nextcloud.com

https://github.com/nextcloud/nextcloudpi?tab=readme-ov-file#links

closed 2025-02-04 by joshtrichards