nextcloud/nextcloudpi #1854: CVE-2023-48239 in nextcloud server

Colfenor, 2023-12-05:
Heyo, 

in the nextcloud server project a high severe CVE has been reported:

https://github.com/nextcloud/security-advisories/security/advisories/GHSA-f962-hw26-g267

which recommends that the Nextcloud Server is upgraded to quote `25.0.13, 26.0.8 or 27.1.3`.

Is there an ETA on when release a new docker image with the upstream merged patch ?

greetings !

REAPERSbattlecry, 2023-12-09:
The nextcloudpi-docker-version is EOL at the moment.
[Look here.](https://help.nextcloud.com/t/nextcloudpi-planning-to-discontinue-its-docker-version-with-nc-25/158895)

So if you are still on docker it is the best to migrate to another ncp instance.

theCalcaholic, 2024-04-22:
@REAPERSbattlecry is right, unfortunately. You can always try to update to specific Nextcloud versions manually, but they will not be tested.

closed 2024-04-22 by theCalcaholic