nextcloud/nextcloudpi #1856: Add booksworm support for install.sh script

djamn, 2023-12-20:
**Description**  
Currently, in etc/ncp.cfg the config entry of the release is "release": "bullseye" but in the meantime, booksworm (one version above bullseye was released stable) and the install.sh returns an error

ttavenard, 2023-12-21:
same problem here, do we just have to modify the ncp.cfg file ?

theCalcaholic, 2023-12-27:
bookworm is not yet supported and will not be until a bit of groundwork has been done (do not expect bookworm support before 2024-03).

closed 2023-12-27 by theCalcaholic

ovpc, 2024-01-15:
@theCalcaholic Adding this info her as it has to do with bookworm and PHP8.2
Im using NextCloudPi v1.53.0 in a Proxmox VM
Using Passwords 2023.12.32 and get notified it is the the last update for NC27 and PHP8.1
see https://git.mdns.eu/nextcloud/passwords/-/wikis/Administrators/Notifications/Platform-Support-Notification 
It includes info about legacy support so there is no reason to worry, the app will continue to work.
I will wait for NCP to support bookworm and PHP8.2.
Let me know if need testing update-os-ncp or install.sh in Bookworm.

theCalcaholic, 2024-01-15:
@ovpc We will soon run Nextcloud via docker, so it will always be inside a container that contains a recent os and PHP version.
That will also greatly simplify the upgrade process for the host OS in the future