nextcloud/nextcloudpi #1861: PHP Version auto update breaks NextCloudPI 1.53.0

rolfkra, 2024-01-05:
Just installed NextCloudPi, latest version on a Raspi 4B (NextCloudPi_RaspberryPi_v1.53.0.img) in Dec 2023.
Today I can see that error : 

This version of Nextcloud is not compatible with PHP>=8.3.<br/>You are currently running 8.3.1.
Nextcloud is still up, Cronjobs are not working anymore and several parts of NextcloudPi Panel report this error.

Seems the Auto Update of the PHP Version of the system breaks parts of the Instance.

djamn, 2024-01-07:
@rolfkra what's your OS? Bookworm or Bullseye?

You can find it out via lsb_release -a

rolfkra, 2024-01-07:
Hello @djamn ,
I am using bullseye.
Meanwhile found a workaround by fixing the PHP to 8.1 via
sudo update-alternatives --config php

naldoxable, 2024-01-29:
I managed to downgrade the version using:
`sudo update-alternatives --set php /usr/bin/php8.1`