nextcloud/nextcloudpi #1865: The documentation site docs.nextcloudpi.com is not loading its content

tjhayes, 2024-01-26:
I'm about to upgrade from Pi 4 to Pi 5 and need to get the backup and restore details right. I wanted to read some of the documentation at: docs.nextcloudpi.com

The site loads, but the article list never appears. Can someone fix the site, or is there a different place to find the documentation for nextcloudpi?  This sounds like what happened [here](https://github.com/nextcloud/nextcloudpi/issues/1784).

Thank you!

schoetju, 2024-01-26:
[docs.nextcloudpi.com](https://docs.nextcloudpi.com) is back online (at least to me). 
FYI: docs.nextcloudpi.com does "only" the presentation of all available NextcloudPi Documentation from 
[here](https://help.nextcloud.com/c/ncpdocs/178)
Basically it's a better UI (compared to help.nextcloud.com) and it includes a search feature.

tjhayes, 2024-01-26:
It's now up for me, too. Thank you!

closed 2024-01-26 by tjhayes