nextcloud/nextcloudpi #1876: Clean Install Max Upload Is 8M

TurtleP, 2024-02-23:
I set up NextcloudPi (fresh install) on my Raspberry Pi 4B yesterday and kept getting "Connection Closed" on the Nextcloud Desktop application. Upon further investigation, uploading files was failing on the web as well citing "Nextcloud read XXX bytes, but Nextcloud wrote 8192 bytes ...". I did some more digging into this error and it turns out that `/etc/php/8.1/fpm/php.ini` had its upload setting set to a maximum of `8M`. Once I upped this value to 16G and ran `sudo systemctl restart php8.1-fpm.service` the problem went away.

It took a bit too long for me to find the solution, but I'm hoping this raises awareness for you guys to fix this in the future. Or at least let users who are setting this up have some way to configure it during setup or in the web dashboard/`ncp-config`.

Hopefully if someone else runs into this issue, they will see this and know how to solve it.

whereswaldon, 2024-04-18:
This problem bit me too, but your fix works. Thanks for saving me even more time debugging!

theCalcaholic, 2024-04-20:
This is configurable in the admin interface via CONFIG/nc-limits. However, it's still strange that you're seeing this, because the default value should be 10G.

whereswaldon, 2024-04-22:
I have this in `/etc/php/8.1/fpm/conf.d/90-ncp.ini`:
```
; disable .user.ini files for performance and workaround NC update bugs
user_ini.filename =

; from Nextcloud .user.ini
upload_max_filesize=10G
post_max_size=10G
memory_limit=715364352
mbstring.func_overload=0
always_populate_raw_post_data=-1
default_charset='UTF-8'
output_buffering=0

; slow transfers will be killed after this time
max_execution_time=3600
max_input_time=3600
```

But I still had to edit:
```
; Maximum size of POST data that PHP will accept.
; Its value may be 0 to disable the limit. It is ignored if POST data reading
; is disabled through enable_post_data_reading.
; https://php.net/post-max-size
post_max_size = 16G

; Maximum allowed size for uploaded files.
; https://php.net/upload-max-filesize
upload_max_filesize = 16G
```

I can't see why the drop-in config file isn't taking precedence here...

theCalcaholic, 2024-04-22:
Weird, thank you for the context. That will help me to investigate the issue.