nextcloud/nextcloudpi #1878: Upgrade from 26 to 27 fails

joneuhauser, 2024-03-01:
```
Running nc-update-nextcloud
Current   Nextcloud version 26.0.12.2
Available Nextcloud version 27.1.7
Download Nextcloud 27.1.7...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20240301_1709323522.tar.gz generated
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20240301_1709323522-26.0.12.2.tar.gz'...
Install Nextcloud 27.1.7...
Fix permissions...
Upgrade...
Nextcloud is not installed - only a limited number of commands are available
Command "upgrade" is not defined.
Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20240301_1709323522-26.0.12.2.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20240301_1709323522-26.0.12.2.tar.gz...
restore files...
rm: cannot remove '/var/www/nextcloud': Directory not empty
Cleanup...
Rollback failed! Data left at /var/www/recovery/ncp-data.Oet4GK
Done. Press any key... 
```

**System information**

Here's the output of ncp-report, although it's probably not useful anymore at this point since it didn't reinstall nextcloud:

<details>
<summary>NextcloudPi diagnostics</summary>
```
grep: /var/www/nextcloud/config/config.php: No such file or directory
grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.53.1
NextcloudPi image    NextCloudPi_10-03-21
OS                   Debian GNU/Linux 11. 5.10.103-v8+ (aarch64)
automount            no
USB devices          sda sdb
datadir               (doesn't exist)
rootfs usage         5.9G/118G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            enxb827eb96a6b1
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1day
```
</details>
<details>
<summary>Nextcloud configuration</summary>
```
Could not open input file: /var/www/nextcloud/occ
```
</details>
<details>
<summary>HTTPd logs</summary>
```
[Fri Mar 01 00:00:06.180759 2024] [ssl:warn] [pid 122255:tid 548003521600] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name                         [Fri Mar 01 00:00:06.181333 2024] [ssl:error] [pid 122255:tid 548003521600] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=arm-build.pitowers.org / issuer: CN=arm-build.pitowers.org / serial: 7DB21552E75827D0B5E58524BE67A36F9E0AA17A / notbefore: May  7 16:05:36 2021 GMT / notafter: May  5 16:05:36 2031 GMT]
[Fri Mar 01 00:00:06.181375 2024] [ssl:error] [pid 122255:tid 548003521600] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Fri Mar 01 00:00:06.238046 2024] [mpm_event:notice] [pid 122255:tid 548003521600] AH00489: Apache/2.4.56 (Debian) OpenSSL/1.1.1w configured -- resuming normal operations                           [Fri Mar 01 00:00:06.238134 2024] [core:notice] [pid 122255:tid 548003521600] AH00094: Command line: '/usr/sbin/apache2'
```
</details>
<details>
<summary>Database logs</summary>
```
2024-02-29 14:17:29 0 [Note] Starting crash recovery...
2024-02-29 14:17:29 0 [Note] Crash recovery finished.
2024-02-29 14:17:29 0 [Note] Server socket created on IP: '127.0.0.1'.
2024-02-29 14:17:30 0 [Note] Reading of all Master_info entries succeeded
2024-02-29 14:17:30 0 [Note] Added new Master_info '' to hash table
2024-02-29 14:17:30 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.39-MariaDB-0+deb10u2'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2024-02-29 15:02:44 0 [Note] InnoDB: Buffer pool(s) load completed at 240229 15:02:44
2024-02-29 15:03:03 11 [Warning] Aborted connection 11 to db: 'nextcloud' user: 'ncadmin' host: 'localhost' (Got an error reading communication packets)
2024-02-29 16:57:59 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
2024-02-29 16:57:59 0 [Note] Event Scheduler: Purging the queue. 0 events
2024-02-29 16:57:59 0 [Note] InnoDB: FTS optimize thread exiting.
2024-02-29 16:58:00 0 [Note] InnoDB: Starting shutdown...
2024-02-29 16:58:00 0 [Note] InnoDB: Dumping buffer pool(s) to /var/lib/mysql/ib_buffer_pool
2024-02-29 16:58:00 0 [Note] InnoDB: Instance 0, restricted to 4096 pages due to innodb_buf_pool_dump_pct=25
2024-02-29 16:58:00 0 [Note] InnoDB: Buffer pool(s) dump completed at 240229 16:58:00
2024-02-29 16:58:18 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2024-02-29 16:58:18 0 [Note] InnoDB: Shutdown completed; log sequence number 42018538067; transaction id 73407460
2024-02-29 16:58:21 0 [Note] /usr/sbin/mysqld: Shutdown complete
```
</details>
grep: /var/www/nextcloud/config/config.php: No such file or directory
<details>
<summary>Nextcloud logs</summary>
```
tail: cannot open '/nextcloud.log' for reading: No such file or directory
```
</details>

TheShanMan, 2024-05-24:
No response to this? What was your resolution? This is exactly why I don't ever want to upgrade my server. I did it once and the user data on my external drive ended up trashed as a result of upgrade problems (I don't remember the specific details).

Since I'm still on 26 I thought I'd see if upgrades are any more bulletproof than last time, hoping that if nothing else, waiting until newer versions have been patched for critical fixes would give me confidence I wouldn't be forced to start over from scratch yet again.

This report leads me to believe it's still far from bulletproof, and I guess I'll forever be stuck on 26 unless and until I'm ever ready to just install a clean server. :(

MegaMarkey, 2024-07-03:
I had the same problem as @joneuhauser, but when upgrading from 27.0.1.2 to 27.1.3 (NextcloudPi v1.53.0). I finally got around to restoring the instance today. It was actually very easy: I was able to successfully restore the backup that was automatically created during the upgrade process (in `/var/www`) using `nc-restore`. I was then able to update NextcloudPi (using `nc-update`) and then Nextcloud (using `nc-update-nextcloud`) to the latest versions without any errors.

I do not know the exact cause of the problem. However, in an immediately preceding call of `nc-update`, the following lines were returned (within the expected output):
```
/usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
/usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
/usr/local/etc/library.sh: line 494: [: 0sh: integer expression expected
```
This is apparently a problem that was fixed in v1.53.1, see https://github.com/nextcloud/nextcloudpi/issues/1670#issuecomment-1922205605. I suspect a connection with the Nextcloud upgrade problem, but according to the logs @joneuhauser seems to have already used v1.53.1.