nextcloud/nextcloudpi #1879: Format USB drive doesn't respect flag from GUI

anjasumma, 2024-03-02:
newest NCP 1.53.1

In NextcloudPi GUI there's the function to format a USB drive. It has a checkbox that asks whether data should be wiped. Ticking this checkbox does not change anything in the script execution, so I think that's a bug and the value of the flag is not properly read.

Problem description: I have used NCP GUI to move data to USB stick ("stick1"). Some time later, I would like to get another USB stick formatted to use, e.g., for backups ("stick2"). I shut down Raspi, remove stick1, and plug in stick2. Now stick2 is the only one and should be formatted. However, because of the fact that NCP GUI knows that I store my data on a USB stick, it doesn't allow me to format the USB stick even though it's a different one and I checked the checkbox. I always see an error message about this, something like "I cannot find the data directory, so it's probably on that drive, and the flag was not set".

m-breitbach, 2024-03-07:
If I understand your problem correctly, this is a known bug. See PR #1855 for a fix (I hope it gets reviewed soon).

anjasumma, 2024-03-08:
Thank you for your answer. Looks like you understood me correctly and your Code is able to fix that bug.
Thank you for taking the time to contribute. I hope your pr gets approved soon.

theCalcaholic, 2024-08-19:
This will be fixed in the next release, thanks to @m-breitbach

closed 2024-08-19 by theCalcaholic