nextcloud/nextcloudpi #1880: Restoring Backup after drive crash is not possible

anjasumma, 2024-03-02:
newest NCP version 1.53.1

I had a running NextcloudPi instance and took a backup (including data and compressed) and exported NCP configuration to another drive. I run NextcloudPi on a Raspi4 from an SD card and had data directory and database moved to an external SSD drive.

My drive crashed (turns out it's a fake drive and unusable in the long run) and now I would like to use the backups to get my Nextcloud(pi) back up and running with a new, working drive.
However, it doesn't work. Both routines (ncp-import and nc-restore) seem to expect a running database to do their work. But due to the crash I don't have a running database anymore. What I can do is format the new drive to use the same name as the crashed one and I can also create the folders ncdata etc.

Is this a bug in the scripts? Both scripts fail with error messages. 
Or am I doing something fundamentally wrong in restoring the updates? But isn't preparing for hardware failure one of the primary reasons to backup data?

OfficialMuffin, 2024-03-13:
Im curious. are you getting the same error as me in #1883? @anjasumma

theCalcaholic, 2024-11-06:
> Is this a bug in the scripts? Both scripts fail with error messages.
No, the backups are meant to allow you to restore to a point in time in a generally working (e.g. freshly installed) setup.

In the case of catastrophic failures that break system services there is just no reliable way to automatically fix the server and a reinstallation (potentially along with a hardware replacement) is the best course of action

closed 2024-11-06 by theCalcaholic