nextcloud/nextcloudpi #1891: Rolling back to backup after successful upgrade 26.0.3 to 27.0.1 !

ssdrun, 2024-04-02:
Hi,
Each time i perform an upgrade of nextcloud i must stop the process at the end ( with a CTRL+C key) because the script want to rollback to backup even if the upgrade is successful . Reason is 
$ rm: cannot remove '/var/www/nextcloud-old/swap': Operation not permitted
Thanks for watching.
Sam

Here the entire update process with the rolling back line at the end:
Running nc-update-nextcloud
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Current   Nextcloud version 26.0.3.2
Available Nextcloud version 27.0.1
Download Nextcloud 27.0.1...
Back up current instance...
Cannot load Zend OPcache - it was already loaded
check free space...
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Maintenance mode enabled
backup database...
backup files...
backup /var/www/nextcloud-bkp_20240402_1712033837.tar.gz generated
Cannot load Zend OPcache - it was already loaded
Maintenance mode disabled
Storing backup at '/var/www/nextcloud-bkp_20240402_1712033837-26.0.3.2.tar.gz'...
Install Nextcloud 27.0.1...
Fix permissions...
Upgrade...
Cannot load Zend OPcache - it was already loaded
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: bookmarks
Disabled incompatible app: calendar
Disabled incompatible app: carnet
Disabled incompatible app: contacts
Disabled incompatible app: cookbook
Disabled incompatible app: files_bpm
Disabled incompatible app: files_mindmap
Disabled incompatible app: maps
Disabled incompatible app: memories
Disabled incompatible app: notify_push
Disabled incompatible app: passwords
Disabled incompatible app: ransomware_protection
Disabled incompatible app: recognize
Disabled incompatible app: tasks
Updating <lookup_server_connector> ...
Updated <lookup_server_connector> to 1.15.0
Updating <oauth2> ...
Updated <oauth2> to 1.15.1
Updating <password_policy> ...
Updated <password_policy> to 1.17.0
Updating <photos> ...
Updated <photos> to 2.3.0
Updating <suspicious_login> ...
Updated <suspicious_login> to 5.0.0
Updating <files> ...
Updated <files> to 1.22.0
Updating <activity> ...
Updated <activity> to 2.19.0
Updating <cloud_federation_api> ...
Updated <cloud_federation_api> to 1.10.0
Updating <dav> ...
Fix broken values of calendar objects

 Starting ...
    0/0 [>---------------------------]   0%
Clean up old calendar subscriptions from deleted users that were not cleaned-up

 Starting ...
    0/0 [->--------------------------]   0%
Updated <dav> to 1.27.0
Updating <files_external> ...
Updated <files_external> to 1.19.0
Updating <files_sharing> ...
Updated <files_sharing> to 1.19.0
Updating <files_trashbin> ...
Updated <files_trashbin> to 1.17.0
Updating <files_versions> ...
Updated <files_versions> to 1.20.0
Updating <sharebymail> ...
Updated <sharebymail> to 1.17.0
Updating <workflowengine> ...
Updated <workflowengine> to 2.9.0
Updating <comments> ...
Updated <comments> to 1.17.0
Updating <firstrunwizard> ...
Updated <firstrunwizard> to 2.16.0
Updating <notifications> ...
Updated <notifications> to 2.15.0
Updating <systemtags> ...
Updated <systemtags> to 1.17.0
Updating <theming> ...
Migrate old user accessibility config

 Starting ...
    0/0 [--->------------------------]   0%
Updated <theming> to 2.2.0
Updating <bruteforcesettings> ...
Updated <bruteforcesettings> to 2.7.0
Updating <contactsinteraction> ...
Updated <contactsinteraction> to 1.8.0
Updating <dashboard> ...
Updated <dashboard> to 7.7.0
Updating <federatedfilesharing> ...
Updated <federatedfilesharing> to 1.17.0
Updating <files_pdfviewer> ...
Updated <files_pdfviewer> to 2.8.0
Updating <files_rightclick> ...
Updated <files_rightclick> to 1.6.0
Updating <logreader> ...
Updated <logreader> to 2.12.0
Updating <privacy> ...
Updated <privacy> to 1.11.0
Updating <provisioning_api> ...
Updated <provisioning_api> to 1.17.0
Updating <recommendations> ...
Updated <recommendations> to 1.6.0
Updating <related_resources> ...
Updated <related_resources> to 1.2.0
Updating <serverinfo> ...
Updated <serverinfo> to 1.17.0
Updating <settings> ...
Updated <settings> to 1.9.0
Updating <text> ...
Updated <text> to 3.8.0
Updating <twofactor_backupcodes> ...
Updated <twofactor_backupcodes> to 1.16.0
Updating <updatenotification> ...
Updated <updatenotification> to 1.17.0
Updating <viewer> ...
Updated <viewer> to 2.1.0
Updating <weather_status> ...
Updated <weather_status> to 1.7.0
Starting code integrity check...
Finished code integrity check
Update successful
Turned off maintenance mode
Resetting log level
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Check indices of the oc_jobs table.
Check indices of the oc_direct_edit table.
Check indices of the oc_preferences table.
Check indices of the oc_mounts table.
Check indices of the oc_systemtag_object_mapping table.
Adding systag_by_tagid index to the oc_systemtag_object_mapping table, this can take some time...
oc_systemtag_object_mapping table updated successfully.
Adding additional textstep_session index to the oc_text_steps table, this can take some time...
oc_text_steps table updated successfully.
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Check columns of the comments table.
Done.
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Check primary keys.
Done.
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
All tables already up to date!
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
Update completed successfully.
Backup stored at /media/NCDATADRIVE/ncdata/data/ncp-update-backups/nextcloud-bkp_20240402_1712033837-26.0.3.2.tar.gz
Clean up...
rm: cannot remove '/var/www/nextcloud-old/swap': Operation not permitted
Abort
Clean up...
rm: cannot remove '/var/www/nextcloud-old/swap': Operation not permitted
Rolling back to backup /media/NCDATADRIVE/ncdata/data/ncp-update-backups/nextcloud-bkp_20240402_1712033837-26.0.3.2.tar.gz...
check free space...
^CDone. Press any key...

theCalcaholic, 2024-04-22:
Do you have a (manually created) swap file inside your nextcloud directory?

theCalcaholic, 2024-05-21:
closed due to inactivity

closed 2024-05-21 by theCalcaholic