nextcloud/nextcloudpi #1897: Feedback for v1.54.0 (Debian 12 bookworm + Raspberry Pi 5)

theCalcaholic, 2024-04-22:
Please give feedback here if you are able to test any of the prerelease images for v1.54.0 :)

nbendall34, 2024-04-22:
Seems good so far, just installed on my RPI5 8G model, installation was quick and painless :)

lucarez, 2024-04-23:
Just my 2 cent:
I'm trying to install nextcloudpi using the install script with CODE_DIR to tamper with php version
```
git clone --branch=v1.54.0-rc3 --depth=1 https://github.com/nextcloud/nextcloudpi.git
vim etc/ncp.cfg ## set php_version to 8.2 (default for Debian 12) ##
cd nextcloudpi
sudo CODE_DIR="$(pwd)" bash install.sh
```
getting:
```
Processing triggers for libapache2-mod-php8.2 (8.2.18-1+0~20240411.52+debian12~1.gbp91aa2d) ...
+ mkdir -p /run/mysqld
+ chown mysql /run/mysqld
+ install_template apache2/http2.conf.sh /etc/apache2/conf-available/http2.conf --defaults
+ local template=apache2/http2.conf.sh
+ local target=/etc/apache2/conf-available/http2.conf
++ mktemp
+ local bkp=/tmp/tmp.zOnWGpFvIt
+ echo 'Installing template '\''apache2/http2.conf.sh'\''...'
Installing template 'apache2/http2.conf.sh'...
++ dirname /etc/apache2/conf-available/http2.conf
+ mkdir -p /etc/apache2/conf-available
+ [[ -f /etc/apache2/conf-available/http2.conf ]]
+ [[ --defaults == \-\-\d\e\f\a\u\l\t\s ]]
+ bash /usr/local/etc/ncp-templates/apache2/http2.conf.sh --defaults
+ rm /tmp/tmp.zOnWGpFvIt
+ mkdir -p /etc/systemd/system/php8.1-fpm.service.d
+ echo '[Service]'
+ echo 'ExecStartPre=mkdir -p /var/run/php'
+ [[ systemd != \s\y\s\t\e\m\d ]]
+ systemctl daemon-reload
+ systemctl restart php8.1-fpm
Failed to restart php8.1-fpm.service: Unit php8.1-fpm.service not found.
```
in the `lamp.sh` script is _hard-coded_ `php8.1` instead of `php${PHPVER}`

Gipada, 2024-04-23:
I've been using it for the last few hours. Vanilla install is working great (managed to log in, configure everything, add an nvme with base, change data to it and use it)
Still, I did not manage to add wifi configuration from Raspberry Pi Imager or ssh connection. I had to use vanilla, find some Ethernet cable to show the initial webservice, begin init and then add Wifi, ssh and so on.

theCalcaholic, 2024-04-23:
@lucarez  That's not how you install NCP from a different branch/ref. Also, PHP 8.2 is not yet supported. Try this instead:

```sh
export BRANCH=v1.54.0-rc3
bash <(wget -O - https://raw.githubusercontent.com/nextcloud/nextcloudpi/${var.branch}/install.sh) # or the curl method
```

theCalcaholic, 2024-04-23:
@Gipada  What do you mean by "Vanilla install"?
I don't expect any special features of Raspberry Pi Imager to work, since the image is no longer based on raspbian (in favor of Armbian). You can use `armbian-config` after installation to configure WIFI.
On which board exactly are you trying to install NCP?

theCalcaholic, 2024-04-23:
@Gipada Also, regarding the installation on NVME SSD - there's a section in the same guide :)

https://docs.armbian.com/User-Guide_Getting-Started/#how-to-install-to-emmc-sata-nvme-usb

lucarez, 2024-04-24:
Hi, thanks for the fast reply and all the hard work.

> @lucarez Thank you for the test!, However, that's not quite how you install NCP from a different branch/ref and also, PHP 8.2 is not yet supported. Every new PHP version requires additional testing and work (because modules are renamed, deprecated, etc.)

I figured that that wasn't the canonical way, but since I wanted to install it with PHP 8.2 (changing `etc/ncp.cfg`) it was the only easy way I found
> 
> To install the new release, try this instead:
> 
> ```shell
> export BRANCH=v1.54.0-rc3
> bash <(wget -O - https://raw.githubusercontent.com/nextcloud/nextcloudpi/${BRANCH}/install.sh) # or the curl method
> ```

Since using this method wouldn't allow me to fix the problems I've encountered (php version hard-coded) and there was a pretty convenient `CODE_DIR` that allows me to choose the branch, edit it and run all the scripts from that folder, I didn't see iit as such a wrong option
> 
> For context: I decided to keep OS and PHP version updates separate, because I can't assume everyone to be on the latest NC version and the required PHP version is tied to the version of NC. PHP versions are therefore updated during NC updates (because only then I can safely assume a minimum NC version) and PHP 8.2 will be no different (will probably come with NC 29).

In the docs for [Nextcloud 28](https://docs.nextcloud.com/server/latest/admin_manual/release_notes/upgrade_to_28.html#upgrade-to-nextcloud-28) under [System requirements](https://docs.nextcloud.com/server/latest/admin_manual/release_notes/upgrade_to_28.html#system-requirements) I found:
 PHP 8.3 is now supported, but 8.2 is recommended.

By the way: even with the error reported above the service went up and running just fine with NC28 (Now automatically upgraded to 29), I had to fix some apache confs but nothing big

TL-Luca-Ricci, 2024-04-24:
Hi @nbendall34 
> It's been working ok so far for me, currently running a data upload to it of about 405 GB on a VPN and locally, temps reach around 50-60°C with a PWM fan connected. However I've had a bit of an issue with sudo su when I'm SSH'd in to the device. Running "sudo su" wields a "this account is not available" error, however "sudo su - /bin/bash" seems to work fine.
> Anyone else seen this yet?
Yes I've got the same problem, it's because `ncp.sh` runs:
```bash
chsh -s /usr/sbin/nologin root
```
that prevent the root from having a shell (I suppose for _security_ reason?) anyway you can put it back by running
```bash
sudo chsh -s /bin/bash root
```

seriengriller, 2024-04-26:
I gave it a try with:
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/v1.54.0-rc3/install.sh | sudo bash`

As the result it quits with 
`Installing NextCloudPi...
ERROR: distro not supported:
Debian GNU/Linux 12 \n \l

`

I've got a brand new Rasperry Pi 5 with latest 64 bit Raspberry PI OS ...
I had expected, that the release v1.54.0 is exactly for bookworm.

Would be nice to get a hint.

seriengriller, 2024-04-26:
Ok in install.sh 
 `BRANCH="${BRANCH:-master}"`

seems to be misleading. After cloning and changing this to 
BRANCH="${BRANCH:-v1.54.0-rc3}"
the script works on...

But end up with:
` systemctl start systemd-resolved
Failed to start systemd-resolved.service: Unit systemd-resolved.service not found.
+ systemctl start resolvconf
Failed to start resolvconf.service: Unit resolvconf.service not found.
`
Same here: Would be nice to get a hint.

theCalcaholic, 2024-04-26:
@seriengriller Thanks for the test. The current release is only fully working yet for the generated images. The installer and the dist-upgrade script for older instances are still being worked on. But that does mean, your feedback is indeed valuable. I'll let you know when this is presumably fixed :)

seriengriller, 2024-04-26:
Thanks for quick reply!
I tested your fix but same issue:
`Installing NextCloudPi...
ERROR: distro not supported:
Debian GNU/Linux 12 \n \l
`

seriengriller, 2024-04-26:
I tested your recommendation with the prebuild image, now.
After some boot action the prompt says 
`NextcloudPi v1.53.3 is up to date`
Im wondering that this is not v1.54.0 as i downloaded the Image named like this.
The image does not work on an Raspi 5.
The last error message says:
`This account is currently not available`. 
I guess I come here in some days back to look for some further testings.
Thanks for your work.

vovchykbratyk, 2024-04-26:
I ran into the same exact issue as @seriengriller

RPI5 8gb

theCalcaholic, 2024-04-26:
@seriengriller @vovchykbratyk Did you try to activate via web browser? You are not supposed to login from the terminal

theCalcaholic, 2024-04-26:
@seriengriller The error message you describe ("distro not supported") means you were testing the wrong branch. Are you sure that you tested with the BRANCH variable set correctly?

seriengriller, 2024-04-27:
@theCalcaholic Im absolut sure that I used the BRANCH variable as described.
I do not understand how this can ever work, when BRANCH is overwritten in install.sh to "-master". But I am not a bash professionel.
I had not tried to activate via Webbrowser. But now I have:
The webserver is up. I can do initialising and login via webinterface.
I activated ssh through webbrowser, but connection gets refused ==>So I can not login through ssh, and I can not login through terminal directly. Might be another lack of knowledge here... but further configuration seems to be impossible for me.

theCalcaholic, 2024-04-27:
`BRANCH="${BRANCH:-master}"` is bash for "set BRANCH to master unless it's set to something else already" (so a default value).

The way to connect to NCP via terminal is by activating SSH for a user on the webinterface, as you did (I'll include a notification in the terminal login screen for the next images, as many seem to be confused by this). However, I don't know why you still can't connect after setting up a new user there and would need more information to troubleshoot that. :thinking: 

However, I'm currently generating a new release with a few fixes, maybe it makes more sense to just wait for that and see if you have more luck with it (will likely land today after I made some more manual tests)

vovchykbratyk, 2024-04-27:
> @seriengriller @vovchykbratyk Did you try to activate NCP via web browser? You are not supposed to login from the terminal.
> 
> The autologin is already disabled on the feature branch, but I didn't publish any images with the fix yet.

Hi @theCalcaholic !  I’m not sure what you mean.  The error I encountered was upon boot, not login.  I flashed the image, put it in the RPI5, powered it on and received the error.  Let me know if there is documentation for another way to perform the test.  Thanks!

seriengriller, 2024-04-27:
Thanks for the explanation and your patience!
I tested a little on the "export issue".
It seems to me that the behaviour in the latest Raspberry PI Os (where I tried the install.sh) is not as expected.
I added some `echo $BRANCH` and `export -p` around in my local copy of install.sh.
The script inherits not the BRANCH variable from environment. Strange but true. So the first tries where like you mentioned from the "wrong" master branch.

I will have another look later on.

@vovchykbratyk Plug in the network cable, boot the raspi, open browser on another device and open https://nextcloudpi.local or IP adress of your raspi. Follow the web wizard to complete installation. It is not possible to login in via terminal directly.

theCalcaholic, 2024-04-27:
> Hi @theCalcaholic ! I’m not sure what you mean. The error I encountered was upon boot, not login. I flashed the image, put it in the RPI5, powered it on and received the error. Let me know if there [...]

Yes, in the test release, the armbian autologin wasn't disabled, which means, when you boot the image, it will automatically try to login as root (which fails, because root login is disabled). You should, however, still be able to reach the instance via its IP address or https://nextcloudpi.local in a webbrowser and proceed with the activation.

vlRein, 2024-04-28:
Hello, I'm new on this.
I just flashed NextcloudPi_RaspberryPi5_v1.54.0-rc3.zip and cannot log because root account logs automatically.
And root accound does not exist.
How can I fix this ?

Thanks

maxlin1, 2024-04-28:
Dosn't work on Rasbian latest Version:

$ git clone https://github.com/nextcloud/nextcloudpi.git
cd nextcloudpi
./build/build-SD-rpi.sh
Cloning into 'nextcloudpi'...
remote: Enumerating objects: 15861, done.
remote: Counting objects: 100% (1935/1935), done.
remote: Compressing objects: 100% (709/709), done.
remote: Total 15861 (delta 1302), reused 1743 (delta 1199), pack-reused 13926
Receiving objects: 100% (15861/15861), 8.88 MiB | 1.80 MiB/s, done.
Resolving deltas: 100% (11094/11094), done.

[ Build NCP Raspberry Pi ]

[ Download RaspiOS ]
2024-04-28 13:25:29 URL:https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-26/2022-09-22-raspios-bullseye-arm64-lite.img.xz [302978228/302978228] -> "cache/raspios_lite.xz" [1]
'cache/raspios_lite.img' -> 'tmp/NextcloudPi_RPi_04-28-24.img'

[ Resize Image ]
WARNING: You are not superuser.  Watch out for permissions.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
RaspiOS image mounted
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/loop0 is mounted on /home/Max/Downloads/nextcloudpi/raspbian_root; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop0 is now 982016 (4k) blocks long.

Image resized
RaspiOS image umounted

[ Update Raspbian Boot UUIDS ]
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
RaspiOS image mounted
RaspiOS image umounted
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
RaspiOS image mounted
RaspiOS image umounted
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
RaspiOS image mounted
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
cp: cannot stat '/usr/bin/qemu-aarch64-static': No such file or directory
RaspiOS image umounted

theCalcaholic, 2024-04-28:
@maxlin1 That's not supported. Please use the brebuilt image from the release page.

If you really want to build your own image, use `./build/build-armbian.sh rpi4b RaspberryPi4` or `./build/build-armbian.sh rpi5b RaspberryPi5` respectively. The new RPi images are not based on Raspbian anymore but use Armbian instead.

theCalcaholic, 2024-04-28:
Thanks to everyone for their feedback. I am now very close to finishing the release and will close this issue.

closed 2024-04-28 by theCalcaholic