nextcloud/nextcloudpi #1907: Adding systemd-resolved breaks DNS during install with LXC / tteck script

namomitk, 2024-05-01:
https://github.com/nextcloud/nextcloudpi/blob/c4288d8e042f861ebfe41dcc01ce1bd545e1fd83/install.sh#L77-L81

See Also: https://github.com/tteck/Proxmox/issues/2893

Name resolution stops working due to systemd-resolved resetting the DNS server provided by Proxmox.

```
root@nextcloudpi:~# nslookup google.com
Server:         127.0.0.53
Address:        127.0.0.53#53

** server can't find google.com: SERVFAIL
```

```
Thanks for using MariaDB!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
 /Suggested packages:
  polkitd | policykit-1
Recommended packages:
  libnss-myhostname libnss-resolve
The following NEW packages will be installed:
  systemd-resolved
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 305 kB of archives.
After this operation, 827 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main amd64 systemd-resolved amd64 252.22-1~deb12u1 [305 kB]
Fetched 305 kB in 0s (5,355 kB/s)    
 \Selecting previously unselected package systemd-resolved.
(Reading database ... 24094 files and directories currently installed.)
Preparing to unpack .../systemd-resolved_252.22-1~deb12u1_amd64.deb ...
Unpacking systemd-resolved (252.22-1~deb12u1) ...
 |Setting up systemd-resolved (252.22-1~deb12u1) ...
Converting /etc/resolv.conf to a symlink to /run/systemd/resolve/stub-resolv.conf...
Creating group 'systemd-resolve' with GID 996.
Creating user 'systemd-resolve' (systemd Resolver) with UID 996 and GID 996.
 /Created symlink /etc/systemd/system/dbus-org.freedesktop.resolve1.service → /lib/systemd/system/systemd-resolved.service.
Created symlink /etc/systemd/system/sysinit.target.wants/systemd-resolved.service → /lib/systemd/system/systemd-resolved.service.
 \Processing triggers for dbus (1.14.10-1~deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
 |Installing nc-nextcloud
Ign:1 https://packages.sury.org/php bookworm InRelease
Ign:2 http://deb.debian.org/debian bookworm InRelease
Ign:3 http://security.debian.org bookworm-security InRelease
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Ign:1 https://packages.sury.org/php bookworm InRelease
Ign:2 http://deb.debian.org/debian bookworm InRelease                
Ign:3 http://security.debian.org bookworm-security InRelease         
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Ign:2 http://deb.debian.org/debian bookworm InRelease
Ign:1 https://packages.sury.org/php bookworm InRelease
Ign:3 http://security.debian.org bookworm-security InRelease
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Err:2 http://deb.debian.org/debian bookworm InRelease
  Temporary failure resolving 'deb.debian.org'
Err:3 http://security.debian.org bookworm-security InRelease            
  Temporary failure resolving 'security.debian.org'
Err:1 https://packages.sury.org/php bookworm InRelease                  
  Temporary failure resolving 'packages.sury.org'
Err:4 http://deb.debian.org/debian bookworm-updates InRelease
  Temporary failure resolving 'deb.debian.org'
Reading package lists... Done    
W: Failed to fetch http://deb.debian.org/debian/dists/bookworm/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch http://deb.debian.org/debian/dists/bookworm-updates/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch http://security.debian.org/dists/bookworm-security/InRelease  Temporary failure resolving 'security.debian.org'
W: Failed to fetch https://packages.sury.org/php/dists/bookworm/InRelease  Temporary failure resolving 'packages.sury.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
[... 108 lines cut ...]
Err:15 http://deb.debian.org/debian bookworm/main amd64 libsmbclient amd64 2:4.17.12+dfsg-0+deb12u1
  Temporary failure resolving 'deb.debian.org'
Err:16 http://deb.debian.org/debian bookworm/main amd64 samba-common all 2:4.17.12+dfsg-0+deb12u1
  Temporary failure resolving 'deb.debian.org'
Err:17 http://deb.debian.org/debian bookworm/main amd64 smbclient amd64 2:4.17.12+dfsg-0+deb12u1
  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/f/fuse/libfuse2_2.9.9-6%2bb1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/f/fuse/fuse_2.9.9-6%2bb1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/f/fuse-exfat/exfat-fuse_1.3.0%2bgit20220115-2_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/e/exfatprogs/exfatprogs_1.2.0-1%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/liba/libarchive/libarchive13_3.6.2-1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/a/avahi/libavahi-common-data_0.8-10_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/a/avahi/libavahi-common3_0.8-10_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/a/avahi/libavahi-client3_0.8-10_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/t/talloc/libtalloc2_2.4.0-f2_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/t/tdb/libtdb1_1.4.8-2_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://deb.debian.org/debian/pool/main/t/tevent/libtevent0_0.14.1-1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/libldb2_2.6.2%2bsamba4.17.12%2bdfsg-0%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/libwbclient0_4.17.12%2bdfsg-0%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/samba-libs_4.17.12%2bdfsg-0%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/libsmbclient_4.17.12%2bdfsg-0%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/samba-common_4.17.12%2bdfsg-0%2bdeb12u1_all.deb  Temporary failure resolving 'deb.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/s/samba/smbclient_4.17.12%2bdfsg-0%2bdeb12u1_amd64.deb  Temporary failure resolving 'deb.debian.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

theCalcaholic, 2024-05-01:
Hm, weird, I had the opposite happening in my Proxmox/LXC setup :thinking: 

Did you install using the official NCP-LXC-Image or via the Proxmox-Script?

namomitk, 2024-05-01:
Install was with the Proxmox-Script.

Resolv.conf is changed from being the DNS server set in Proxmox / the LXC to:

```
nameserver 127.0.0.53
options edns0 trust-ad
search .
```

theCalcaholic, 2024-05-02:
@namomitk I did some more testing and believe that the issue can best be fixed in the install script, not NCP itself (I'll discuss this with ttek in more detail in the upstream post tomorrow).

For now, you can still install NCP on Proxomox by heading to any storage pool that has CT Templates enabled, go to CT Templates, click "Download from URL" and enter the link to the latest LXC image from the [Nextcloudpi releases page](https://github.com/nextcloud/nextcloudpi/releases).
Then create a container from that template.

[image]

In my test, I had to set the network device to "static IP" - DHCP didn't seem to work (but I'm don't think that that's an issue with the container image - I'll need to look into that). But other than that everything worked fine.

[image]

closed 2024-05-02 by theCalcaholic

namomitk, 2024-05-02:
Thanks for looking into it. Always rough building from master. I understand why you couldn’t change the commits. Thanks for the how to.