nextcloud/nextcloudpi #1909: Update to 1.54 with dist-upgrade killed the system

N0m4n904, 2024-05-02:
Hi, I updated my nextcloudpi to 1.54. After that I saw the message that there is a dist-upgrade, so I ran it. I did make nextcloud backups before running it.

the command ran and at some point it started to uninstall stuff and then crashed the system.

After I restarted my Raspberry Pi, I got the message that the dist-upgrade is available, but when I run it again, it says that there is no update for the current version.

I can't access my nextcloud anymore and when I try to run nc-update-nextcloud I get the message "Nextcloud is currently down".

How am I able to rescue my system?

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.0
NextcloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.2T/1.9T
rootfs usage         5.8G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               26min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(377): Doctrine\DBAL\Query\QueryBuilder->executeQuery()
#5 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(280): Doctrine\DBAL\Query\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(421): OC\DB\QueryBuilder\QueryBuilder->execute()
#7 /var/www/nextcloud/lib/private/AppConfig.php(187): OC\AppConfig->loadConfigValues()
[... 63 lines cut ...]
```

{"reqId":"Z5r6gHXItxvJ2vH0jLzU","level":3,"time":"2024-05-02T07:05:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Exception thrown: Doctrine\\DBAL\\Exception","userAgent":"--","version":"28.0.4.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1654,"function":"connect","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1081,"function":"getWrappedConnection","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":261,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":337,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":377,"function":"executeQuery","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":280,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":421,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":187,"function":"loadConfigValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":377,"function":"getApps","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":128,"function":"getValues","class":"OC\\AppConfig","type":"->"},{"file":"/var/www/nextcloud/lib/private/App/AppManager.php","line":149,"function":"getInstalledAppsValues","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":231,"function":"getInstalledApps","class":"OC\\App\\AppManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":90,"function":"getEnabledApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":700,"function":"runInitialRegistration","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}
{"reqId":"ZjM7H3FTaDd3xaNBFKgNngAAAAU","level":3,"time":"2024-05-02T07:05:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}
{"reqId":"ZjM7W5u67K5phIXujcLVtwAAAAA","level":3,"time":"2024-05-02T07:06:04+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}
{"reqId":"ZjM7l0GxmgPt0wdBpYTHcQAAAAE","level":3,"time":"2024-05-02T07:07:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}
{"reqId":"ZjM70zzLa9DtzL_6F8UJuwAAAAI","level":3,"time":"2024-05-02T07:08:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}
{"reqId":"ZjM8D8YLFq3Z_j6eY4dCgQAAAAM","level":3,"time":"2024-05-02T07:09:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data": [... cut ...]
{"reqId":"ZjM-ozzLa9DtzL_6F8UJvQAAAAI","level":3,"time":"2024-05-02T07:20:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}

```
</details>

theCalcaholic, 2024-05-02:
Did you, at any point, manually install a php version (because those packages look like it). If so, that's unfortunately likely to happen, because manually managing PHP versions breaks NCPs automation...

Regarding getting back you're instance - according to the logs, your database is down, but there are no database errors. Can you try restarting mariadb and see if that works (run `sudo systemctl restart mariadb && sudo systemctl status mariadb`)?

N0m4n904, 2024-05-02:
> Did you, at any point, manually install a php version (because those packages look like it). If so, that's unfortunately likely to happen, because manually managing PHP versions breaks NCPs [...]

I needed to install php manually, because it got uninstalled during the upgrade process.
during the uninstall process I got a kernel panic, so php got completely uninstalled.
I restarted mariadb and status says it is running.

I created a new report now.

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.0
NextcloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.2T/1.9T
rootfs usage         5.9G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql                     Nextcloud check      error
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               24min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
OCP\HintException: [0]: Memcache OC\Memcache\Redis not available for local cache (Is the matching PHP module installed and enabled?)

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu May 02 10:20:38.368918 2024] [ssl:error] [pid 34002] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu May 02 10:20:38.410053 2024] [ssl:warn] [pid 34003] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 3.0.13 30 Jan 2024 (OpenSSL 3.0.11 19 Sep 2023), version currently loaded is 0x300000B0) - may result in undefined or erroneous behavior
[Thu May 02 10:20:38.418423 2024] [ssl:warn] [pid 34003] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu May 02 10:20:38.419905 2024] [ssl:error] [pid 34003] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: 0112307086B5A933D74576F54C8FD74C8207A8AF / notbefore: Nov 27 22:55:31 2020 GMT / notafter: Nov 25 22:55:31 2030 GMT]
[Thu May 02 10:20:38.419976 2024] [ssl:error] [pid 34003] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu May 02 10:20:38.426372 2024] [http2:warn] [pid 34003] AH10034: The mpm module (prefork.c) is not supported by mod_http2. The mpm determines how things are processed in your server. HTTP/2 has more demands in this regard and the currently selected mpm will just not do. This is an advisory warning. Your server will continue to work, but the HTTP/2 protocol will be inactive.
[Thu May 02 10:20:38.565170 2024] [mpm_prefork:notice] [pid 34003] AH00163: Apache/2.4.59 (Debian) OpenSSL/3.0.11 configured -- resuming normal operations
[... 38 lines cut ...]
```

{"reqId":"ZjNDF8YLFq3Z_j6eY4dChgAAAAM","level":3,"time":"2024-05-02T07:39:03+00:00","remoteAddr":"2a02:810d:5fbf:d41c:df3d:3e14:7516:9d11","user":"--","app":"PHP","method":"GET","url":"/ocs/v2.php/apps/serverinfo/api/v1/info?format=json&skipUpdate=false&skipApps=false","message":"Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver at /var/www/nextcloud/lib/private/DB/Connection.php#139","userAgent":"python-requests/2.31.0","version":"28.0.4.1","data":{"app":"PHP"}}
{"reqId":"b42asIMZTELE9GDvvOSC","level":3,"time":"2024-05-02T07:40:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Exception thrown: Doctrine\\DBAL\\Exception","userAgent":"--","version":"28.0.4.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1654,"function":"connect","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1081,"function":"getWrappedConnection","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":261,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":337,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":377,"function":"executeQuery","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":280,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":421,"function":"execute","class":"OC\\DB\\QueryBuilder\ [... cut ...] nextcloud/lib/base.php","line":700,"function":"runInitialRegistration","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}

```
</details>

theCalcaholic, 2024-05-02:
> Is there a way to completely reinstall all needed packages?

The dist upgrade doesn't actually uninstall anything important, it only upgrades things and removes unneeded dependencies afterwards.

Does `sudo apt update` give you any warnings?

N0m4n904, 2024-05-02:
> > Is there a way to completely reinstall all needed packages?
> 
> The dist upgrade doesn't actually uninstall anything important, it only upgrades things and removes unneeded dependencies afterwards.
> 
> Does `sudo apt update` give you any warnings?

weird, why got php removed? I needed to install it again, because the command php was not known anymore.

I ran apt update know and get this output:

[image]

theCalcaholic, 2024-05-02:
Okay, unfortunately none of that gives me any hint... :thinking: 

Can you share your /var/log/apt/history.log?

> weird, why got php removed
The only explanation I can think of is that something broke during `apt upgrade` or `apt full-upgrade` and therefore packages would be removed but their new versions would not yet be installed.
However, I would expect that apt complains about this and offers to complete the upgrade, so I'm not entirely show how you ended up there. Hopefully /var/log/apt/history.log will give more clues.

Also, the next person to run into this issue should find their full dist-upgrade log on disk (just pushed that change)

theCalcaholic, 2024-05-02:
Also, can you please share the kind of system your running NCP on (what installation method did you use and what hardware are you running)?

theCalcaholic, 2024-05-02:
You can try to run the following to reinstall php - but no guarantees that it will work:

```sh
sudo apt-get install --no-install-recommends -t bookworm php8.1 php8.1-curl php8.1-gd php8.1-fpm php8.1-cli php8.1-opcache php8.1-mbstring php8.1-xml php8.1-zip php8.1-fileinfo php8.1-ldap php8.1-intl php8.1-bz2
```

N0m4n904, 2024-05-02:
> Okay, unfortunately none of that gives me any hint... 🤔
> 
> Can you share your /var/log/apt/history.log?
> 
> > weird, why got php removed
> > The only explanation I can think of is that something broke during `apt upgrade` or `apt full-upgrade` and therefore packages would be removed but their new versions would not yet be installed.
> > However, I would expect that apt complains about this and offers to complete the upgrade, so I'm not entirely show how you ended up there. Hopefully /var/log/apt/history.log will give more clues.
> 
> Also, the next person to run into this issue should find their full dist-upgrade log on disk (just pushed that change)

here is the history.log
[history.txt](https://github.com/nextcloud/nextcloudpi/files/15186794/history.txt)

I am running on a Raspberry Pi 4B 8GB Model. And I installed it via flashing the sd card. From there I always upgraded using the ncp scripts.

When I try to install php again, I get this output

[php.txt](https://github.com/nextcloud/nextcloudpi/files/15186902/php.txt)

theCalcaholic, 2024-05-02:
Hm, that's interesting. Can you run `sudo cat /etc/apt/sources.list.d/*` and share the output?

N0m4n904, 2024-05-02:
> Hm, that's interesting. Can you run `sudo cat /etc/apt/sources.list.d/*` and share the output?

Here is the output:

```
sudo cat /etc/apt/sources.list.d/*
deb [arch=arm64 signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/debian bullseye stable
#deb https://packages.sury.org/php/ bookworm main
deb http://archive.raspberrypi.org/debian/ bullseye main
Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bullseye main
Tailscale packages for raspbian bullseye
deb https://pkgs.tailscale.com/stable/raspbian bullseye main
```

theCalcaholic, 2024-05-02:
Okay, I see the issue now. Can you 
1) give me the output of `sudo ls -l /etc/apt/sources.list.d`
2) update all files in this directory by replacing 'bullseye' with 'bookworm' and run the following commands:

```sh
apt-get update && apt-get upgrade -y --without-new-pkgs
apt-get full-upgrade -y
sudo apt-get --purge  autoremove -y
```

N0m4n904, 2024-05-02:
> Okay, I see the issue now. Can you
> 
> 1. give me the output of `sudo ls -l /etc/apt/sources.list.d`
> 2. update all files in this directory by replacing 'bullseye' with 'bookworm' and run the following commands:
> 
> ```shell
> apt-get update && apt-get upgrade -y --without-new-pkgs
> apt-get full-upgrade -y
> sudo apt-get --purge  autoremove -y
> ```

Output:
```
sudo ls -l /etc/apt/sources.list.d
total 16
-rw-r--r-- 1 root root 131 May 20  2022 docker.list
-rw-r--r-- 1 root root  50 May  2 07:56 php.list
-rw-r--r-- 1 root root 191 May 20  2022 raspi.list
-rw-r--r-- 1 root root 104 May 20  2022 tailscale.list
```

Replaced bullseye with bookworm and running the commands now

N0m4n904, 2024-05-02:
okay, the command ran without any issue

N0m4n904, 2024-05-02:
> Awesome! Then you should just need the following 3 commands to complete the upgrade and then, hopefully, everything should be back to normal:
> 
> ```shell
> sudo ncc maintenance:mode --off
> sudo sed -i -e 's/bullseye/bookworm/' /usr/local/etc/ncp.cfg
> sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade
> ```

[image]

theCalcaholic, 2024-05-02:
I think you might need to repeat the PHP installation step https://github.com/nextcloud/nextcloudpi/issues/1909#issuecomment-2090098981

N0m4n904, 2024-05-02:
I still get this output:

[php.txt](https://github.com/nextcloud/nextcloudpi/files/15187323/php.txt)

theCalcaholic, 2024-05-02:
@N0m4n904  See the update to my previous comment :)

N0m4n904, 2024-05-02:
> @N0m4n904 See the update to my previous comment :)

same with the redis package. `E: Package 'php8.1-redis' has no release candidate`

theCalcaholic, 2024-05-02:
Did you update /etc/apt/sources.list.d/php.list and changed 'bullseye' to 'bookworm'?

N0m4n904, 2024-05-02:
[image]

theCalcaholic, 2024-05-02:
You need to remove the `#` (no idea how it got there).

N0m4n904, 2024-05-02:
now my ncp-report is like this:

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.0
NextcloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.2T/1.9T
rootfs usage         6.2G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
OCP\HintException: [0]: Memcache OC\Memcache\Redis not available for local cache (Is the matching PHP module installed and enabled?)

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu May 02 13:24:50.479529 2024] [mpm_prefork:notice] [pid 1584] AH00171: Graceful restart requested, doing restart
[Thu May 02 13:24:50.612331 2024] [ssl:warn] [pid 1584] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 3.0.13 30 Jan 2024 (OpenSSL 3.0.11 19 Sep 2023), version currently loaded is 0x300000B0) - may result in undefined or erroneous behavior
[Thu May 02 13:24:50.617436 2024] [ssl:warn] [pid 1584] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu May 02 13:24:50.618770 2024] [ssl:error] [pid 1584] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: 0112307086B5A933D74576F54C8FD74C8207A8AF / notbefore: Nov 27 22:55:31 2020 GMT / notafter: Nov 25 22:55:31 2030 GMT]
[Thu May 02 13:24:50.618799 2024] [ssl:error] [pid 1584] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Thu May 02 13:24:50.625363 2024] [http2:warn] [pid 1584] AH10034: The mpm module (prefork.c) is not supported by mod_http2. The mpm determines how things are processed in your server. HTTP/2 has more demands in this regard and the currently selected mpm will just not do. This is an advisory warning. Your server will continue to work, but the HTTP/2 protocol will be inactive.
[Thu May 02 13:24:50.713667 2024] [mpm_prefork:notice] [pid 1584] AH00163: Apache/2.4.59 (Debian) OpenSSL/3.0.11 configured -- resuming normal operations
[Thu May 02 13:24:50.713724 2024] [core:notice] [pid 1584] AH00094: Command line: '/usr/sbin/apache2'
[Thu May 02 13:29:36.214817 2024] [mpm_prefork:notice] [pid 1584] AH00170: caught SIGWINCH, shutting down gracefully
[Thu May 02 13:29:58.411392 2024] [ssl:warn] [pid 1646] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 3.0.13 30 Jan 2024 (OpenSSL 3.0.11 19 Sep 2023), version currently loaded is 0x300000B0) - may result in undefined or erroneous behavior
[Thu May 02 13:29:58.449235 2024] [ssl:warn] [pid 1646] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu May 02 13:29:58.451175 2024] [ssl:error] [pid 1646] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: 0112307086B5A933D74576F54C8FD74C8207A8AF / notbefore: Nov 27 22:55:31 2020 GMT / notafter: Nov 25 22:55:31 2030 GMT]
[... 33 lines cut ...]
```

{"reqId":"ZjN4Q4TgaJSBdwA-OE3lvQAAAAU","level":3,"time":"2024-05-02T11:25:55+00:00","remoteAddr":"176.2.75.242","user":"--","app":"index","method":"GET","url":"/favicon.ico","message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","userAgent":"Mozilla/5.0 (Linux; Android 10; K) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/124.0.0.0 Mobile Safari/537.36","version":"28.0.4.1","exception":{"Exception":"Doctrine\\DBAL\\Exception","Message":"Failed to connect to the database: An exception occurred in the driver: could not find driver","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1654,"function":"connect","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php","line":1081,"function":"getWrappedConnection","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":261,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":341,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->"},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php","line":377,"function":"executeQuery","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->"},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":280,"function":"execute","class":"Doctrine\\DBAL\\Query\\Query [... cut ...] /Template/Base.php","line":132,"function":"fetchPage","class":"OC_Template","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Template.php","line":291,"function":"printPage","class":"OC\\Template\\Base","type":"->"},{"file":"/var/www/nextcloud/index.php","line":104,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"}],"File":"/var/www/nextcloud/lib/private/DB/Connection.php","Line":139,"CustomMessage":"Exception thrown: Doctrine\\DBAL\\Exception"}}

```
</details>

N0m4n904, 2024-05-02:
when I ssh into my raspi now, I still get this message:

[image]

theCalcaholic, 2024-05-02:
Then I assume you still need to complete these steps: https://github.com/nextcloud/nextcloudpi/issues/1909#issuecomment-2090231862

N0m4n904, 2024-05-02:
i ran all three commands, but I can do it again.

theCalcaholic, 2024-05-02:
In particular the `rm -f ...` command should remove that message

N0m4n904, 2024-05-02:
I now get this message again: https://github.com/nextcloud/nextcloudpi/issues/1909#issuecomment-2090234354

theCalcaholic, 2024-05-02:
Is php8.1-redis installed?

N0m4n904, 2024-05-02:
> Is php8.1-redis installed?

the installation completed, lemme run it again

theCalcaholic, 2024-05-02:
You could try all of the above php packages again (I added the missing ones), for good measure

N0m4n904, 2024-05-02:
> You could try all of the above php packages again (I added the missing ones), for good measure

will try that

N0m4n904, 2024-05-02:
This message did not appear before. Should I do it?

[image]

theCalcaholic, 2024-05-02:
@N0m4n904 It won't hurt, although it's probably already enabled

N0m4n904, 2024-05-02:
still this error: https://github.com/nextcloud/nextcloudpi/issues/1909#issuecomment-2090234354

theCalcaholic, 2024-05-02:
Hm... Can you try if a reboot fixes things?

theCalcaholic, 2024-05-02:
Hm... I'm wondering if it is still broken, because you installed the system php packages - can you try removing those again?

N0m4n904, 2024-05-02:
the php8.1 packages from above?

theCalcaholic, 2024-05-02:
> the php8.1 packages from above?

No, the php (without version) packages

theCalcaholic, 2024-05-02:
Those:

>I reinstalled the following packages, but it did not make any difference:

> 
>     libapache2-mod-php
>     php-bz2
>     php-gd
>     php-mysql
>     php-curl
>     php-mbstring
>     php-imagick
>     php-zip
>     php-ctype
>     php-curl
>     php-dom
>     php-json
>     php-posix
>     php-bcmath
>     php-xml
>     php-intl
>     php-gmp
>     zip
>     unzip
>     wget

theCalcaholic, 2024-05-02:
But only the ones starting with php

N0m4n904, 2024-05-02:
okay, now I get this error
[image]

N0m4n904, 2024-05-02:
> Then you are probably still missing some of the php packages above. Did you copy and paste the command again?

yes did copy paste

theCalcaholic, 2024-05-02:
Okay, maybe just try the installation again - maybe some of them were depending on the non-versioned packages and got removed.

https://github.com/nextcloud/nextcloudpi/issues/1909#issuecomment-2090098981

N0m4n904, 2024-05-02:
[image]

theCalcaholic, 2024-05-02:
Hm.. Not sure why that is... Maybe apache is still using the wrong php version? What's the output of `sudo apachectl -M`?

theCalcaholic, 2024-05-02:
sorry, wrong command

theCalcaholic, 2024-05-02:
I think, what I need is: `ls -l /etc/apache2/conf-enabled/`

N0m4n904, 2024-05-02:
[image]

N0m4n904, 2024-05-02:
hmmm, too bad. I don't want to do a clean install. especially because I have some data on that drive

theCalcaholic, 2024-05-02:
One more thing to check: Can you give me the output of

```sh
php --version
php -m
```

N0m4n904, 2024-05-02:
I uninstalled PHP completely and installed 8.1 again. maintenance command is now working

N0m4n904, 2024-05-02:
> ```shell
> sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade
> ```

sudo dpkg -l | grep php
rc  libapache2-mod-php8.2               8.2.18-1~deb12u1                             arm64        server-side, HTML-embedded scripting language (Apache 2 module)       rc  libapache2-mod-php8.3               8.3.6-1+0~20240424.32+debian12~1.gbp9a7ce5   arm64        server-side, HTML-embedded scripting language (Apache 2 module)       ii  php-common                          2:94+0~20240205.51+debian12~1.gbp6faa2e      all          Common files for PHP packages                                         rc  php7.3-common                       7.3.31-1~deb10u1                             arm64        documentation, examples and common module for PHP                     rc  php7.3-readline                     7.3.31-1~deb10u1                             arm64        readline module for PHP
ii  php8.1                              8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  all          server-side, HTML-embedded scripting language (metapackage)           ii  php8.1-bcmath                       8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        Bcmath module for PHP                                                 ii  php8.1-bz2                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        bzip2 module for PHP
ii  php8.1-cli                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        command-line interpreter for the PHP scripting language               ii  php8.1-common                       8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        documentation, examples and common module for PHP
ii  php8.1-curl                         8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        CURL module for PHP                                                   ii  php8.1-fpm                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        server-side, HTML-embedded scripting language (FPM-CGI binary)
ii  php8.1-gd                           8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        GD module for PHP                                                     ii  php8.1-gmp                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        GMP module for PHP                                                    ii  php8.1-igbinary                     3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8  arm64        igbinary PHP serializer                                               rc  php8.1-imagick                      3.7.0-4+0~20231125.43+debian11~1.gbpbf7e27   arm64        Provides a wrapper to the ImageMagick library
ii  php8.1-intl                         8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        Internationalisation module for PHP
ii  php8.1-ldap                         8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        LDAP module for PHP                                                   ii  php8.1-mbstring                     8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        MBSTRING module for PHP                                               ii  php8.1-mysql                        8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        MySQL module for PHP
ii  php8.1-opcache                      8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        Zend OpCache module for PHP                                           ii  php8.1-readline                     8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        readline module for PHP                                               ii  php8.1-redis                        6.0.2-1+0~20231128.56+debian12~1.gbp9f3071   arm64        PHP extension for interfacing with Redis                              ii  php8.1-xml                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        DOM, SimpleXML, XML, and XSL module for PHP                           ii  php8.1-zip                          8.1.28-1+0~20240424.64+debian12~1.gbpd7b516  arm64        Zip module for PHP
rc  php8.2-bcmath                       8.2.18-1~deb12u1                             arm64        Bcmath module for PHP                                                 rc  php8.2-bz2                          8.2.18-1~deb12u1                             arm64        bzip2 module for PHP
rc  php8.2-cli                          8.2.18-1~deb12u1                             arm64        command-line interpreter for the PHP scripting language
rc  php8.2-common                       8.2.18-1~deb12u1                             arm64        documentation, examples and common module for PHP
rc  php8.2-curl                         8.2.18-1~deb12u1                             arm64        CURL module for PHP                                                   rc  php8.2-gd                           8.2.18-1~deb12u1                             arm64        GD module for PHP                                                     rc  php8.2-gmp                          8.2.18-1~deb12u1                             arm64        GMP module for PHP                                                    rc  php8.2-imagick                      3.7.0-4                                      arm64        Provides a wrapper to the ImageMagick library
rc  php8.2-intl                         8.2.18-1~deb12u1                             arm64        Internationalisation module for PHP
[... cut ...]
rc  php8.3-opcache                      8.3.6-1+0~20240424.32+debian12~1.gbp9a7ce5   arm64        Zend OpCache module for PHP                                           rc  php8.3-readline                     8.3.6-1+0~20240424.32+debian12~1.gbp9a7ce5   arm64        readline module for PHP                                               rc  php8.3-xml                          8.3.6-1+0~20240424.32+debian12~1.gbp9a7ce5   arm64        DOM, SimpleXML, XML, and XSL module for PHP                           rc  php8.3-zip                          8.3.6-1+0~20240424.32+debian12~1.gbp9a7ce5   arm64        Zip module for PHP

N0m4n904, 2024-05-02:
When I try to access my cloud now I get this:

```
<?php
/**
 * @copyright Copyright (c) 2016, ownCloud, Inc.
 *
 * @author Christoph Wurst <christoph@winzerhof-wurst.at>
 * @author Joas Schilling <coding@schilljs.com>
 * @author Jörn Friedrich Dreyer <jfd@butonic.de>
 * @author Lukas Reschke <lukas@statuscode.ch>
 * @author Morris Jobke <hey@morrisjobke.de>
 * @author Robin Appelman <robin@icewind.nl>
 * @author Roeland Jago Douma <roeland@famdouma.nl>
 * @author Sergio Bertolín <sbertolin@solidgear.es>
 * @author Thomas Müller <thomas.mueller@tmit.eu>
 * @author Vincent Petry <vincent@nextcloud.com>
 *
 * @license AGPL-3.0
 *
 * This code is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Affero General Public License, version 3,
 * as published by the Free Software Foundation.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU Affero General Public License for more details.
 *
 * You should have received a copy of the GNU Affero General Public License, version 3,
 * along with this program. If not, see <http://www.gnu.org/licenses/>
 *
 */
require_once __DIR__ . '/lib/versioncheck.php';

use OCP\Security\Bruteforce\MaxDelayReached;
use Psr\Log\LoggerInterface;

try {
	require_once __DIR__ . '/lib/base.php';

	OC::handleRequest();
} catch (\OC\ServiceUnavailableException $ex) {
	\OC::$server->get(LoggerInterface::class)->error($ex->getMessage(), [
		'app' => 'index',
		'exception' => $ex,
	]);

	//show the user a detailed error page
	OC_Template::printExceptionErrorPage($ex, 503);
} catch (\OCP\HintException $ex) {
	try {
		OC_Template::printErrorPage($ex->getMessage(), $ex->getHint(), 503);
	} catch (Exception $ex2) {
		try {
			\OC::$server->get(LoggerInterface::class)->error($ex->getMessage(), [
				'app' => 'index',
				'exception' => $ex,
			]);
			\OC::$server->get(LoggerInterface::class)->error($ex2->getMessage(), [
				'app' => 'index',
				'exception' => $ex2,
			]);
		} catch (Throwable $e) {
			// no way to log it properly - but to avoid a white page of death we try harder and ignore this one here
		}

		//show the user a detailed error page
		OC_Template::printExceptionErrorPage($ex, 500);
	}
} catch (\OC\User\LoginException $ex) {
	$request = \OC::$server->getRequest();
	/**
	 * Routes with the @CORS annotation and other API endpoints should
	 * not return a webpage, so we only print the error page when html is accepted,
	 * otherwise we reply with a JSON array like the SecurityMiddleware would do.
	 */
	if (stripos($request->getHeader('Accept'), 'html') === false) {
		http_response_code(401);
		header('Content-Type: application/json; charset=utf-8');
		echo json_encode(['message' => $ex->getMessage()]);
		exit();
	}
	OC_Template::printErrorPage($ex->getMessage(), $ex->getMessage(), 401);
} catch (MaxDelayReached $ex) {
	$request = \OC::$server->getRequest();
	/**
	 * Routes with the @CORS annotation and other API endpoints should
	 * not return a webpage, so we only print the error page when html is accepted,
	 * otherwise we reply with a JSON array like the BruteForceMiddleware would do.
	 */
	if (stripos($request->getHeader('Accept'), 'html') === false) {
		http_response_code(429);
		header('Content-Type: application/json; charset=utf-8');
		echo json_encode(['message' => $ex->getMessage()]);
		exit();
	}
	http_response_code(429);
	OC_Template::printGuestPage('core', '429');
} catch (Exception $ex) {
	\OC::$server->get(LoggerInterface::class)->error($ex->getMessage(), [
		'app' => 'index',
		'exception' => $ex,
	]);

	//show the user a detailed error page
	OC_Template::printExceptionErrorPage($ex, 500);
} catch (Error $ex) {
	try {
		\OC::$server->get(LoggerInterface::class)->error($ex->getMessage(), [
			'app' => 'index',
			'exception' => $ex,
		]);
	} catch (Error $e) {
		http_response_code(500);
		header('Content-Type: text/plain; charset=utf-8');
		print("Internal Server Error\n\n");
		print("The server encountered an internal error and was unable to complete your request.\n");
		print("Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.\n");
		print("More details can be found in the webserver log.\n");

		throw $ex;
	}
	OC_Template::printExceptionErrorPage($ex, 500);
}
```

theCalcaholic, 2024-05-02:
That's progress! You probably just need 

```sh
sudo a2enconf php8.1-fpm
sudo systemctl reload apache2
```

N0m4n904, 2024-05-02:
I get the login screen and I am able to login. Thank you soooooo much.

N0m4n904, 2024-05-02:
I am able to Update to 28.0.5 now as well. thanks a lot for the help

theCalcaholic, 2024-05-02:
Awesome, very glad to hear it!

closed 2024-05-02 by theCalcaholic