nextcloud/nextcloudpi #1910: Update to 1.54.0 fails due to password missing

GRaptor, 2024-05-02:
Hi, 
I tried to automatically and manually update to 1.54.0.
Both ways failed due to password error.

Outpup of automatic update NextcloudPi panel (v1.53.3):
`[ nc-update ] (Thu May 2 08:50:13 CEST 2024)
Downloading updates
Performing updates
Update root login prevention method...
root:x:0:0:root:/root:/usr/sbin/nologin
Password: chsh: PAM: Authentication failure
Error while applying update 1.54.0. Exiting...
`

This is similar to command line output.
Tried NCP admin password as well as my PI login password.

ncp-report shows:

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.53.3
NextcloudPi image    NextcloudPi_02-10-24
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            no
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      30G/917G
rootfs usage         30G/917G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               14days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
[... 88 lines cut ...]
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 728","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\Notifications\\BackgroundJob\\SendNotificationMails job with ID 728","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\Notifications\\BackgroundJob\\SendNotificationMails job with ID 728 in 0 seconds","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2249","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2250","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2251","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2252","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2254","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2256","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2258","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2259","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\FilesReminders\\BackgroundJob\\ScheduledNotifications job with ID 2259","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\FilesReminders\\BackgroundJob\\ScheduledNotifications job with ID 2259 in 0 seconds","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2269","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2270","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2271","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2272","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2273","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2274","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}
{"reqId":"BdZJFz4sEeGCaqKFSCxq","level":0,"time":"2024-05-02T07:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 2471","userAgent":"--","version":"28.0.3.2","data":{"app":"cron"}}

```
</details> 

Any ideas?

vcbranco, 2024-05-02:
> Hi, thank you for the report. Can you try if `sudo ncp-update fix/1.54.0-pam-error` fixes your issue?

No, is not fixed

Same error

theCalcaholic, 2024-05-02:
@vcbranco Should be now :)
Already merged in master, so `sudo ncp-update` should suffice

closed 2024-05-02 by theCalcaholic

vcbranco, 2024-05-02:
Now its fixed

Thanks

GRaptor, 2024-05-02:
Now works like a charme :-)
Thanks!