nextcloud/nextcloudpi #1911: installation via curl failed

ksd73429udmn, 2024-05-02:
Installed via # curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash on RP5 on fresh Bookworm. 
 ncp-config returns
cat: /usr/local/etc/ncp-version: No such file or directory
but starts up. 
no options available. 
"NextCloudPi configuration ver. "
Does not show any version.

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-version: No such file or directory
Nextcloud is not installed - only a limited number of commands are available
grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version
OS                   Debian GNU/Linux 12. 6.6.28+rpt-rpi-2712 (aarch64)
automount            no
USB devices          none
datadir               (doesn't exist)
rootfs usage         5.2G/28G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            wlan0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               38min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>
-->

theCalcaholic, 2024-05-02:
That likely means that there was a previous error in the installation process. Can you check that or share the full installation log?

Other than that, feel free to use the available, readily built and tested image for the RPi5 from the [official release page](https://github.com/nextcloud/nextcloudpi/releases)

ksd73429udmn, 2024-05-02:
Sure. 
[ncp.log](https://github.com/nextcloud/nextcloudpi/files/15186353/ncp.log)

theCalcaholic, 2024-05-02:
Thank you, that will be very helpful. However, I will focus on issues with the upgrade script first (I prioritize preventing breakage of existing installations over issues with setting up new ones, I hope you understand :) )

theCalcaholic, 2024-05-02:
How exactly did you run the installer? Something about those logs doesn't look right...

ksd73429udmn, 2024-05-02:
Sorry for my noobism, but 

sudo curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash |& /var/log/ncp-install.log
zsh: permission denied: /var/log/ncp-install.log

I can create the file manually and also permissions seem right, but I can not manage to write the logfile

theCalcaholic, 2024-05-02:
Ah, sorry, my bad! The correct command is:

```sh
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash |& sudo tee /var/log/ncp-install.log
```

ksd73429udmn, 2024-05-02:
❯ curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash |& sudo tee /var/log/ncp-install.log
>>> WARNING: existing mysqld configuration will be changed <<<
The 'nextcloud' database already exists. Aborting

ksd73429udmn, 2024-05-02:
The nextcloud installtion seems to be in place, all folders are  in /var/www/nextcloud, but ncp-config looks like this and the web-ui is also not accesible. 
[image]
[image]

theCalcaholic, 2024-05-02:
You need to reflash the sd with a clean debian installation before attempting the installation of NCP again

ksd73429udmn, 2024-05-02:
Hmm, still no luck. 
[ncp-install.log](https://github.com/nextcloud/nextcloudpi/files/15190114/ncp-install.log)

ksd73429udmn, 2024-05-02:
https://forums.debian.net/viewtopic.php?p=790454&sid=c78c1b2ca4badc26978cd4d656f63dbf#p790454

theCalcaholic, 2024-05-02:
> Hmm, still no luck. [ncp-install.log](https://github.com/nextcloud/nextcloudpi/files/15190114/ncp-install.log)
> 
> systemctl status systemd-resolved Unit systemd-resolved.service could not be found.

You're still not installing into a fresh Debian system according to these logs, so you I'm not surprised that this breaks again

theCalcaholic, 2024-05-02:
I'm closing this issue for now, because it's an unsupported use case and I'm not sure, if there's actually an issue with the installer if run on a fresh debian system as supposed to.

Feel free to post here again, if you try installing in a supported environment

closed 2024-05-02 by theCalcaholic

ksd73429udmn, 2024-05-02:
Ok, I see, you are right. I will try flashing the provided image. Thanks for your help so far!

theCalcaholic, 2024-05-02:
You're welcome and good luck! :)

NuclearPhoenixx, 2024-05-03:
Getting the same error with a clean install of the latest Raspberry Pi OS Lite (Bookworm) that I flashed using the Raspberry Pi Imager:

```
Applying workaround for dnsmasq bug (compare issue #1446)
+ mkdir -p /etc/systemd/resolved.conf.d
+ systemctl status systemd-resolved
Unit systemd-resolved.service could not be found.
+ systemctl stop resolvconf
Failed to stop resolvconf.service: Unit resolvconf.service not loaded.
```

it stops after that with an unfinished install.

ksd73429udmn, 2024-05-03:
Same here. I also tried flashing the provided image (NextcloudPi_RaspberryPi5_v1.54.0.zip). It installed and did the inital setup. I was not able to access via SSH though.

NuclearPhoenixx, 2024-05-03:
> Ah, sorry, my bad! The correct command is:
> 
> ```shell
> curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash |& sudo tee /var/log/ncp-install.log
> ```

Here is the log file:
[ncp-install.log](https://github.com/nextcloud/nextcloudpi/files/15201133/ncp-install.log)

This is the same error as described above:

```
+ systemctl status systemd-resolved
Unit systemd-resolved.service could not be found.
+ systemctl stop resolvconf
Failed to stop resolvconf.service: Unit resolvconf.service not loaded.
```

Notably, look at these lines just before that, relating to the `dnsmasq` install:

```
+ apt-get install --no-install-recommends -y dnsmasq
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
  resolvconf
The following NEW packages will be installed:
  dnsmasq
0 upgraded, 1 newly installed, 0 to remove and 36 not upgraded.
```

For whatever reason you specify `--no-install-recommends`, however this results in `resolvconf` not being installed as well, so obviously it can neither start nor stop the service and stops the install.

I ended up nuking mysql using `sudo apt purge mysql*` to get rid of the `WARNING: existing mysqld configuration will be changed`.

Then I manually installed `resolvconf` and ran the installer again. This time it successfully concluded the install. I'm guessing this whole problem is a result of the new Debian version either not pre-installing `resolvconf` or no longer requiring it for the install of `dnsmasq`. Whatever it is, once again, just manually installing resolvconf fixed everything for me. I have opened a PR for this: #1917

How things like this aren't tested even once before pushing a new release is beyond me. Please just run the installer once before release to mitigate issues like this. I wanted to migrate from Bullseye to Bookworm today, which would've been an easy process mainly just re-installing NCP on a fresh image since using a new install is the recommended way for Debian to upgrade. Instead we all had to deal with this now.

ksd73429udmn, 2024-05-04:
There is no need to be upset.

theCalcaholic, 2024-05-04:
@ksd73429udmn You need to activate NCP via the web browser (go to https://nextcloudpi.local or https://[ncp-ip]). After that you can enable SSH from the admin webinterface

theCalcaholic, 2024-05-04:
@NuclearPhoenixx 

> For whatever reason you specify --no-install-recommends, however this results in resolvconf not being installed 

Because that wasn't necessary in any tested scenario and would break (not fix) most installations, because it collides with alternative network managers like systemd-resolved. NCP favors systemd-resolved over resolvconf, because it's the standard for most non-VM/non-cloud setups nowadays, however resolvconf also works.
Generally though, the installer expects you to already have a working network configuration.

> How things like this aren't tested even once before pushing a new release is beyond me
Behind me lie about two weeks of nonstop testing this release. I don't think you understand the complexity of testing entire operating system setups. However, for your information:

There are automatic tests which are ran on every release (and prevent a release to be published if they fail) which cover the following scenarios:
- NCP installed on a new debian VM
- NCP updated from a debian VM running the previous release of NCP
- NCP updated from a debian VM running the last NCP version on a previous debian release, including distribution upgrade
- NCP installed on a new debian LXD container
- NCP updated from a debian LXD container running the previous release of NCP
- NCP updated from a debian LXD container running the last NCP version on a previous debian release, including distribution upgrade
- all built armbian images using an ARM emulator (qemu)

These automatic tests cover the following:
- Startup of all relevant services
- Availability of relevant tools and binaries
- Successful activation of NCP
- Successful login into Nextcloud without errors being shown in the admin section
- Various configuration checks
- Successful ncp-update to a newer (fake) NCP release

There are more tests planned, but that's what we have now.

Additional, before a release like this, I'm running countless manual tests on various configurations (aided by volunteers within the NCP community) to see if there are issues that might not be covered by the automatic tests. I'm sorry to hear that the installer didn't work for your specific system configuration, but that risk always exists, especially shortly after swapping the used OS version, which changes lots of services, packages and configurations.

NuclearPhoenixx, 2024-05-04:
> Because that wasn't necessary in any tested scenario and would break (not fix) most installations, because it collides with alternative network managers like systemd-resolved. NCP favors [...]

I see, so does adding `resolvconf` to the installer break other installs by any chance? Don't want to cause more harm than good.

> Generally though, the installer expects you to already have a working network configuration.

I would expect so too if I'm able to curl the installer. If I understand correctly now, `resolvconf` is not necessarily the best way to go with new installs, because `systemd-resolved` is favored. Then why does the installer check `resolvconf` and crashes if the service isn't there? Sounds like a deeper issue in that it shouldn't be absolutely expecting `resolvconf` to be there in order to work.

In that case, the installer shouldn't be installing `resolvconf` like in my PR and instead check for different DNS resolvers for better compat with different systems, no?

> I'm sorry to hear that the installer didn't work for your specific system configuration, but that risk always exists, especially shortly after swapping the used OS version, which changes lots of services, packages and configurations.

Like I said, I am using the official, unmodified image of Raspberry Pi OS Lite (64-bit). So nothing out of the ordinary for a Raspberry Pi. I'd have fully expected this issue to get caught in any of the automatic tests. Of course there is always a risk of breakage, more so with a new OS version. However, I'm not running some weird edge case or something, so I'm guessing then this is only a problem with Raspberry Pi OS?

theCalcaholic, 2024-05-04:
> Then why does the installer check resolvconf and crashes if the service isn't there?
It first checks for systemd-resolved and if that's not found checks for resolvconf. If that's not found either, the installation fails.

> In that case, the installer shouldn't be installing resolvconf like in my PR and instead check for different DNS resolvers for better compat with different systems, no?
Yes, exactly, but that's basically what it's doing in dnsmasq.conf. It doesn't seem to find neither resolvconf nor systemd-resolved so maybe Raspberry Pi OS is using something different still? I will check next week when I'm home and have a Raspberry Pi available for testing.

> Like I said, I am using the official, unmodified image of Raspberry Pi OS Lite (64-bit). So nothing out of the ordinary for a Raspberry Pi
Generally, because of how hard it is to test images that are meant to run on specific hardware, we don't have installation tests for those boards, but rather provide readily built images (that are, of course, tested).
That means that the installer will always be more likely to fail when installing on a custom OS for single board computers (other than Armbian, because that's the one we use to build everything).
It's still something I intend to support, but I can only react to issues once users found them, while I can actively test the official NCP images.

NuclearPhoenixx, 2024-05-04:
> It doesn't seem to find neither resolvconf nor systemd-resolved so maybe Raspberry Pi OS is using something different still?

That's my best bet now too. Uninstalling resolvconf after the install did not break anything either, so I presume some package is simply not checked for at install that's being used in Raspberry Pi OS. From what I can tell, Raspberry Pi OS uses the `network-manager` package. Any chance you can re-open the issue to get more attention to this problem?

@ksd73429udmn Did you try to replicate my method by any chance? Would be good to hear a second opinion on this, if not don't worry about it.

ksd73429udmn, 2024-05-04:
Yes, I did exactly the same. But could not get it running on my RP5 like you on RP4. I had another Redis error afterwards. Will try again though with a fresh image and report back.

ksd73429udmn, 2024-05-04:
*also running the latest Raspberry Pi OS Lite (64-bit), like you. Vanilla.

REAPERSbattlecry, 2024-05-04:
Just for the record. Had the same issue (aborting install at resolve dns) with Debian 12 clean in a VM. 
Only thing. I use an own dns specified in /etc/network/interfaces.
This seesm to cause some trouble. Installed systemd-resolved manually an restarted "networking". Now everything works fine.

ksd73429udmn, 2024-05-06:
So I tried again with installing systemd-resolved prior to runnning the script and ran into this now: 
...
+ echo 'Starting mariaDB'
Starting mariaDB
+ :
+ sudo -u mysql mysqld
+ [[ -S /run/mysqld/mysqld.sock ]]
+ sleep 0.5
2024-05-06  9:10:05 0 [Note] Starting MariaDB 10.11.6-MariaDB-0+deb12u1 source revision  as process 14189
2024-05-06  9:10:05 0 [Note] InnoDB: Compressed tables use zlib 1.2.13
2024-05-06  9:10:05 0 [Note] InnoDB: Number of transaction pools: 1
2024-05-06  9:10:05 0 [Note] InnoDB: Using ARMv8 crc32 + pmull instructions
2024-05-06  9:10:05 0 [Note] InnoDB: Using liburing
2024-05-06  9:10:05 0 [Note] InnoDB: Initializing buffer pool, total size = 256.000MiB, chunk size = 4.000MiB
2024-05-06  9:10:05 0 [Note] InnoDB: Completed initialization of buffer pool
2024-05-06  9:10:05 0 [ERROR] InnoDB: The Auto-extending data file './ibdata1' is of a different size 767 pages than specified by innodb_data_file_path
2024-05-06  9:10:05 0 [ERROR] InnoDB: Plugin initialization aborted with error Generic error
2024-05-06  9:10:05 0 [Note] InnoDB: Starting shutdown...
2024-05-06  9:10:05 0 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
2024-05-06  9:10:05 0 [Note] Plugin 'FEEDBACK' is disabled.
2024-05-06  9:10:05 0 [ERROR] Could not open mysql.plugin table: "Table 'mysql.plugin' doesn't exist". Some plugins may be not loaded
2024-05-06  9:10:05 0 [Warning] 'innodb-large-prefix' was removed. It does nothing now and exists only for compatibility with old my.cnf files.
2024-05-06  9:10:05 0 [Warning] 'innodb-file-format' was removed. It does nothing now and exists only for compatibility with old my.cnf files.
2024-05-06  9:10:05 0 [Warning] 'innodb-buffer-pool-instances' was removed. It does nothing now and exists only for compatibility with old my.cnf files.
2024-05-06  9:10:05 0 [ERROR] Unknown/unsupported storage engine: InnoDB
2024-05-06  9:10:05 0 [ERROR] Aborting
+ :
+ [[ -S /run/mysqld/mysqld.sock ]]
+ sleep 0.5
+ :
+ [[ -S /run/mysqld/mysqld.sock ]]
+ sleep 0.5
+ :
...

theCalcaholic, 2024-05-07:
@ksd73429udmn I suspect, that there are some remnants from a previous installation of mariadb/mysql - are you running the installer on a system where you tried to run it (or something else) before?