nextcloud/nextcloudpi #1913: Did ncp-dist-upgrade break your (Raspberry Pi) installation? Here's how to get it back

theCalcaholic, 2024-05-02:
## 1. Fix your package sources

- Edit /etc/apt/sources.list.d/raspi.list and replace all occurrences of `bullseye` with `bookworm`. Also make sure, no line starts with an `#` (indicates that it is ignored). The result should look something like this:
```

```
- Do the same for /etc/apt/sources.list.d/php.list

## 2. Upgrade packages

Run the following commands:
```sh
sudo apt-get update && apt-get upgrade -y --without-new-pkgs
apt-get full-upgrade -y
sudo apt-get --purge  autoremove -y
```

## 3. Install missing PHP packages

Run the following command
```sh
sudo apt-get install --no-install-recommends php8.1 php8.1-curl php8.1-gd php8.1-fpm php8.1-cli php8.1-opcache php8.1-mbstring php8.1-xml php8.1-zip php8.1-fileinfo php8.1-ldap php8.1-intl php8.1-bz2 php8.1-redis php8.1-exif php8.1-bcmath php8.1-gmp php8.1-mysql
```

## 4. Reload apache

```sh
sudo a2enconf php8.1-fpm
sudo systemctl reload apache2
```

## 4. Complete the dist-upgrade

Run the following commands:
```sh
sudo ncc maintenance:mode --off
sudo sed -i -e 's/bullseye/bookworm/' /usr/local/etc/ncp.cfg
sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade
```

**Now, Nextcloud should be back up again** (if not, try rebooting and check again).

nposnie, 2024-05-03:
Thanks for the guide. I successfully completed step 1. When i try running `sudo apt-get update && apt-get upgrade -y --without-new-pkgs` from step two I get the following error message:

E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?

Any suggestions? Thanks a lot!

nposnie, 2024-05-03:
> Thanks for the guide. I successfully completed step 1. When i try running `sudo apt-get update && apt-get upgrade -y --without-new-pkgs` from step two I get the following error message:
>
> E: Could [...]

Problem solved... I had to go into sudo -s mode and it worked... Nextcloud is back... However, I still get the following info, when I login to the raspberry pi:
Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade

Would running this command again break it again? Can I safely run the ncp updates?

Thanks for the great work!

theCalcaholic, 2024-05-03:
@nposnie Thank you for the feedback, I fixed the commands without sudo.

Can you give me the output of the following command? That should help me understand why you're still seeing the message on login:

```sh
ls -l /etc/update-motd.d/
```

You don't need to worry about it, though. The message won't break anything

nposnie, 2024-05-04:
Hi @theCalcaholic,

thanks for getting back on this! This is the output you asked for: 
-rwxr-xr-x 1 root root  48 Oct  4  2022 10logo
-rwxr-xr-x 1 root root  45 Oct  4  2022 20updates
-rwxr-xr-x 1 root root 167 May  3 09:33 30ncp-dist-upgrade

Thanks!

nposnie, 2024-05-04:
> You don't need to worry about it, though. The message won't break anything

Sorry for the confusion... The message per se is not worrying. I was wondering wether I could/should run the `sudo ncp-dist-upgrade` again without breaking it again?

theCalcaholic, 2024-05-04:
@nposnie If you ran `sudo ncp-dist-upgrade` it would just tell you that you are not on a debian version that can be upgraded (you basically completed the upgrade manually).

Can you try if the following commands remove the login message?

```sh
sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade
sudo rm -f /usr/local/etc/ncp-recommended.cfg
```

if yes, I'll adjust the guide above

sebeweiss, 2024-05-04:
> @nposnie If you ran `sudo ncp-dist-upgrade` it would just tell you that you are not on a debian version that can be upgraded (you basically completed the upgrade manually).
> 
> Can you try if the following commands remove the login message?
> 
> ```shell
> sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade
> sudo rm -f /usr/local/etc/ncp-recommended.cfg
> ```
> 
> if yes, I'll adjust the guide above

```
sudo rm -f /usr/local/etc/ncp-recommended.cfg
```
remove the message for me.

nposnie, 2024-05-04:
> ```shell
> sudo rm -f /usr/local/etc/ncp-recommended.cfg
> ```

Excellent! Message is gone! Thanks a lot for your quick help!

Schniefel, 2024-05-08:
@theCalcaholic is it now safe to use dist-upgrade of nextcloudpi on raspberry pi os (bullseye)? Or will you integrate additional fixes for 1.54.x?

sjibbert, 2024-05-08:
I am running a Pi4 with ncp installed to an SSD. I just updated from v1.53.0 to v1.54.0 and it seems i ran into the same issues as [https://github.com/nextcloud/nextcloudpi/issues/1921](url). Like @sebeweiss, I ran `sudo apt-get install zstd` in an effort to fix the "4-dphys-swapfile_20100506-7.1+rpt3_all.deb" error message. Subsequently, like him, I ran all the commands in the above step-by-step guide, sans step 3. After rebooting the raspberry and navigating to the NC portal, I still am looking at the message `Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.`

ncp system-info shows mainly positive messages, although these two don't look too good:
‣ You are using an outdated distro release. You should upgrade to latest Debian/Raspbian
HPB service | down

some more info from said page, since it complains about the distro to be outdated:
NextcloudPi version | v1.54.0
NextcloudPi image | NextCloudPi_RaspberryPi_v1.51.2.img
OS | Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)

As you can see below, the very first step from your howto still results in an issue regarding the swap file for some reason:

`ncpadmin@nextcloudpi:~ $ sudo apt-get update && sudo apt-get upgrade -y --witho                                                                                                                                                                                                ut-new-pkgs
[sudo] password for ncptijmen:
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://archive.raspberrypi.org/debian bookworm InRelease
Hit:3 https://packages.sury.org/php bookworm InRelease
Hit:4 http://deb.debian.org/debian bullseye InRelease
Hit:5 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  alsa-utils console-setup console-setup-linux keyboard-configuration
  kms++-utils libcamera-apps-lite libkms++0 libnm0 network-manager
  php8.1-bcmath php8.1-bz2 php8.1-cli php8.1-common php8.1-curl php8.1-fpm
  php8.1-gd php8.1-gmp php8.1-intl php8.1-ldap php8.1-mbstring php8.1-mysql
  php8.1-opcache php8.1-readline php8.1-redis php8.1-xml php8.1-zip
  python3-kms++ python3-libcamera python3-picamera2 python3-pidng
  python3-simplejpeg python3-spidev raspi-utils
The following packages will be upgraded:
  dphys-swapfile
1 upgraded, 0 newly installed, 0 to remove and 33 not upgraded.
Need to get 0 B/22.6 kB of archives.
After this operation, 2,048 B of additional disk space will be used.
apt-listchanges: Reading changelogs...
dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rp                                                                                                                                                                                                t3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
Traceback (most recent call last):
  File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
    output = subprocess.check_output(command)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archi                                                                                                                                                                                                ves/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', '                                                                                                                                                                                                Architecture', 'Status']' returned non-zero exit status 2.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/bin/apt-listchanges", line 323, in <module>
    main(config)
  File "/usr/bin/apt-listchanges", line 104, in main
    pkg = DebianFiles.Package(deb)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
    parser.readdeb(self.path)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
    raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-                                                                                                                                                                                                7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-sw                                                                                                                                                                                                apfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture                                                                                               [... cut ...]
dpkg: error processing archive /var/cache/apt/archives/dphys-swapfile_20100506-7                                                                                                                                                                                                .1+rpt3_all.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

If you need more info from me, let me know

theCalcaholic, 2024-05-08:
@sjibbert That should already be fixed. Can you run `ncp-update` again (should show a succesful update to v1.54.2)?

@Schniefel Yes, every known issue should be fixed. Just make sure you're on v1.54.2

sjibbert, 2024-05-08:
I just did, unfortunately, this results in errors, causing the update (to 1.54.1, not 1.54.2!) to be cancelled: 

> ncpadmin@nextcloudpi:~ $ sudo ncp-update
> Downloading updates
> Performing updates
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Update root login prevention method...
> done.
> Fixing trusted proxies list...
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> done.
> Updating PHP package signing key...
> Get:1 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
> Get:2 http://archive.raspberrypi.org/debian bookworm InRelease [23.6 kB]
> Hit:3 https://packages.sury.org/php bookworm InRelease
> Get:4 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages [410 kB]
> Get:5 http://archive.raspberrypi.org/debian bookworm/main armhf Packages [419 kB]
> Hit:6 http://deb.debian.org/debian bullseye InRelease
> Hit:7 http://deb.debian.org/debian bullseye-updates InRelease
> Fetched 901 kB in 6s (152 kB/s)
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> gnupg2 is already the newest version (2.2.27-2+deb11u2).
> 0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
> Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
> Executing: /tmp/apt-key-gpghome.cZDjjEtrsu/gpg.1.sh --fetch-keys https://packages.sury.org/php/apt.gpg
> gpg: requesting key from 'https://packages.sury.org/php/apt.gpg'
> gpg: key B188E2B695BD4743: "DEB.SURY.ORG Automatic Signing Key <deb@sury.org>" not changed
> gpg: Total number processed: 1
> gpg:              unchanged: 1
> done.
> Installing dependencies...
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> tmux is already the newest version (3.1c-1+deb11u1).
> 0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
> done.
> Updating obsolete theming URL
> Cannot load Zend OPcache - it was already loaded
> done.
> NextCloudPi updated to version v1.54.0
> Reenable erroneously disabled package sources
> done
> Hit:1 http://deb.debian.org/debian bullseye InRelease
> Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
> Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
> Hit:4 https://packages.sury.org/php bookworm InRelease
> Hit:5 http://archive.raspberrypi.org/debian bookworm InRelease
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Calculating upgrade... Done
> The following packages have been kept back:
>   alsa-utils console-setup console-setup-linux keyboard-configuration kms++-utils libcamera-apps-lite libkms++0 libnm0 network-manager php8.1-bcmath php8.1-bz2 php8.1-cli php8.1-common php8.1-curl php8.1-fpm php8.1-gd php8.1-gmp php8.1-intl php8.1-ldap php8.1-mbstring
>   php8.1-mysql php8.1-opcache php8.1-readline php8.1-redis php8.1-xml php8.1-zip python3-kms++ python3-libcamera python3-picamera2 python3-pidng python3-simplejpeg python3-spidev raspi-utils
> The following packages will be upgraded:
>   dphys-swapfile
> 1 upgraded, 0 newly installed, 0 to remove and 33 not upgraded.
> Need to get 0 B/22.6 kB of archives.
> After this operation, 2,048 B of additional disk space will be used.
> apt-listchanges: Reading changelogs...
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> Traceback (most recent call last):
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
>     output = subprocess.check_output(command)
>   File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
>     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
>   File "/usr/lib/python3.9/subprocess.py", line 528, in run
>     raise CalledProcessError(retcode, process.args,
> subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/usr/bin/apt-listchanges", line 323, in <module>
>     main(config)
>   File "/usr/bin/apt-listchanges", line 104, in main
>     pkg = DebianFiles.Package(deb)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
>     parser.readdeb(self.path)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
>     raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
> RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> dpkg: error processing archive /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb (--unpack):
>  dpkg-deb --control subprocess returned error exit status 2
> Errors were encountered while processing:
>  /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)
> Error while applying update 1.54.1. Exiting...
>

theCalcaholic, 2024-05-09:
@sjibbert Oh, thank you, that makes sense. Can you try running `ncp-update` again? I just pushed a small fix.

sjibbert, 2024-05-09:
I just ran ncp-update again and I seem to have succesfully updated. I won't copy all output, but the operation ended with the following:

> NextCloudPi updated to version v1.54.0
> Reenable erroneously disabled package sources
> done
> Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
> Hit:2 http://deb.debian.org/debian bullseye InRelease
> Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
> Hit:4 http://archive.raspberrypi.org/debian bookworm InRelease
> Hit:5 https://packages.sury.org/php bookworm InRelease
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> zstd is already the newest version (1.4.8+dfsg-2.1).
> 0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
> NextCloudPi updated to version v1.54.1
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> zstd is already the newest version (1.4.8+dfsg-2.1).
> 0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
> Cannot load Zend OPcache - it was already loaded
> Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade
> Cannot load Zend OPcache - it was already loaded
> 
> Broadcast message from ncpadmin@nextcloudpi (pts/0) (Thu May  9 12:36:56 2024)
> 
> Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade
> 
> NextCloudPi updated to version v1.54.2

i did not do `sudo ncp-dist-upgrade yet`. I did however reboot the system and tried to access the portal. yet again yielding the message 
> Internal Server Error etc etc etc

NC admin dashboard system info still says:
>HPB service | down

I suppose I should do the dist-upgrade?

theCalcaholic, 2024-05-09:
@sjibbert Since you ran ncp-dist-upgrade already, you will most likely need to follow the manual instructions above instead of rerunning the script (no harm in trying though).

sjibbert, 2024-05-09:
package sources under step 1 already/still contained bookworm instead of bullseye. no adjustment neccecary

`sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs`

yields:

> ncpadmin@nextcloudpi:~ $ sudo apt-get update && sudo apt-get upgrade -y --witho                                                                                                                                                                                                ut-new-pkgs
> Hit:1 http://deb.debian.org/debian bullseye InRelease
> Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
> Hit:3 https://packages.sury.org/php bookworm InRelease
> Get:4 http://archive.raspberrypi.org/debian bookworm InRelease [23.6 kB]
> Get:5 http://archive.raspberrypi.org/debian bookworm/main arm64 Packages [410 kB                                                                                                                                                                                                ]
> Get:6 http://archive.raspberrypi.org/debian bookworm/main armhf Packages [419 kB                                                                                                                                                                                                ]
> Hit:7 http://security.debian.org/debian-security bullseye-security InRelease
> Fetched 852 kB in 6s (152 kB/s)
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Calculating upgrade... Done
> The following packages have been kept back:
>   alsa-utils console-setup console-setup-linux keyboard-configuration
>   kms++-utils libcamera-apps-lite libkms++0 libnm0 network-manager
>   php8.1-bcmath php8.1-bz2 php8.1-cli php8.1-common php8.1-curl php8.1-fpm
>   php8.1-gd php8.1-gmp php8.1-intl php8.1-ldap php8.1-mbstring php8.1-mysql
>   php8.1-opcache php8.1-readline php8.1-redis php8.1-xml php8.1-zip
>   python3-kms++ python3-libcamera python3-picamera2 python3-pidng
>   python3-simplejpeg python3-spidev raspi-utils
> The following packages will be upgraded:
>   dphys-swapfile
> 1 upgraded, 0 newly installed, 0 to remove and 33 not upgraded.
> Need to get 0 B/22.6 kB of archives.
> After this operation, 2,048 B of additional disk space will be used.
> apt-listchanges: Reading changelogs...
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rp                                                                                                                                                                                                t3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> Traceback (most recent call last):
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
>     output = subprocess.check_output(command)
>   File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
>     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
>   File "/usr/lib/python3.9/subprocess.py", line 528, in run
>     raise CalledProcessError(retcode, process.args,
> subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archi                                                                                                                                                                                                ves/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', '                                                                                                                                                                                                Architecture', 'Status']' returned non-zero exit status 2.
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/usr/bin/apt-listchanges", line 323, in <module>
>     main(config)
>   File "/usr/bin/apt-listchanges", line 104, in main
>     pkg = DebianFiles.Package(deb)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
>     parser.readdeb(self.path)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
>     raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
> RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-                                                                                                                                                                                                7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-sw                                                                                                                                                                                                apfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture                                                                                                                                                                                                ', 'Status']' returned non-zero exit status 2.
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rp                                                                                                                                                                                                t3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
[... cut ...] /var/cache/apt/archives/dphys-swapfile_20100506-7                                                                                                                                                                                                .1+rpt3_all.deb (--unpack):
>  dpkg-deb --control subprocess returned error exit status 2
> Errors were encountered while processing:
>  /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)

theCalcaholic, 2024-05-09:
@sjibbert But you have zstd installed, right? Can you check if `apt-get install zstd` changes anything?

theCalcaholic, 2024-05-09:
@sjibbert I see, this seems to be linked to an old version of dpkg. Can you run `apt-get upgrade dpkg` and then try the dist-upgrade again?

sjibbert, 2024-05-09:
this doesn't look promising: 

> ncpadmin@nextcloudpi:~ $ sudo apt-get upgrade dpkg
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> dpkg is already the newest version (1.20.13).
> Calculating upgrade... Done
> The following packages have been kept back:
>   alsa-utils console-setup console-setup-linux keyboard-configuration kms++-utils libcamera-apps-lite libkms++0 libnm0 network-manager php8.1-bcmath php8.1-bz2 php8.1-cli php8.1-common php8.1-curl php8.1-fpm php8.1-gd php8.1-gmp php8.1-intl php8.1-ldap php8.1-mbstring
>   php8.1-mysql php8.1-opcache php8.1-readline php8.1-redis php8.1-xml php8.1-zip python3-kms++ python3-libcamera python3-picamera2 python3-pidng python3-simplejpeg python3-spidev raspi-utils
> The following packages will be upgraded:
>   dphys-swapfile
> 1 upgraded, 0 newly installed, 0 to remove and 33 not upgraded.
> Need to get 0 B/22.6 kB of archives.
> After this operation, 2,048 B of additional disk space will be used.
> Do you want to continue? [Y/n] y
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> Traceback (most recent call last):
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
>     output = subprocess.check_output(command)
>   File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
>     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
>   File "/usr/lib/python3.9/subprocess.py", line 528, in run
>     raise CalledProcessError(retcode, process.args,
> subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/usr/bin/apt-listchanges", line 323, in <module>
>     main(config)
>   File "/usr/bin/apt-listchanges", line 104, in main
>     pkg = DebianFiles.Package(deb)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
>     parser.readdeb(self.path)
>   File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
>     raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
> RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> dpkg: error processing archive /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb (--unpack):
>  dpkg-deb --control subprocess returned error exit status 2
> Errors were encountered while processing:
>  /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)

theCalcaholic, 2024-05-09:
What's your `dpkg --version`?

sjibbert, 2024-05-09:
Debian 'dpkg' package management program version 1.20.13 (arm64).

theCalcaholic, 2024-05-09:
I see. zstd support was added to dpkg in version 1.21.18.
However, this should be the latest version in the bookworm repositories. I'm wondering, why it doesn't get updated...
Have you checked all files in `/etc/apt/sources.list.d/*` and `/etc/apt/sources.list` if they are configured for bookworm instead of bullseye?

theCalcaholic, 2024-05-09:
I think I see the issue:

> ncpadmin@nextcloudpi:~ $ sudo apt-get update && sudo apt-get upgrade -y --witho ut-new-pkgs
> Hit:1 http://deb.debian.org/debian bullseye InRelease
> Hit:2 http://deb.debian.org/debian bullseye-updates InRelease

There are still bullseye package sources configured in your system.

sjibbert, 2024-05-09:
That's correct, step one only mentioned "/etc/apt/sources.list.d/raspi.list" and "/etc/apt/sources.list.d/php.list", so I only manually altered those two.
I just changed "/etc/apt/sources.list" in the same manner to:
> deb http://deb.debian.org/debian bookworm main contrib non-free
> deb http://security.debian.org/debian-security bookworm-security main contrib non-free
> deb http://deb.debian.org/debian bookworm-updates main contrib non-free

`sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs` resulted in a whole lot of packages being updated. somewhere during the process, I spotted this:
> dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
> Traceback (most recent call last):
>   File [...]

`dpkg --version`
> Debian 'dpkg' package management program version 1.21.22 (arm64).

`sudo apt-get install zstd`
> The following packages will be upgraded:
>   libstdc++6 zstd
> 2 upgraded, 1 newly installed, 0 to remove and 174 not upgraded.

again, i prompt `sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs`. No errors this time.
`sudo apt-get full-upgrade -y`. a lot of preparing, unpacking and setting up. No errors. Somewhere during this process, it was suggested I reboot, so i did.
`sudo apt-get --purge  autoremove -y`
`sudo apt-get install --no-install-recommends php8.1 php8.1-curl php8.1-gd php8.1-fpm php8.1-cli php8.1-opcache php8.1-mbstring php8.1-xml php8.1-zip php8.1-fileinfo php8.1-ldap php8.1-intl php8.1-bz2 php8.1-redis php8.1-exif php8.1-bcmath php8.1-gmp php8.1-mysql`
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

`sudo a2enconf php8.1-fpm`
> Conf php8.1-fpm already enabled

`sudo systemctl reload apache2`
`sudo ncc maintenance:mode --off`
> Cannot load Zend OPcache - it was already loaded
> Maintenance mode already disabled

`sudo sed -i -e 's/bullseye/bookworm/' /usr/local/etc/ncp.cfg`
`sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade `
`sudo rm -f /usr/local/etc/ncp-recommended.cfg`
`sudo reboot`

Still:

> Internal Server Error
> 
> The server encountered an internal error and was unable to complete your request.
> Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
> More details can be found in the server log.

and

> HPB service | down

Schniefel, 2024-05-09:
dist-upgrade didn't work for me, so i am now completely reinstalling after a few tries.
Is it still possible to install via the curl-script on a freshly flashed raspberry pi os (bookworm) or is this not supported and i need a full debian or to directly flash the ncp-image? didn't directly find something in the documentation. I wonder because the image is now based on armbian.

theCalcaholic, 2024-05-11:
@Schniefel Some issues have been reported with the curl installer on Raspberry Pi OS specifically that will probably take me another week to fix.

So you could either wait for that or use the Armbian image (or run NCP via LXD/LXC inside Raspberry Pi OS)

Schniefel, 2024-05-11:
@theCalcaholic look at: https://github.com/nextcloud/nextcloudpi/issues/1926

I tried it out and got it working but had to manually install systemd-resolved before running the script

theCalcaholic, 2024-05-20:
@Halloran Thank you, yeah that's a typo :)

janvlug, 2024-05-25:
I ran today into this issue: https://github.com/nextcloud/nextcloudpi/issues/1930
Using NextcloudPi version: v1.54.2.

pokaalinkanssayohon, 2024-06-08:
I just ran ` sudo ncp-dist-upgrade`  and after ca. 1h still output is full of dots only...
[image]
So if everything is not ok, hopefully script has timeout (?).

pokaalinkanssayohon, 2024-06-08:
Nope, I see, that nothing happens, script is hanging and when I try to follow these instructions:
` `sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs
` Errors were encountered while processing:
 /tmp/apt-dpkg-install-tMOEmX/00-dphys-swapfile_20100506-7.1+rpt3_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

tomcat-xx, 2024-06-19:
I also ran into this issue with v1.54.2 (on Buster, not sure if this is the same for all).

Before I got that far, I first had to solve issue with gcc-8-base, described in #1460. Seems still exist.

While following instruction above, struggled with following:
1. I edited /etc/apt/sources.list.d/raspi.list, but there is no /etc/apt/sources.list.d/php.list. So I ignored this step.
2. Maybe because above, when trying to install PHP, I got message that "php8.1" is interpreted as regex and php8.1-common is used, but ended with errors also. When changing all 8.1 to 8.2 there is no message, installation working, but then message from NCP that 8.2 is not supported.

Any idea? Thanks.
Thomas

m-breitbach, 2024-06-21:
Starting from 1.52.2 I too ran into the following:
```
The following packages have unmet dependencies:
 libc-dev-bin : Depends: libc6 (< 2.32) but 2.36-9+deb12u7 is to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+deb11u10) but 2.36-9+deb12u7 is to be installed
E: Broken packages
```
Your steps fixed it luckily, thanks.

tomcat-xx, 2024-07-04:
Thanks  to wiborgit I finally made it: 
[https://help.nextcloud.com/t/nextcloudpi-updates-von-php7-3-auf-7-4-incl-auf-bullseye/135086/22](https://help.nextcloud.com/t/nextcloudpi-updates-von-php7-3-auf-7-4-incl-auf-bullseye/135086/22)
-> step 8, give up, new installation ;-)

dwprod38, 2024-07-07:
Same problem with issue # fresh RPi5 install from SD flashed with "https://github.com/nextcloud/nextcloudpi/releases/download/v1.54.0/NextcloudPi_RaspberryPi5_v1.54.0.zip"

After activation page and SSH : reboot of RPi5 (takes 10min !!!! To get prompt login !!!

And launching "sudo ncp-update", run until "Processing Triggers for man-db (2.11,2-2)" and nothing happening during 24h...

[image]

JayGatsby7, 2024-08-31:
my system broke today doing this update.  going to reinstall but figured I would let you know what the error looked like.``Errors were encountered while processing:
 /tmp/apt-dpkg-install-7aMcve/01-raspi-utils-core_20240402-4_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)``

JayGatsby7, 2024-09-02:
> looks like the images are all still based on bullseye. Is the only way to update to bookworm via ncp-dist-upgrade?

correction -- the new images are on bookworm.  thanks again - love this project been using it for years.

theCalcaholic, 2024-11-08:
I'll close this issue now, since all bugs have been fixed since and there shouldn't be any new occurrences of the original bug.

closed 2024-11-08 by theCalcaholic

Slash241, 2024-12-19:
I get this error and have not be able to figure out how to get back it? Any Idea

pi@PiHole:~ $ sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs
Hit:1 http://archive.raspberrypi.org/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  alsa-utils console-setup console-setup-linux dpkg keyboard-configuration libcamera-apps-lite libgstreamer-plugins-base1.0-0 python3-gpiozero
  python3-spidev raspberrypi-net-mods raspberrypi-sys-mods raspi-config raspinfo rpi-eeprom
The following packages will be upgraded:
  dphys-swapfile
1 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Need to get 0 B/22.6 kB of archives.
After this operation, 2048 B of additional disk space will be used.
apt-listchanges: Reading changelogs...
dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
Traceback (most recent call last):
  File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
    output = subprocess.check_output(command)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/bin/apt-listchanges", line 323, in <module>
    main(config)
  File "/usr/bin/apt-listchanges", line 104, in main
    pkg = DebianFiles.Package(deb)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
    parser.readdeb(self.path)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
    raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
dpkg-deb: error: archive '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@PiHole:~ $

janvlug, 2024-12-20:
> Any Idea

Check: https://github.com/nextcloud/nextcloudpi/issues/1921#issue-2279097145

bluejay76, 2025-03-14:
I've run these steps and still cannot use my NCP (since last 6 months or so; usage broke without my intervention, with some background update). I get the Internal Server Error and unable to load the dashboard. I can access Panel, which shows Nextcloud check: error, HPB service:down. PHP I believe used to be down, but at least that appears fixed. I'm on v1.55.3, image NextCloudPi_10-08-21. Have even been using AI to diagnose and help with fixing to no avail. Can anyone help?