nextcloud/nextcloudpi #1914: Missing Image Raspberry Pi 3. update v1.54.0

teddyhoss, 2024-05-02:
On latest release is missing image for Pi 3 and older. Image for Rasberry pi 4 is kernel paniking my b3+

theCalcaholic, 2024-05-02:
I don't intend to readd support for the RPi3 unless this gets a ton of requests, because a) it's an 8 year old board and b) has not been well recommended for NCP for many years (due to its lack of performance).

But if you want to run NCP on a RPi3, you can always install an older image and update from there.

closed 2024-05-02 by theCalcaholic

teddyhoss, 2024-05-02:
My idea when I opened the thread is not a periodic support for RB3 I think a newer build based on Debian 12 / Latest Arabian release is better to start than a "dist-upgrade".

theCalcaholic, 2024-05-02:
Yes, however, as of v1.54.0, all builds (including RPi 4&5) are based on Armbian and afaik, there's no upstream support for the RPi3

teddyhoss, 2024-05-02:
> > @teddyhoss wrote:
> 
> > I know that you can download armbian base and install everything via the script but for the end user it would be better to have (in my opinion) a last image (EOL) for raspebbies other than 4 and 5 and a writing on the chagelog saying which will no longer be compiled in Github releases.
> 
> 
> 
> The main reason, why I dropped support for building RPi3 images, is, that as of v1.54.0 all images (including RPi 4&5) are based on Armbian and there is no upstream support for the RPi3

Armbian is still working on RPi 3A+ 3B 3B+.(tested on Feb 13, 2024 Jammy Relase) 

3B+ support x64 too.

theCalcaholic, 2024-05-02:
There is no board config for it available in https://github.com/armbian/build/tree/main/config/boards

Maybe it's community maintained, but nothing we can easily build off.

theCalcaholic, 2024-05-02:
Allthough, [according to the documentation](https://www.armbian.com/rpi4b/) the rpi4 image is supposed to work on RPi 3 - if it doesn't, there's not much I can do about it, though.

dorsiflexion, 2025-05-18:
In the Installation section on the [website](https://nextcloudpi.com/), the Raspberry Pi tab should say *Raspberry Pi (4 and later)* instead of *Raspberry Pi (3 and later)* then.