nextcloud/nextcloudpi #1915: ncp-dist-upgrade to bookworm failed

lxto, 2024-05-02:
When ncp-1.54.0 came out this morning I already had upgraded my NC installation to NC-29.0-stable. My system is a Raspi-4 8GB, booting from SSD. After successfully upgrading ncp to 1.54.0 i tried to perform the ncp-dist-upgrade to bookworm which failed :

> This is a dangerous process that is only guaranteed to work properly if you
have not made manual changes in the system. Backup the SD card first and
proceed at your own risk.

>Note that this is not a requirement for NCP to continue working properly.
The current distribution will keep receiving updates for some time.

>Do you want to continue? [y/N]y

>You still have PHP version 7.4 installed. Please update to the latest supported version of nextcloud (which will also update your PHP version) before proceeding with the distribution upgrade.
Exiting.
root@nextcloudpi:/home/pi# 

I manually updated php to 8.3.3 some weeks before. php 7.4 is no longer installed (>sudo apt-get purge php7.*) 
The php-version used might be altered via "update-alternatives --config php":

>root@nextcloudpi:/home/pi# php -v
PHP 8.3.3-1+0~20240216.17+debian11~1.gbp87e37b (cli) (built: Feb 16 2024 10:33:07) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.3.3, Copyright (c) Zend Technologies
    with Zend OPcache v8.3.3-1+0~20240216.17+debian11~1.gbp87e37b, Copyright (c), by Zend Technologies

Is there a chance to upgrade my system although I'm on NC-29 and php-8.3  ?

theCalcaholic, 2024-05-02:
I'm afraid with manually upgrading PHP you lost some of the automation capabilities of NCP and the dist-upgrade will likely fail. 

You could try to revert PHP back to 8.1, but I can not give you any guarantee that this will work. Nextcloud 29 should still support PHP 8.1 though and NCP will probably automatically upgrade PHP to 8.3 when upgrading to NC 29.0.1

lxto, 2024-05-05:
Today i could successfully upgrade my ncp installation to Bookworm following your guide . It worked although my NC already is on version 29.0.0. I used
“sudo update-alternatives --config php” to adjust my php to v. 8.1.
I also edited /etc/apt/resources.list and replaced all "bullseye" with "bookworm"
To be safe, i cloned the installation-SSD before proceding. :wink:
Thanks for your advice.

closed 2024-05-06 by theCalcaholic