nextcloud/nextcloudpi #1920: Error when updating: Table 'nextcloud.oc_group_user' doesn't exist

janvlug, 2024-05-04:
When I try to run `sudo ncp-update` on the command line, I get this error message:

```
ERROR 1146 (42S02) at line 1: Table 'nextcloud.oc_group_user' doesn't exist
admin user not found
```

janvlug, 2024-05-04:
I guess this error comes from this line:

https://github.com/nextcloud/nextcloudpi/blob/c75fa441176666db988498e36d539c1aa6c9b7ce/etc/library.sh#L571

There is a prefix `oc_` used, but in my installation, the tables do not have a prefix.

This issue is similar to issue #1818

theCalcaholic, 2024-05-04:
Hi, thank you for reporting!

Can you try running `sudo ncp-update devel`?

janvlug, 2024-05-04:
> Can you try running `sudo ncp-update devel`?

I ran this from the command line, and all went fine. Thanks!

theCalcaholic, 2024-05-04:
Great, thank youbfor testing

theCalcaholic, 2024-05-06:
Fix is now in master

closed 2024-05-06 by theCalcaholic