nextcloud/nextcloudpi #1921: ncp-dist-upgrade to bookworm failed: 'uses unknown compression for member 'control.tar.zst'' --> solution

sebeweiss, 2024-05-04:
When I tried to upgrade my instance (NCP 1.54.0, NC 28.0.5) on Raspberry Pi 4 Model B with `ncp-dist-upgrade` I ran into the following error:

```
dpkg-deb: error: archive '/tmp/apt-dpkg-install-OUyk8N/4-dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
```

<details>
  <summary>Click me for long message</summary>

```
  File "/usr/share/apt-listchanges/DebianFiles.py", line 124, in readdeb
    output = subprocess.check_output(command)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/bin/apt-listchanges", line 323, in <module>
    main(config)
  File "/usr/bin/apt-listchanges", line 104, in main
    pkg = DebianFiles.Package(deb)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 358, in __init__
    parser.readdeb(self.path)
  File "/usr/share/apt-listchanges/DebianFiles.py", line 127, in readdeb
    raise RuntimeError(_("Error processing '%(what)s': %(errmsg)s") %
RuntimeError: Error processing '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb': Command '['dpkg-deb', '-f', '/var/cache/apt/archives/dphys-swapfile_20100506-7.1+rpt3_all.deb', 'Package', 'Source', 'Version', 'Architecture', 'Status']' returned non-zero exit status 2.
apt-listchanges: Reading changelogs...
Preconfiguring packages ...
Fetched 101 MB in 8s (12.2 MB/s)
(Reading database ... 42307 files and directories currently installed.)
Preparing to unpack .../base-files_12.4+deb12u5_arm64.deb ...
Unpacking base-files (12.4+deb12u5) over (11.1+deb11u9) ...
Setting up base-files (12.4+deb12u5) ...
Installing new version of config file /etc/debian_version ...
Installing new version of config file /etc/issue ...
Installing new version of config file /etc/issue.net ...
(Reading database ... 42307 files and directories currently installed.)
Preparing to unpack .../0-init_1.65.2_arm64.deb ...
Unpacking init (1.65.2) over (1.60) ...
Preparing to unpack .../1-systemd-sysv_252.22-1~deb12u1_arm64.deb ...
Unpacking systemd-sysv (252.22-1~deb12u1) over (247.3-7+deb11u4) ...
Preparing to unpack .../2-console-setup-linux_1.221rpt1_all.deb ...
Unpacking console-setup-linux (1.221rpt1) over (1.205+rpt1) ...
Preparing to unpack .../3-python3-debconf_1.5.82_all.deb ...
Unpacking python3-debconf (1.5.82) over (1.5.77) ...
Preparing to unpack .../4-libtext-wrapi18n-perl_0.06-10_all.deb ...
Unpacking libtext-wrapi18n-perl (0.06-10) over (0.06-9) ...
Preparing to unpack .../5-debconf-i18n_1.5.82_all.deb ...
Unpacking debconf-i18n (1.5.82) over (1.5.77) ...
Preparing to unpack .../6-debconf-utils_1.5.82_all.deb ...
Unpacking debconf-utils (1.5.82) over (1.5.77) ...
Preparing to unpack .../7-debconf_1.5.82_all.deb ...
Unpacking debconf (1.5.82) over (1.5.77) ...
Setting up debconf (1.5.82) ...
(Reading database ... 42313 files and directories currently installed.)
Preparing to unpack .../0-xkb-data_2.35.1-1_all.deb ...
Unpacking xkb-data (2.35.1-1) over (2.29-2) ...
Preparing to unpack .../1-console-setup_1.221rpt1_all.deb ...
Unpacking console-setup (1.221rpt1) over (1.205+rpt1) ...
Preparing to unpack .../2-keyboard-configuration_1.221rpt1_all.deb ...
Unpacking keyboard-configuration (1.221rpt1) over (1.205+rpt1) ...
Preparing to unpack .../3-dc_1.07.1-3_arm64.deb ...
Unpacking dc (1.07.1-3) over (1.07.1-2+b2) ...
dpkg-deb: error: archive '/tmp/apt-dpkg-install-OUyk8N/4-dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /tmp/apt-dpkg-install-OUyk8N/4-dphys-swapfile_20100506-7.1+rpt3_all.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Preparing to unpack .../5-php8.1-igbinary_3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8_arm64.deb ...
Unpacking php8.1-igbinary (3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8) over (3.2.15-1+0~20231202.49+debian11~1.gbpe9ffe8) ...
Preparing to unpack .../6-tasksel-data_3.73_all.deb ...
Unpacking tasksel-data (3.73) over (3.68+deb11u1) ...
Preparing to unpack .../7-tasksel_3.73_all.deb ...
Unpacking tasksel (3.73) over (3.68+deb11u1) ...
Preparing to unpack .../8-libpam-runtime_1.5.2-6+rpt2+deb12u1_all.deb ...
Unpacking libpam-runtime (1.5.2-6+rpt2+deb12u1) over (1.4.0-9+deb11u1+rpt2) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-OUyk8N/4-dphys-swapfile_20100506-7.1+rpt3_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

</details>

As suggested [here](https://github.com/yktoo/indicator-sound-switcher/issues/113), I installed `zstd` using:
```
sudo apt-get install zstd
```
After that, I followed the instructions in [Did ncp-dist-upgrade break your (Raspberry Pi) installation? Here's how to get it back #1913](https://github.com/nextcloud/nextcloudpi/issues/1913), but without  point 3.

The upgrade has now been successfully completed. Everything seems to be working properly.

theCalcaholic, 2024-05-04:
Thank you for reporting! That's a really obscure issue, tbh.

Would you mind telling me what your setup is? (Hardware and OS)?

I'll add zstd anyway, shouldn't hurt (and was probably just present on all systems I tested the dist-upgrade on)

sebeweiss, 2024-05-04:
Yes, of course.
Raspberry PI 4 Model B (booting and running from SSD).
It was my installation from the Bullseye NCP-image I installed in November 2022 and kept updating (NCP and NC) until I finally updated to NCP 1.54.0 and NC 28.0.5 in the last days.

theCalcaholic, 2024-05-06:
A fix is now in master

closed 2024-05-06 by theCalcaholic