nextcloud/nextcloudpi #1922: Failed to connect to the database after 1.54.0 dist-upgrade. An exception occurred in the driver

zedocrob, 2024-05-05:
Hi, after sudo ncp-dist-upgrade i can't connect to the database:

`An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:139
`

Rebooted the server (OdroidHC4) but same problem, should i update Nextcloud to 28.0.5? or restore mt last NC Backup maybe?

Thanks

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: Aucun fichier ou dossier de ce type
NextcloudPi version  v1.54.0
NextcloudPi image    NextCloudPi_08-16-22
OS                   Armbian 24.2.1 bookworm \l . 6.6.16-current-meson64 (aarch64)
automount            yes
USB devices          none
datadir              /media/DATAcloud/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      940G/3,7T
rootfs usage         6,4G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            end0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               15min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1654): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1081): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(337): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(377): Doctrine\DBAL\Query\QueryBuilder->executeQuery()
#5 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(280): Doctrine\DBAL\Query\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(421): OC\DB\QueryBuilder\QueryBuilder->execute()
#7 /var/www/nextcloud/lib/private/AppConfig.php(187): OC\AppConfig->loadConfigValues()
#8 /var/www/nextcloud/lib/private/AppConfig.php(377): OC\AppConfig->getApps()
#9 /var/www/nextcloud/lib/private/App/AppManager.php(128): OC\AppConfig->getValues()
#10 /var/www/nextcloud/lib/private/App/AppManager.php(149): OC\App\AppManager->getInstalledAppsValues()
#11 /var/www/nextcloud/lib/private/legacy/OC_App.php(231): OC\App\AppManager->getInstalledApps()
#12 /var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php(90): OC_App::getEnabledApps()
#13 /var/www/nextcloud/lib/base.php(700): OC\AppFramework\Bootstrap\Coordinator->runInitialRegistration()
#14 /var/www/nextcloud/lib/base.php(1200): OC::init()
#15 /var/www/nextcloud/console.php(48): require_once('...')
#16 /var/www/nextcloud/occ(11): require_once('...')
#17 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun May 05 02:25:02.891216 2024] [authz_host:error] [pid 2753:tid 281473086058784] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/dashboard.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:02.923306 2024] [authz_host:error] [pid 2753:tid 281472993063200] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/settings-white.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:02.938729 2024] [authz_host:error] [pid 2753:tid 281473001517344] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/search.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:02.966001 2024] [authz_host:error] [pid 2753:tid 281473153282336] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/wizard.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:02.979839 2024] [authz_host:error] [pid 2753:tid 281472984609056] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/defaults-white.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:03.026289 2024] [authz_host:error] [pid 2753:tid 281473035333920] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/nc-button.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:03.047246 2024] [authz_host:error] [pid 2753:tid 281472967700768] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/logs.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:03.050720 2024] [authz_host:error] [pid 2753:tid 281472976154912] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/info-white.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:03.051746 2024] [authz_host:error] [pid 2753:tid 281473018425632] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Sun May 05 02:25:03.074655 2024] [authz_host:error] [pid 2753:tid 281472959246624] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/files.svg failed, reason: unable to get the remote host name, referer: https://192.168.0.9:4443/css/ncp.css
[Sun May 05 02:25:03.268604 2024] [authz_host:error] [pid 2753:tid 281472916975904] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Sun May 05 02:25:03.475073 2024] [authz_host:error] [pid 2753:tid 281472916975904] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/favicon.png failed, reason: unable to get the remote host name
[Sun May 05 02:25:13.959725 2024] [authz_host:error] [pid 2753:tid 281472908521760] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.161492 2024] [authz_host:error] [pid 2753:tid 281472908521760] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.196487 2024] [authz_host:error] [pid 2753:tid 281473161736480] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/delete.svg failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.196487 2024] [authz_host:error] [pid 2753:tid 281472900067616] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.197127 2024] [authz_host:error] [pid 2753:tid 281472933884192] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/download.svg failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.414139 2024] [authz_host:error] [pid 2753:tid 281472900067616] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.416801 2024] [authz_host:error] [pid 2753:tid 281473161736480] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/delete.svg failed, reason: unable to get the remote host name
[Sun May 05 02:25:14.427639 2024] [authz_host:error] [pid 2753:tid 281472933884192] [remote 192.168.0.54:55343] AH01753: access check of 'localhost' to /img/download.svg failed, reason: unable to get the remote host name

```
</details>

<details>
<summary>Database logs</summary>

```

tail: impossible d'ouvrir '/var/log/mysql/*.log' en lecture: Aucun fichier ou dossier de ce type

```
</details>

theCalcaholic, 2024-05-05:
Hi, can you give me the output of `systemctl status mariadb`?

zedocrob, 2024-05-05:
Hi, sure

```
● mariadb.service - MariaDB 10.11.6 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/mariadb.service.d
             └─ncp-delay-automount.conf
     Active: active (running) since Sun 2024-05-05 02:23:51 CEST; 14h ago
       Docs: man:mariadbd(8)
             https://mariadb.com/kb/en/library/systemd/
    Process: 3706 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
    Process: 3707 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
    Process: 3709 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && syst>
    Process: 3789 ExecStartPre=/bin/sleep 20 (code=exited, status=0/SUCCESS)
    Process: 3834 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
    Process: 3839 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
   Main PID: 3822 (mariadbd)
     Status: "Taking your SQL requests now..."
      Tasks: 9 (limit: 4133)
     Memory: 367.3M
        CPU: 12.375s
     CGroup: /system.slice/mariadb.service
             └─3822 /usr/sbin/mariadbd

mai 05 02:23:51 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:51 0 [Warning] 'innodb-file-format' was removed. It does nothing now and exists only for compatibi>
mai 05 02:23:51 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:51 0 [Warning] 'innodb-buffer-pool-instances' was removed. It does nothing now and exists only for>
mai 05 02:23:51 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:51 0 [Warning] You need to use --log-bin to make --expire-logs-days or --binlog-expire-logs-second>
mai 05 02:23:51 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:51 0 [Note] Server socket created on IP: '127.0.0.1'.
mai 05 02:23:51 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:51 0 [Note] /usr/sbin/mariadbd: ready for connections.
mai 05 02:23:51 nextcloudpi mariadbd[3822]: Version: '10.11.6-MariaDB-0+deb12u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 12
mai 05 02:23:51 nextcloudpi systemd[1]: Started mariadb.service - MariaDB 10.11.6 database server.
mai 05 02:23:52 nextcloudpi /etc/mysql/debian-start[3861]: Checking for insecure root accounts.
mai 05 02:23:52 nextcloudpi /etc/mysql/debian-start[3865]: Triggering myisam-recover for all MyISAM tables and aria-recover for all Aria tables
mai 05 02:23:55 nextcloudpi mariadbd[3822]: 2024-05-05  2:23:55 0 [Note] InnoDB: Buffer pool(s) load completed at 240505  2:23:55
```

theCalcaholic, 2024-05-06:
@zedocrob I have the suspicion that you have either a wrong version of PHP installed or your PHP installation broke during the upgrade. Can you give me the output to the following commands?

```bash
php --version
php -m
```

zedocrob, 2024-05-06:
@theCalcaholic Thanks for your help, but in the meantime i've changed my SDcard for a newer one and restored everything fine after flashing the latest image 1.54.0.
We'll nerver know if it was a broken PHP but i think your suspicion was probably right.

theCalcaholic, 2024-05-06:
Alright, then I'll close this issue :)

closed 2024-05-06 by theCalcaholic