nextcloud/nextcloudpi #1926: Unit systemd-resolved.service could not be found

Schniefel, 2024-05-09:
When trying to install nextcloudpi via the curl-script on a freshly flashed Raspberry Pi OS (Bookworm) i get the following error:

Downloading updates
Performing updates
PHP Warning:  Undefined variable $CONFIG in Command line code on line 1
PHP Warning:  Trying to access array offset on value of type null in Command line code on line 1
PHP Warning:  Undefined variable $CONFIG in Command line code on line 1
PHP Warning:  Trying to access array offset on value of type null in Command line code on line 1
Installing clear-php-opcache
Installing dnsmasq
+ apt-get update
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease                          
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease        
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Hit:5 https://packages.sury.org/php bookworm InRelease
Reading package lists... Done
+ apt-get install --no-install-recommends -y dnsmasq
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libraspberrypi0 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  resolvconf
The following NEW packages will be installed:
  dnsmasq
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 32.4 kB of archives.
After this operation, 120 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 dnsmasq all 2.89-1 [32.4 kB]
Fetched 32.4 kB in 0s (567 kB/s)   
Selecting previously unselected package dnsmasq.
(Reading database ... 246159 files and directories currently installed.)
Preparing to unpack .../dnsmasq_2.89-1_all.deb ...
Unpacking dnsmasq (2.89-1) ...
Setting up dnsmasq (2.89-1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/dnsmasq.service â†’ /lib/systemd/system/dnsmasq.service.
+ rc=0
+ service dnsmasq status
+ [[ 0 -eq 3 ]]
+ echo 'Applying workaround for dnsmasq bug (compare issue #1446)'
Applying workaround for dnsmasq bug (compare issue #1446)
+ mkdir -p /etc/systemd/resolved.conf.d
+ systemctl status systemd-resolved
**Unit systemd-resolved.service could not be found.
+ systemctl stop resolvconf
Failed to stop resolvconf.service: Unit resolvconf.service not loaded.**

Schniefel, 2024-05-10:
Update: I was able to install nextcloudpi by running apt install systemd-resolved before running the script

vcbranco, 2024-05-11:
Is also a problem with the curl installation on Debian 12 x86_64.

vcbranco, 2024-05-11:
In my case, all I had to do was install systemd-resolved, configure the dns server address and then run the script.
The curl installation was done correctly.

Schniefel, 2024-05-11:
I haven't even configured a dns-server.

vcbranco, 2024-05-11:
I had to configure the address of the dns server because after installing systemd-resolved I restarted the server and was left without dns.
It probably wouldn't have been necessary to configure the dns if I ran the script without restarting the server after installing the package. But I haven't tested that.

Schniefel, 2024-05-11:
right, a restart fixed it for me

darguez, 2024-05-13:
When I tried to install nextcloudpi on my Raspberry Pi OS Lite 64-bit Debian 12 (bookworm) via curl script:
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash`
I got the same error: **_Unit systemd-resolved.service could not be found_**

So, I manually installed the missing package `apt install systemd-resolved` but I missed DNS with the change and I had to manually add DNS in `/etc/systemd/resolved.conf`. Also I had to drop the ncp created mysql database `sudo mysqladmin drop nextcloud` to start over with ncp installation script.

Then the installation went on, but finished with a weird sql error (not sure if the script finished at all): **_Table "nextcloud.appconfig" doesn't exist_**

However, I checked ncp installation by accessing https://my-ip/ and the web installer was working, so after the initial setup everything seemed to work fine.

The only thing that wasn't work properly was letsencrypt, because after enabling it (without errors), the certificate was not updated, so I had to edit apache2 site config file to set the correct paths to the certificates in `/etc/apache2/sites-enabled/001-nextcloud.conf`.

And now everything seems to work fine.

darguez, 2024-05-13:
> And now everything seems to work fine.

Well, I've just realized the HPB service is always down, so something went wrong with installation after all... Also there is no `/var/www/nextcloud/apps/notify_push` folder and command `/occ notify_push:setup` returns:  

> **_There are no commands defined in the "notify_push" namespace._**

darguez, 2024-05-13:
> Well, I've just realized the HPB service is always down, so something went wrong with installation after all... Also there is no `/var/www/nextcloud/apps/notify_push` folder and command `/occ [...]

It seems "Client Push" app was not installed, so after installing it from appstore and executing `occ notify_push:setup` the service is up now

vcbranco, 2024-05-29:
You can configure the systemd-resolved to have DNS.
https://unix.stackexchange.com/questions/442598/how-to-configure-systemd-resolved-and-systemd-networkd-to-use-local-dns-server-f

ovpc, 2024-06-05:
Ran curl installer again to get and save full output
System is running in Proxmox VM
I used debian-12.5-amd64-netinst.iso installed basic system tools and ssh. After reboot installed curl net-tools and sudo, before running installer.
[ncp-installer-log.txt](https://github.com/user-attachments/files/15584912/ncp-installer-log.txt)

m-breitbach, 2024-08-06:
Encountering same problem on fresh install (RaspberryPi OS Lite, arm64, Bookworm). Any plans to fix the install script?

e8johan, 2024-08-24:
Can confirm on a fresh version of Raspberry Pi OS Lite (64-bit), release date 2024-07-04.

e8johan, 2024-08-25:
I had another go tonight. Started from a fresh RaspberryPi OS Lite (64-bit) image, then applies the following tricks to covert to systemd-networkd (over wlan0): https://raspberrypi.stackexchange.com/questions/108592/use-systemd-networkd-for-general-networking . Finally, I ran the nextcloudpi install script and ran into the following (new) error:

```
Setting up unattended-upgrades (2.9.1+nmu3) ...
locale: Cannot set LC_ALL to default locale: No such file or directory

Creating config file /etc/apt/apt.conf.d/20auto-upgrades with new version

Creating config file /etc/apt/apt.conf.d/50unattended-upgrades with new version
Created symlink /etc/systemd/system/multi-user.target.wants/unattended-upgrades.service → /lib/systemd/system/unattended-upgrades.service.
Synchronizing state of unattended-upgrades.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable unattended-upgrades
Processing triggers for man-db (2.11.2-2) ...
Running unattended-upgrades
Unattended upgrades active: yes (autoreboot true)
NextcloudPi updated to version v1.54.3
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libavahi-core7 libdaemon0
Suggested packages:
  avahi-autoipd
Recommended packages:
  libnss-mdns
The following NEW packages will be installed:
  avahi-daemon libavahi-core7 libdaemon0
The following held packages will be changed:
  avahi-daemon
0 upgraded, 3 newly installed, 0 to remove and 1 not upgraded.
E: Held packages were changed and -y was used without --allow-change-held-packages.
```
I can see where the holding comes from, but also got the impression that systemd-resolved and avahi-daemon sort of, kind of, did the same job, so I was surprised to see the mix of avahi-daemon and systemd-resolved.

e8johan, 2024-08-25:
So, third try it is. I did exactly what I did above, except holding avahi-daemon. Now it seems to work. I'm activating NextCloudPi right now. :-)

theCalcaholic, 2024-08-26:
@e8johan Glad to hear you made it work. 

Out of curiosity: is there a reason why using [the official NCP image for RPi 5](https://github.com/nextcloud/nextcloudpi/releases) was not an option?

e8johan, 2024-08-26:
I guess two parts: first reason is that I did that last time installing NextCloudPi (on a 32-bit install on a Pi3), second reason is the feeling of control - having the user and ssh keys that I'm used to from start, instead of having to add that afterwards for that seamless ssh admin experience ;-)

ksd73429udmn, 2024-09-01:
Are you still planning to make the curl script work some day? It would also help me a lot.

theCalcaholic, 2024-09-04:
@ksd73429udmn I do, after NC 30 support

theCalcaholic, 2024-09-05:
@HervePruvost There's an approximate timeline for the NC29 update (should arrive until the end of next week). Other than that, I can't give an ETA

wulfsystems, 2024-10-06:
Hi guys,
is there something new?

Schniefel, 2024-10-24:
Nextcloud 30 is available, when can we expect support for it and this bug fix?

remz1337, 2024-11-06:
@theCalcaholic I tested the latest patch and issue persist. Script fails after installing `systemd-resolved`

Logs:
```
The following NEW packages will be installed:
  systemd-resolved
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 307 kB of archives.
After this operation, 833 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main amd64 systemd-resolved amd64 252.30-1~deb12u2 [307 kB]
Fetched 307 kB in 0s (3,646 kB/s)    
Selecting previously unselected package systemd-resolved.
(Reading database ... 24133 files and directories currently installed.)
Preparing to unpack .../systemd-resolved_252.30-1~deb12u2_amd64.deb ...
Unpacking systemd-resolved (252.30-1~deb12u2) ...
Setting up systemd-resolved (252.30-1~deb12u2) ...
Converting /etc/resolv.conf to a symlink to /run/systemd/resolve/stub-resolv.conf...
Creating group 'systemd-resolve' with GID 996.
Creating user 'systemd-resolve' (systemd Resolver) with UID 996 and GID 996.
Created symlink /etc/systemd/system/dbus-org.freedesktop.resolve1.service → /lib/systemd/system/systemd-resolved.service.
Created symlink /etc/systemd/system/sysinit.target.wants/systemd-resolved.service → /lib/systemd/system/systemd-resolved.service.
Processing triggers for dbus (1.14.10-1~deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
Installing nc-nextcloud
Ign:1 http://security.debian.org bookworm-security InRelease
Ign:2 https://packages.sury.org/php bookworm InRelease             
Ign:3 http://deb.debian.org/debian bookworm InRelease              
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Ign:3 http://deb.debian.org/debian bookworm InRelease
Ign:1 http://security.debian.org bookworm-security InRelease            
Ign:2 https://packages.sury.org/php bookworm InRelease                  
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Ign:1 http://security.debian.org bookworm-security InRelease
Ign:3 http://deb.debian.org/debian bookworm InRelease              
Ign:2 https://packages.sury.org/php bookworm InRelease             
Ign:4 http://deb.debian.org/debian bookworm-updates InRelease
Err:2 https://packages.sury.org/php bookworm InRelease
  Temporary failure resolving 'packages.sury.org'
Err:3 http://deb.debian.org/debian bookworm InRelease                
  Temporary failure resolving 'deb.debian.org'
Err:1 http://security.debian.org bookworm-security InRelease         
  Temporary failure resolving 'security.debian.org'
Err:4 http://deb.debian.org/debian bookworm-updates InRelease
  Temporary failure resolving 'deb.debian.org'
```

And here's the content of /etc/resolv.conf:
```
# This is /run/systemd/resolve/stub-resolv.conf managed by man:systemd-resolved(8).
# Do not edit.
#
# This file might be symlinked as /etc/resolv.conf. If you're looking at
# /etc/resolv.conf and seeing this text, you have followed the symlink.
#
# This is a dynamic resolv.conf file for connecting local clients to the
# internal DNS stub resolver of systemd-resolved. This file lists all
# configured search domains.
#
# Run "resolvectl status" to see details about the uplink DNS servers
# currently in use.
#
# Third party programs should typically not access this file directly, but only
# through the symlink at /etc/resolv.conf. To manage man:resolv.conf(5) in a
# different way, replace this symlink by a static file or a different symlink.
#
# See man:systemd-resolved.service(8) for details about the supported modes of
# operation for /etc/resolv.conf.

nameserver 127.0.0.53
options edns0 trust-ad
search .
```

theCalcaholic, 2024-11-06:
@remz1337 A fix for this is in devel. Can you try running the installer like this?

```bash
BRANCH=devel DBG=x bash <(curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh)
```

theCalcaholic, 2024-11-08:
Should be fixed with v1.55.3

closed 2024-11-08 by theCalcaholic

remz1337, 2024-11-08:
Tested and can confirm the issue is now fixed. Thanks

PR #2000 (2024-11-06): NC 30 + fixes
Fixes #1992 
Fixes #1926
commit a3ccf157 (2024-11-06): Merge pull request #2000 from nextcloud/fix/networking-raspbian

- Add NC 30 support
- Fixes #1992
- Fixes #1926