nextcloud/nextcloudpi #1927: Upgrade to ncp 1.54.2 stuck on fresh installed RPi5 1.54.0 image

lovenemesis, 2024-05-09:
**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.0
NextcloudPi image    
OS                   Armbian-unofficial 24.2.1 Bookworm \l . 6.6.18-current-bcm2712 (aarch64)
automount            yes
USB devices          none
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.8G/908G
rootfs usage         4.8G/908G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            end0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               9:00

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "202.117.147.60",
            "1": "192.168.199.216",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "28.0.5.1",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/opt\/ncdata\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/opt\/ncdata\/data\/nextcloud.log",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***"
    }
}

```
</details>

After setup wizard using RPi5 1.54.0 image, I was greeted with an upgrade prompt to ncp 1.54.2 from ssh session. However, the upgrade process seems to stuck somewhere after installing some updates:

```
$ sudo ncp-update
[sudo] password for pi: 
Downloading updates
Performing updates
Update root login prevention method...
done.
Fixing trusted proxies list...
done.
Updating PHP package signing key...
Hit:1 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Get:2 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease [56.5 kB]
Get:4 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease [48.0 kB]
Hit:5 https://mirrors.ustc.edu.cn/armbian bookworm InRelease
Get:6 https://mirrors.ustc.edu.cn/debian-security bookworm-security/main arm64 Packages [155 kB]
Hit:7 https://packages.sury.org/php bookworm InRelease                      
Fetched 315 kB in 2s (166 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnupg2 is already the newest version (2.2.40-1.1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.kcqZ0rygpb/gpg.1.sh --fetch-keys https://packages.sury.org/php/apt.gpg
gpg: requesting key from 'https://packages.sury.org/php/apt.gpg'
gpg: key B188E2B695BD4743: "DEB.SURY.ORG Automatic Signing Key <deb@sury.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
done.
Installing dependencies...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
tmux is already the newest version (3.3a-3).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
done.
Updating obsolete theming URL
done.
NextCloudPi updated to version v1.54.0
Reenable erroneously disabled package sources
done
Hit:1 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Hit:2 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
Hit:3 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
Hit:4 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease
Hit:5 https://mirrors.ustc.edu.cn/armbian bookworm InRelease
Hit:6 https://packages.sury.org/php bookworm InRelease      
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  zstd
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/584 kB of archives.
After this operation, 1,956 kB of additional disk space will be used.
Selecting previously unselected package zstd.
(Reading database ... 39207 files and directories currently installed.)
Preparing to unpack .../zstd_1.5.4+dfsg2-5_arm64.deb ...
Unpacking zstd (1.5.4+dfsg2-5) ...
Setting up zstd (1.5.4+dfsg2-5) ...
Processing triggers for man-db (2.11.2-2) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libdav1d6
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 275 kB of archives.
After this operation, 65.5 kB of additional disk space will be used.
Get:1 https://mirrors.ustc.edu.cn/debian-security bookworm-security/main arm64 libdav1d6 arm64 1.0.0-2+deb12u1 [275 kB]
Fetched 275 kB in 0s (929 kB/s)
```

It's a fresh installation with only change on debian mirrors. I tried the same upgrade on my existing RPi3 instance running Debian Bullseys. ncp managed to upgrade itself to 1.54.2 without issue there.

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.54.2
NextcloudPi image    NextCloudPi_03-02-22
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          none
datadir              /mnt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      223G/361G
rootfs usage         223G/361G
swapfile             /swapfile
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              closed
Port 443             closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               9:22

```
</details>

Looking through the commit history, it appears the 1.54.1 and 1.54.2 were focused on solving issues on `ncp-dist-upgrade`. Should I hold on the upgrade on RPi5 image since it's already on Debian Bookworm?

lovenemesis, 2024-05-10:
Never mind.

I re-flashed the image again and gave the update another try. It worked.

Perhaps a reboot before making any further configuration would be a good practice.

closed 2024-05-10 by lovenemesis

xa-y-ax, 2024-07-04:
Same Problem here. Still after re-flash and reboot after activating ssh. What was your solution?

Ts451, 2024-07-04:
same here.

i did a fresh install, enabled ssh, did "sudo apt update" and "sudo apt upgrade" first, then rebooted and then did the "sudo ncp-upgrade." 

it took a few minutes to complete, then did one final reboot.

dwprod38, 2024-07-07:
Same problem with fresh RPi5 install from SD flashed with "https://github.com/nextcloud/nextcloudpi/releases/download/v1.54.0/NextcloudPi_RaspberryPi5_v1.54.0.zip"

After activation page and SSH : reboot of RPi5 (takes 10min !!!! To get prompt login !!!

And launching "sudo ncp-update", run until "Processing Triggers for man-db (2.11,2-2)" and nothing happening during 24h...

[image]