nextcloud/nextcloudpi #1928: Admin Settings result in Internal Server Error

Schmiddiii, 2024-05-11:
Opening the admin settings `/settings/admin` results in an internal server error for me. All other settings pages seem to work.

Looking at the NextCloud logs provided below, the error seems to be in [SettingsService.runCommand](https://github.com/nextcloud/nextcloudpi/blob/0c023d95a5f5c51a5e036751ef53f58ed2da4eb0/ncp-app/lib/Service/SettingsService.php#L128).

Note that I don't know PHP well, so my analysis may be wrong. I will nevertheless try:

- `AdminSettings.getForm` requests the current settings via `SettingsService.getConfig` with argument `ncp`.
- `SettingsService.getConfig` calls the `runCommand` that is failing with command `bash -c \"sudo /home/www/ncp-app-bridge.sh config ncp`.
    - I checked and the file `/home/www/ncp-app-bridge.sh` exists. I did not execute it, but reading the source code it just echos `/usr/local/etc/ncp.cfg` (which I also included below, but think it is not relevant; also note that the `php_version` field has the wrong PHP version, but I also don't think this is an issue).
    - The file has permission `-rwx------ 1 root root 964 Jan 21 06:26 /home/www/ncp-app-bridge.sh` which I think means it can only be executed by root. As the command is called with `sudo`, I don't think this is the problem.
    - The folder `/home/www` has `drwx------ 2 www-data www-data 4.0K May 11 12:24` which I also don't think should be the problem.
- `SettingsService.runCommand` uses an argument `$pipes`, which I would guess is `null` for some reason looking at the logs.

I have recently updated my PHP to 8.3, so maybe something changed in that regard. Also, maybe the file permissions are wrong somehow.

I am unsure if this issue just affects myself or if others are also affected. As I think this is more of a bug instead of a support request, I opened the issue here instead of asking on the forum, feel free to move it to the forum if I am wrong in this regard.

**System information**

<details>
<summary>`ncp.cfg`</summary

{
  "nextcloud_version": "28.0.5",
  "php_version": "8.1",
  "release": "bullseye"
}
</details>

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.2
NextcloudPi image    NextCloudPi_07-21-19
OS                   Raspbian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      223G/466G
rootfs usage         6.9G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               30min

```
</details>

[... 132 lines cut ...]
```

{"reqId":"abJRi0OvmtKTwVtKhpB5","level":1,"time":"2024-05-11T09:50:56+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"abJRi0OvmtKTwVtKhpB5","level":1,"time":"2024-05-11T09:50:56+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"28.0.5.1","data":{"app":"updater"}}
{"reqId":"Zj9LCmW8dg24evQNEU1TPAAAAAo","level":3,"time":"2024-05-11T10:40:11+00:00","remoteAddr":"192.168.2.196","user":"censored-user","app":"PHP","method":"GET","url":"/settings/admin","message":"proc_open(): posix_spawn() failed: No such file or directory at /var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php#135","userAgent":"Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) QtWebEngine/6.7.0 Chrome/118.0.5993.220 Safari/537.36","version":"28.0.5.1","data":{"app":"PHP"}}
{"reqId":"Zj9LCmW8dg24evQNEU1TPAAAAAo","level":3,"time":"2024-05-11T10:40:11+00:00","remoteAddr":"192.168.2.196","user":"censored-user","app":"PHP","method":"GET","url":"/settings/admin","message":"Trying to access array offset on null at /var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php#136","userAgent":"Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) QtWebEngine/6.7.0 Chrome/118.0.5993.220 Safari/537.36","version":"28.0.5.1","data":{"app":"PHP"}}
{"reqId":"Zj9LCmW8dg24evQNEU1TPAAAAAo","level":3,"time":"2024-05-11T10:40:11+00:00","remoteAddr":"192.168.2.196","user":"censored-user","app":"index","method":"GET","url":"/settings/admin","message":"stream_get_contents(): Argument #1 ($stream) must be of type resource, null given in file '/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php' line 136","userAgent":"Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) QtWebEngine/6.7.0 Chrome/118.0.5993.220 Safari/537.36","version":"28.0.5.1","exception":{"Exception":"Exception","Message":"stream_get_contents(): Argument #1 ($stream) must be of type resource, null given in file '/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php' line 136","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":184,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":315,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1069,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":39,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","Line":169,"Previous":{"Exception":"TypeError","Message":"stream_get_contents(): Argument #1 ($stream) must be of type resource, null given","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php","line":136,"function":"stream_get_contents"},{"file":"/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php","line":30,"function":"runCommand","class":"OCA\\NextcloudPi\\Service\\SettingsService","type":"->"},{"file":"/var/www/nextcloud/apps/nextcloudpi/lib/Settings/AdminSettings.php","line":27,"function":"getConfig","class":"OCA\\NextcloudPi\\Service\\SettingsService","type":"->"},{"file":"/var/www/nextcloud/apps/settings/lib/Controller/CommonSettingsTrait.php","line":129,"function":"getForm","class":"OCA\\NextcloudPi\\Settings\\AdminSettings","type":"->"},{"file":"/var/www/nextcloud/apps/settings/lib/Controller/AdminSettingsController.php","line":86,"function":"formatSettings","class":"OCA\\Settings\\Controller\\AdminSettingsController","type":"->"},{"file":"/var/www/nextcloud/apps/settings/lib/Controller/CommonSettingsTrait.php","line":149,"function":"getSettings","class":"OCA\\Settings\\Controller\\AdminSettingsController","type":"->"},{"file":"/var/www/nextcloud/apps/settings/lib/Controller/AdminSettingsController.php","line":71,"function":"getIndexResponse","class":"OCA\\Settings\\Controller\\AdminSettingsController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":230,"function":"index","class":"OCA\\Settings\\Controller\\AdminSettingsController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":137,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":184,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":315,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1069,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":39,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/apps/nextcloudpi/lib/Servic [... cut ...] pe":"::"}],"File":"/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php","Line":136},"message":"stream_get_contents(): Argument #1 ($stream) must be of type resource, null given in file '/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php' line 136","exception":{},"CustomMessage":"stream_get_contents(): Argument #1 ($stream) must be of type resource, null given in file '/var/www/nextcloud/apps/nextcloudpi/lib/Service/SettingsService.php' line 136"}}

```
</details>

theCalcaholic, 2024-08-19:
> I have recently updated my PHP to 8.3

That's what's causing your issue. PHP 8.3 is not yet supported by NCP and you should never manually update PHP versions unless you are prepared to break and restore your system.

closed 2024-08-19 by theCalcaholic

bendschs, 2024-11-09:
i am get the same errore since last updating nextcloud and i did not recently update my php, still on 8.1.

theCalcaholic, 2024-11-10:
@bendschs By now, PHP 8.3 *is* supported and will automatically installed when upgrading Nextcloud beyond version 28.

What's NC version are you running?

bendschs, 2024-11-10:
thanks for trying to help, i think i figured it out: it was my "passwords app" that caused it. upgraded it and now it's working again.