nextcloud/nextcloudpi #1930: ncp-dist-upgrade to bookworm failed on Rock64 ncp 1.54.2

ff0255, 2024-05-12:
Hello.
Thank you for all the work on this project.
Today I tried ncp-dist-upgrade to bookworm but it failed.
Last part of upgrade output:
```
...
Unpacking libxml2:arm64 (2.9.14+dfsg-0.1+0~20230421.14+debian11~1.gbpf14485) over (2.9.10+dfsg-6.7+deb11u4) ...
Setting up libmpdec3:arm64 (2.5.1-2+0~20240210.5+debian11~1.gbp216c03) ...
Setting up libxml2:arm64 (2.9.14+dfsg-0.1+0~20230421.14+debian11~1.gbpf14485) ...
Processing triggers for libc-bin (2.31-13+deb11u10) ...
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://security.debian.org bookworm-security InRelease [48.0 kB]
Get:3 https://packages.sury.org/php bookworm InRelease [7,542 B]
Get:4 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:5 http://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Get:6 http://armbian.tnahosting.net/apt bookworm InRelease [53.3 kB]
Get:7 http://security.debian.org bookworm-security/main armhf Packages [149 kB]
Get:8 http://security.debian.org bookworm-security/main arm64 Packages [152 kB]
Get:9 http://deb.debian.org/debian bookworm/main armhf Packages [11.7 MB]
Get:10 https://packages.sury.org/php bookworm/main armhf Packages [385 kB]
Get:11 http://deb.debian.org/debian bookworm/contrib arm64 Packages [54.7 kB]
Get:12 http://deb.debian.org/debian bookworm/non-free arm64 Packages [94.4 kB]
Get:13 http://deb.debian.org/debian bookworm/contrib armhf Packages [50.7 kB]
Get:14 http://deb.debian.org/debian bookworm/non-free armhf Packages [68.4 kB]
Get:15 http://deb.debian.org/debian bookworm/main arm64 Packages [11.9 MB]
Get:16 https://packages.sury.org/php bookworm/main arm64 Packages [388 kB]
Get:17 http://deb.debian.org/debian bookworm-updates/non-free armhf Packages [492 B]
Get:18 http://deb.debian.org/debian bookworm-updates/non-free arm64 Packages [12.0 kB]
Get:19 http://deb.debian.org/debian bookworm-updates/contrib arm64 Packages [768 B]
Get:20 http://deb.debian.org/debian bookworm-updates/main armhf Packages [13.2 kB]
Get:21 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [13.7 kB]
Get:22 http://deb.debian.org/debian bookworm-backports/non-free armhf Packages [1,288 B]
Get:23 http://deb.debian.org/debian bookworm-backports/main armhf Packages [194 kB]
Get:24 http://deb.debian.org/debian bookworm-backports/non-free arm64 Packages [1,288 B]
Get:25 http://deb.debian.org/debian bookworm-backports/contrib arm64 Packages [4,688 B]
Get:26 http://deb.debian.org/debian bookworm-backports/main arm64 Packages [190 kB]
Get:27 http://deb.debian.org/debian bookworm-backports/contrib armhf Packages [4,692 B]
Get:28 http://armbian.tnahosting.net/apt bookworm/bookworm-utils armhf Packages [18.4 kB]
Get:29 http://armbian.tnahosting.net/apt bookworm/main all Packages [4,650 B]
Get:30 http://armbian.tnahosting.net/apt bookworm/bookworm-desktop arm64 Packages [2,952 B]
Get:31 http://armbian.tnahosting.net/apt bookworm/bookworm-desktop all Packages [933 B]
Get:32 http://armbian.tnahosting.net/apt bookworm/main arm64 Packages [188 kB]
Get:33 http://armbian.tnahosting.net/apt bookworm/bookworm-utils all Packages [3,068 B]
Get:34 http://armbian.tnahosting.net/apt bookworm/bookworm-desktop armhf Packages [2,983 B]
Get:35 http://armbian.tnahosting.net/apt bookworm/main armhf Packages [86.0 kB]
Get:36 http://armbian.tnahosting.net/apt bookworm/bookworm-utils arm64 Packages [21.1 kB]
Fetched 26.1 MB in 7s (3,986 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc-dev-bin : Depends: libc6 (< 2.32) but 2.36-9+deb12u7 is to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+deb11u10) but 2.36-9+deb12u7 is to be installed
E: Broken packages
```

I tried manual upgrade (https://github.com/nextcloud/nextcloudpi/issues/1913), but after `apt-get upgrade -y dpkg` got the same error.
After that I manually changed back to bullseye everything in sources.list
How to fix those packages?

**System information**

[... 146 lines cut ...]
```

{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Add background job to cleanup login flow v2 tokens","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Remove potentially over exposing share links","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairInfoEvent: Repair info: No need to remove link shares.","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Clear access cache of projects","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Reset generated avatar flag","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Keep legacy encryption enabled","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Check encryption key format","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Remove old dashboard app config data","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Repair DAV shares","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Add background job to set the lookup server share state for users","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:20:54+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"OC\\Repair\\Events\\RepairStepEvent: Repair step: Add token cleanup job","userAgent":"--","version":"24.0.12.1","data":{"app":"updater"}}
[... cut ...]
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:21:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"25.0.13.2","data":{"app":"updater"}}
{"reqId":"ZYVB4oMaMFMTui2lt9xh","level":1,"time":"2024-05-11T01:21:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"25.0.13.2","data":{"app":"updater"}}

```
</details>

pairatec, 2024-05-15:
I have the same issue with my RaspberryPI 4b+

I wanted to upgrade my OS from bullseye to bookworm via sudo ncp-dist-upgrade on my Raspberry PI 4B+. However, the script terminates with the following error:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc-dev-bin : Depends: libc6 (< 2.32) but 2.36-9+rpt2+deb12u7 is to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+rpt2+rpi1+deb11u10) but 2.36-9+rpt2+deb12u7 is to be installed
```

I tried upgrading the dependencies with the following commands (listed below), but after that, i could not execute sudo ncp-dist-ugprade anymore, since it was already upgraded to Bookworm. But after switching the SD from my Raspberry PI 4 to a Raspberry PI 5, i got "No bootable partitions" so I believe the dist-upgrade was not finished?
```
sudo apt-get update
sudo apt-get upgrade libc-dev-bin libc6-dev
```

**System information**
This is the log after running ncp-dist-upgrade the first time
<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.2
NextcloudPi image    NextcloudPi_12-27-23
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda sdb 
datadir              /media/clouddrive/ncpdata
data in SD           no
data filesystem      btrfs
data disk usage      67G/932G
rootfs usage         6.7G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    27.1.7.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               4min

```
</details>

<details>
[... 91 lines cut ...]
[Tue May 14 23:24:27.027880 2024] [core:warn] [pid 35443:tid 548634046528] AH00045: child process 35445 still did not exit, sending a SIGTERM
[Tue May 14 23:24:29.030168 2024] [core:error] [pid 35443:tid 548634046528] AH00046: child process 35445 still did not exit, sending a SIGKILL
[Tue May 14 23:24:38.683906 2024] [ssl:warn] [pid 548:tid 547907891264] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 23:24:38.693169 2024] [ssl:error] [pid 548:tid 547907891264] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspi5 / issuer: CN=raspi5 / serial: 74CC02906A0698300FBBC982FCE9664B3B17BEB4 / notbefore: Dec 26 21:46:54 2023 GMT / notafter: Dec 23 21:46:54 2033 GMT]
[Tue May 14 23:24:38.693253 2024] [ssl:error] [pid 548:tid 547907891264] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue May 14 23:24:38.777358 2024] [ssl:warn] [pid 852:tid 547907891264] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 23:24:38.777779 2024] [ssl:error] [pid 852:tid 547907891264] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspi5 / issuer: CN=raspi5 / serial: 74CC02906A0698300FBBC982FCE9664B3B17BEB4 / notbefore: Dec 26 21:46:54 2023 GMT / notafter: Dec 23 21:46:54 2033 GMT]
[Tue May 14 23:24:38.777838 2024] [ssl:error] [pid 852:tid 547907891264] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue May 14 23:24:38.786778 2024] [mpm_event:notice] [pid 852:tid 547907891264] AH00489: Apache/2.4.59 (Debian) OpenSSL/1.1.1w configured -- resuming normal operations
[Tue May 14 23:24:38.786901 2024] [core:notice] [pid 852:tid 547907891264] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>

Schniefel, 2024-05-16:
@pairatec I had the exact same problem which i mentioned here: https://github.com/nextcloud/nextcloudpi/issues/1913

I reinstalled Raspberry pi os bookworm and installed via curl-insraller. **Note: Before running the installer run sudo apt install systemd-resolved and reboot.**

After that, simply restore your backup.

FadeFx, 2024-05-18:
Yo run NC25.x? I doubt this will work with bookworm. There is at least an issue with PHP i guess

sebeweiss, 2024-05-19:
Maybe try this: https://stackoverflow.com/questions/30451939/cant-install-libc6-package

new-on-github, 2024-05-20:
Same problem here. 
Now a cannot run sudo ncp-distro-upgrade. But NCP tells me to do so.  I tried now
`sudo apt autoclean
sudo apt clean
sudo apt update
sudo apt upgrade`

Does someone have a easy solution? I am not eager to reinstall everything. But maybe everything is ok. How to find out, if something is wrong?

Thanks a lot already!

FadeFx, 2024-05-21:
@theCalcaholic you think it is possible to check the NC Version before upgrading to bookworm? I know it ist not related to the Problems people have here, but upgrading to bookworm while running NC25.x will most likely require a lot of work to get it running because of PHP version mismatch.

theCalcaholic, 2024-05-21:
@ff0255 @pairatec You can try following the instructions in #1913  - it should complete the dist-upgrade process.

However @pairatec moving your existing RPi 4 installation to a RPi 5 is unsupported and won't work with or without the dist-upgrade.

theCalcaholic, 2024-05-21:
Regarding the original issue, the best I can probably do is to add an `apt-get autoclean` before the dist-upgrade. Let's hope that this fixes the issue - however, I can't really test and verify this.

closed 2024-05-21 by theCalcaholic

janvlug, 2024-05-25:
I ran also into this issue today:
```
Calculating upgrade...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc-dev-bin : Depends: libc6 (< 2.32) but 2.36-9+rpt2+deb12u7 is to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+rpt2+rpi1+deb11u10) but 2.36-9+rpt2+deb12u7 is to be installed
E: Broken packages
```
Maybe this issue should be re-opened. I checked open issues before doing the `ncp-dist-upgrade`, because I did not want to run into trouble. But I missed this closed issue.

janvlug, 2024-05-25:
I tried to follow the instructions at the top issue #1913 to finalize the upgrade.
I forgot to put Nextcloud into maintenance mode before trying the fix. Maybe and instruction to enable maintenance mode  should be added to #1913.

Later I tried to set the NextcloudPi to maintenance mode, but this resulted now in this error:
```
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20210902/redis.so (/usr/lib/php/20210902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20210902/redis.so.so (/usr/lib/php/20210902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
An unhandled exception has been thrown:
OCP\HintException: [0]: Memcache OC\Memcache\Redis not available for local cache (Is the matching PHP module installed and enabled?)
```

Now back to what I did to fix the system:

My `/etc/apt/sources.list.d/raspi.list` file after running into the issue looked like:
```
deb http://archive.raspberrypi.org/debian/ bookworm main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bookworm main
```
So this looked like this file was correctly upgraded.
Also the `/etc/apt/sources.list.d/php.list` file looked fine after running into the issue:
```
deb https://packages.sury.org/php/ bookworm main
```
Next I did point 2 in the instructions of #1913:
```
sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs
```
I saw an error, but do not have the details unfortunately. It had to do with an unsupported compression type. It was printed in an root shell. When I did Ctrl-D from the root shell, the terminal was refreshed. And I could not copy the error message any more.
I tried to run this again:
```
deb https://packages.sury.org/php/ bookworm main
```
But I got:
```
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease                                           
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease                            
Hit:4 http://archive.raspberrypi.org/debian bookworm InRelease                                          
Hit:5 https://packages.sury.org/php bookworm InRelease                                                  
Reading package lists... Done
E: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 1211494 (unattended-upgr)
N: Be aware that removing the lock file is not a solution and may break your system.
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
```
Top shows that `unattended-upgr` is using 100% CPU time at the moment. So, I guess I should leave this run until it finishes.
I've no idea how long this process will take. I will provide more details in this issue later.

janvlug, 2024-05-25:
I found the error message about the compression in `/var/log/apt/term.log`:
```
dpkg-deb: error: archive '/tmp/apt-dpkg-install-nA9zZV/4-dphys-swapfile_20100506-7.1+rpt3_all.deb' uses unknown compression for member 'control.tar.zst', giving up^M
dpkg: error processing archive /tmp/apt-dpkg-install-nA9zZV/4-dphys-swapfile_20100506-7.1+rpt3_all.deb (--unpack):^M
 dpkg-deb --control subprocess returned error exit status 2^M
Preparing to unpack .../5-php8.1-igbinary_3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8_arm64.deb ...^M
Unpacking php8.1-igbinary (3.2.15-1+0~20231202.49+debian12~1.gbpe9ffe8) over (3.2.15-1+0~20231202.49+debian11~1.gbpe9ffe8) ...^M
Preparing to unpack .../6-tasksel-data_3.73_all.deb ...^M
Unpacking tasksel-data (3.73) over (3.68+deb11u1) ...^M
Preparing to unpack .../7-tasksel_3.73_all.deb ...^M
Unpacking tasksel (3.73) over (3.68+deb11u1) ...^M
Preparing to unpack .../8-libpam-runtime_1.5.2-6+rpt2+deb12u1_all.deb ...^M
Unpacking libpam-runtime (1.5.2-6+rpt2+deb12u1) over (1.4.0-9+deb11u1+rpt2) ...^M
Errors were encountered while processing:^M
 /tmp/apt-dpkg-install-nA9zZV/4-dphys-swapfile_20100506-7.1+rpt3_all.deb^M
```

janvlug, 2024-05-25:
Further update:
I killed the "hanging" `unattended-upgr` process with kill -9
I installed zst: `sudo apt install zst`.
And started this again:
```
sudo apt-get update && sudo apt-get upgrade -y --without-new-pkgs
```
The upgrade is running now...

janvlug, 2024-05-25:
I further followed the instructions of #1913 and NextcloudPi seems to be working again.

FadeFx, 2024-05-25:
I had afterwards the problem that despite nextcloud had been working normally, ncp panel was reporting that nextcloud and HCP service was down. Also the Cron jobs were not running. This was caused by PHP 8.3 being installed, which I removed and rebooted the server. Then everything was up and running again.

ff0255, 2024-06-09:
Thank you, after carefully following https://github.com/nextcloud/nextcloudpi/issues/1913 dist upgrade compleded. The only problem I had - it was exiting interactive mode (found somewhere Ctrl+D)