nextcloud/nextcloudpi #1931: RaspberryPi5 v1.54.0 - Internal Server Error

Ste-denko, 2024-05-14:
Hi All,

I installed  NextcloudPi_RaspberryPi5_v1.54.0.zip.
When logged in as ncp admin or any other user and trying to enter the settings or administrator settings from the menu (top right button) I get an info message about an internal server error (see attached screenshot).
This already happens directly after the server is newly started.
I moved the data folder to an attached usb stick. Dont know if this is important to know here.
Today I updated to v1.54.2, same problem.
When shutting down the server via terminal the server often hangs for very long and does not get to the point of shutting down, dont know if its related.

[image]

Thank you very much for everything.
Cheers

theCalcaholic, 2024-05-20:
Hi, please attach your Nextcloud logs from right after this occurs (you'll find them in the data directory in a good called nextcloud.log)

Ste-denko, 2024-05-21:
> Hi, please attach your Nextcloud logs from right after this occurs (you'll find them in the data directory in a good called nextcloud.log)

By checking the nextcloud.log file I could already help myself:
I had a Nextcloud-App installed called "W2G2".
As soon as I activate this app the above described problem occurs, once its deactivated all is fine.
So that was apparently the problem -> Dont use this app.

Thank you very much @theCalcaholic.

Cheers and thanks for all the great NextcloudPi work !!!

Ste-denko, 2024-05-21:
Done!

closed 2024-05-21 by Ste-denko