nextcloud/nextcloudpi #1932: Some packages could not be installed - sudo ncp-dist-upgrade

pairatec, 2024-05-14:
I wanted to upgrade my OS from bullseye to bookworm via sudo ncp-dist-upgrade on my Raspberry PI 4B+. However, the script terminates with the following error:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc-dev-bin : Depends: libc6 (< 2.32) but 2.36-9+rpt2+deb12u7 is to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+rpt2+rpi1+deb11u10) but 2.36-9+rpt2+deb12u7 is to be installed
```

I tried upgrading the dependencies with the following commands (listed below), but after that, i could not execute sudo ncp-dist-ugprade anymore, since it was already upgraded to Bookworm. But after switching the SD from my Raspberry PI 4 to a Raspberry PI 5, i got "No bootable partitions" so I believe the dist-upgrade was not finished?
```
sudo apt-get update
sudo apt-get upgrade libc-dev-bin libc6-dev
```

**System information**
This is the log after running ncp-dist-upgrade the first time
<--! Paste this in GitHub report -->

<details>
<summary>NextcloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.2
NextcloudPi image    NextcloudPi_12-27-23
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda sdb 
datadir              /media/clouddrive/ncpdata
data in SD           no
data filesystem      btrfs
data disk usage      67G/932G
rootfs usage         6.7G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    27.1.7.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               4min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
Nextcloud is in maintenance mode, no apps are loaded.
Commands provided by apps are unavailable.
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
	REMOVED
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "27.1.7.2",
        "overwrite.cli.url": "https:\/\/daadii.ddns.net\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "default_phone_region": "DE",
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/clouddrive\/ncpdata\/tmp",
        "mail_smtpmode": "smtp",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "loglevel": "1",
        "log_type": "file",
        "maintenance": true,
        "logfile": "\/media\/clouddrive\/ncpdata\/nextcloud.log",
        "theme": "",
        "htaccess.RewriteBase": "\/",
        "updater.release.channel": "stable",
        "default_language": "de",
        "simpleSignUpLink.shown": false,
        "app_install_overwrite": [
            "user_usage_report",
            "nextcloudpi",
            "files_downloadactivity",
            "issuetemplate",
            "previewgenerator",
            "limit_login_to_ip"
        ],
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "ssl",
        "mail_smtpauthtype": "LOGIN",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "465",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "data-fingerprint": "REMOVED"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue May 14 23:16:16.793201 2024] [ssl:warn] [pid 35443:tid 548634046528] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 23:16:16.793519 2024] [ssl:error] [pid 35443:tid 548634046528] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspi5 / issuer: CN=raspi5 / serial: 74CC02906A0698300FBBC982FCE9664B3B17BEB4 / notbefore: Dec 26 21:46:54 2023 GMT / notafter: Dec 23 21:46:54 2033 GMT]
[Tue May 14 23:16:16.793557 2024] [ssl:error] [pid 35443:tid 548634046528] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue May 14 23:16:16.806282 2024] [mpm_event:notice] [pid 35443:tid 548634046528] AH00489: Apache/2.4.59 (Debian) OpenSSL/1.1.1w configured -- resuming normal operations
[Tue May 14 23:16:16.806407 2024] [core:notice] [pid 35443:tid 548634046528] AH00094: Command line: '/usr/sbin/apache2'
[Tue May 14 23:24:19.442192 2024] [mpm_event:notice] [pid 35443:tid 548634046528] AH00492: caught SIGWINCH, shutting down gracefully
[Tue May 14 23:24:19.535630 2024] [proxy_fcgi:error] [pid 35445:tid 548617568512] [remote 192.168.1.52:8807] AH01067: Failed to read FastCGI header
[Tue May 14 23:24:19.535819 2024] [proxy_fcgi:error] [pid 35445:tid 548617568512] (104)Connection reset by peer: [remote 192.168.1.52:8807] AH01075: Error dispatching request to :4443: 
[Tue May 14 23:24:23.023291 2024] [core:warn] [pid 35443:tid 548634046528] AH00045: child process 35445 still did not exit, sending a SIGTERM
[Tue May 14 23:24:25.025594 2024] [core:warn] [pid 35443:tid 548634046528] AH00045: child process 35445 still did not exit, sending a SIGTERM
[Tue May 14 23:24:27.027880 2024] [core:warn] [pid 35443:tid 548634046528] AH00045: child process 35445 still did not exit, sending a SIGTERM
[Tue May 14 23:24:29.030168 2024] [core:error] [pid 35443:tid 548634046528] AH00046: child process 35445 still did not exit, sending a SIGKILL
[Tue May 14 23:24:38.683906 2024] [ssl:warn] [pid 548:tid 547907891264] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 23:24:38.693169 2024] [ssl:error] [pid 548:tid 547907891264] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspi5 / issuer: CN=raspi5 / serial: 74CC02906A0698300FBBC982FCE9664B3B17BEB4 / notbefore: Dec 26 21:46:54 2023 GMT / notafter: Dec 23 21:46:54 2033 GMT]
[Tue May 14 23:24:38.693253 2024] [ssl:error] [pid 548:tid 547907891264] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue May 14 23:24:38.777358 2024] [ssl:warn] [pid 852:tid 547907891264] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 23:24:38.777779 2024] [ssl:error] [pid 852:tid 547907891264] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspi5 / issuer: CN=raspi5 / serial: 74CC02906A0698300FBBC982FCE9664B3B17BEB4 / notbefore: Dec 26 21:46:54 2023 GMT / notafter: Dec 23 21:46:54 2033 GMT]
[Tue May 14 23:24:38.777838 2024] [ssl:error] [pid 852:tid 547907891264] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue May 14 23:24:38.786778 2024] [mpm_event:notice] [pid 852:tid 547907891264] AH00489: Apache/2.4.59 (Debian) OpenSSL/1.1.1w configured -- resuming normal operations
[Tue May 14 23:24:38.786901 2024] [core:notice] [pid 852:tid 547907891264] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>

pairatec, 2024-05-15:
Duplicate to #1930

theCalcaholic, 2024-05-20:
> But after switching the SD from my Raspberry PI 4 to a Raspberry PI 5

That's unsupported. The old operating system for the Raspberry Pi <=4 does not support the Raspberry Pi 5. You do have to reinstall from scratch and restore a backup.