nextcloud/nextcloudpi #1935: nc-update-nextcloud does not prevent updating to unsupported versions

HansCronau, 2024-05-20:
At the time of writing the latest version of NC supported by NCP ([1.54.0](https://github.com/nextcloud/nextcloudpi/releases/tag/v1.54.0)) is 28.0.5.
I would therefore expect `nc-update-nextcloud` to not update past this version, or at least throw a warning.
Instead it is possible to (accidentally) enter "29.0.0" and the update will just start running without warning.
(If this behavior is intended, I think documentation could be improved to better explain why this will not cause issues.)

theCalcaholic, 2024-05-21:
Well it is intended but may cause issues. If you want to update to the latest supported version, you can simply enter "0". If you enter a specific version, it will perform the upgrade as requested.

I think this is mostly a UX issue that will be resolved once I get around to rewriting the web interface, but I don't think I'll do it before that.

HansCronau, 2024-05-21:
@theCalcaholic Thanks for your comment and amazing work. 

I should probably have noted that the observation goes for both the web UI as the terminal application `ncp-config`, which I was using.

Because updating to the latest supported version (using "0") is not possible when a machine is multiple majors behind, I was manually updating it to "27.0.0", "28.0.0", "29.0.0", "Wait... That one shouldn't even be supported, right?" _Double checks release notes while update to NC29 is running._ "No, I think it _shouldn't_. So... whoops?"
But if not, why had it just started, without warning? This created  the tiniest bit of doubt for me: Did I misunderstand something about `nc-update-nextcloud`? I'm glad to learn I didn't and this is indeed mostly a UX issue.

FadeFx, 2024-07-17:
I think it is OK, that it is possible to update to unsupported versions, but probably there should be a warning if going beond latest officially supported version.

theCalcaholic, 2024-07-19:
I'll soon release an improved update mechanism that properly selects the next version that can be updated to when either using auto updates or updating manually to a version that would skip a major version update (so, if you were on v25.0.1 and would run nc-update with v28.0.7, you'd first update to the latest v25.x.x, then, when running it again, to the latest v26.x.x and so forth, until you reach your desired version).

Also, auto updates will automatically update minor versions.

That updating with 0 unexpectedly updates to a NC version past the latest supported one will not be fixed for the time being.

theCalcaholic, 2024-08-19:
The new updater is now released

closed 2024-08-19 by theCalcaholic