nextcloud/nextcloudpi #1938: Error while applying update 1.54.0. Exiting...

EnricoK18, 2024-06-05:
I am not able to update from 1.53.3 to 1.54.2.

When i perform sudo ncp-update the output is:

```
sudo ncp-update
Downloading updates
Performing updates
Update root login prevention method...
done.
Fixing trusted proxies list...
done.
Updating PHP package signing key...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Get:3 http://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Ign:4 http://ppa.launchpad.net/ondrej/php/ubuntu bullseye InRelease
Get:5 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu oracular InRelease
Hit:7 https://packages.sury.org/php bullseye InRelease
Err:8 http://ppa.launchpad.net/ondrej/php/ubuntu bullseye Release
  404  Not Found [IP: 2620:2d:4000:1::81 80]
Err:9 http://ppa.launchpad.net/ondrej/php/ubuntu oracular Release
  404  Not Found [IP: 2620:2d:4000:1::81 80]
Get:10 http://deb.debian.org/debian-security bullseye-security/main arm64 Packages [270 kB]
Get:11 http://deb.debian.org/debian-security bullseye-security/main armhf Packages [271 kB]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/ondrej/php/ubuntu bullseye Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/ondrej/php/ubuntu oracular Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Error while applying update 1.54.0. Exiting...
```

What can i do?

**System information**

<details>
<summary>NextcloudPi diagnostics</summary>

```

NextcloudPi version  v1.53.3
NextcloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 11. 6.1.21-v8+ (aarch64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      28G/29G
rootfs usage         28G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    28.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
[... 164 lines cut ...]
```

{"reqId":"AT2mscOafOJ8vYkAm3Pd","level":3,"time":"2024-05-31T10:20:34+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /kathrin/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"AT2mscOafOJ8vYkAm3Pd","level":3,"time":"2024-05-31T10:20:40+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"AT2mscOafOJ8vYkAm3Pd","level":3,"time":"2024-05-31T10:20:47+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:09+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /enrico/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:15+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /enrico/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:22+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /kathrin/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:29+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /kathrin/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:35+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"j9DOvHtSonhBmb2NGOfC","level":3,"time":"2024-05-31T10:40:42+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:14+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /enrico/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:20+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /enrico/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:27+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /kathrin/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:33+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /kathrin/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:40+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_bilder: Invalid request for /bilder%20und%20filme/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"33sV3cIbtnbT6JdP42i8","level":3,"time":"2024-05-31T11:00:46+00:00","remoteAddr":"","user":"--","app":"memories","method":"","url":"--","message":"Failed to index folder /ncp/files/qnap_daten: Invalid request for /daten/.nomedia (InvalidArgumentException)","userAgent":"--","version":"28.0.5.1","data":{"app":"memories"}}
{"reqId":"ZlwWAiqRQ8X1q0zmJMoKcAAAAAU","level":3,"time":"2024-06-02T06:49:39+00:00","remoteAddr":"2a02:8071:6400:d5e0:baa3:4de3:ed1a:740b","user":"enrico","app":"webdav","method":"PUT","url":"/remote.php/dav/files/enrico/qnap_bilder/_unsortiert/Handy%20Enrico/2024/06/.pending-1717915773-_PXL_2092169964941368_1717310973140_2092169965109076.jpg","message":"X-OC-MTime header must be a valid positive integer","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.29.0","version":"28.0.5.1","exception":{"Exception":"InvalidArgumentException","Message":"X-OC-MTime header must be a valid positive integer","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/dav/lib/Connec [... cut ...] ction":"initSession","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1200,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/index.php","line":37,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Security/Crypto.php","Line":158,"message":"Could not decrypt or decode encrypted session data","exception":{},"CustomMessage":"Could not decrypt or decode encrypted session data"}}

```
</details>

christoph-bittmann, 2024-06-08:
Hi @EnricoK18,

where comes the http://ppa.launchpad.net/ondrej dependency from? I would suggest to disable these repos and retry the update. In a second step it could be an idea to uninstall all related packaged and/or replace them which versions from official debian repo.

EnricoK18, 2024-06-08:
Hi @christoph-bittmann,

I think this dependancy cam from previous test with different PHP version.

I was able to delete them. And now the update finished successfully.

Thanks

closed 2024-06-08 by EnricoK18