nextcloud/nextcloudpi #1942: Pre-set users/passwords on NextCloudPi

Madnevil, 2024-07-02:
Hi there!

I'm having an issue where I have a fresh install of NCPi 1.54 and I cannot access the activation page via:
https://nextcloudpi.local/
https://nextcloudpi/
local IP

So I decided to plugin monitor and keyboard to see what's going on but I cannot access console with credentials provided here (u:root p:vagrant):
[What are pre-set users/passwords on NextCloudPi?
](https://help.nextcloud.com/t/frequently-asked-questions-faq-ncp/126325#what-are-pre-set-userspasswords-on-nextcloudpi-10)

Are those outdated or am I missing something?

Thanks,
M

pglevy, 2024-07-21:
I think I'm also stuck here with fresh install on new Pi 4. My [hapless attempt described here](https://help.nextcloud.com/t/noob-stuck-on-boot-screen-asking-for-login-with-pi4/198601) on Nextcloud site.

theCalcaholic, 2024-07-22:
There are no default account passwords on NCP. root/vagrant was only for the VM image, which is currently unsupported

theCalcaholic, 2024-07-22:
@pglevy I would recommend to install NCP from the official RPi4-Image (see https://github.com/nextcloud/nextcloudpi/releases).

Installation onto RPi OS is not possible at this time

pglevy, 2024-07-25:
Thank you! My mental model of what I was trying to do was a bit off. (I was focused on trying to log in to the Pi, which I was looking at via the connected monitor.) 

But I reflashed the SD card with image and got it up and running! (Not sure why trying to activate through .local didn't work but going to the IP address directly did. Instructions a bit vague here. And I still don't know what it means that the Pi itself is sitting on a login prompt. But I see files now!)

I did run into a crash on Android when trying to upload some photos, which I reported here: https://github.com/nextcloud/android/issues/13315

Looking forward to digging in further now to set up more devices and backup.

closed 2024-08-04 by theCalcaholic