nextcloud/nextcloudpi #1943: Message "Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade" is shown after dist upgrade completed

Andi-82, 2024-07-02:
I upgraded my nextcloudpi host system from Bullseye to Bookworm. This was running thru without issues.
My expectation would be that also the message from the SSH Bash Login screen 
NextCloudPi v1.54.2 is up to date
Update to bookworm available. Type 'sudo ncp-dist-upgrade' to upgrade
should be not shown anymore. On my installation this is still shown.

If i run 'sudo ncp-dist-upgrade' again, the bash shows me 'No dist-upgrade available for OS version: Debian 12 (bookworm).'

<details>
<summary>NextcloudPi diagnostics</summary>

```

grep: /etc/apache2/sites-available/nextcloud.conf: No such file or directory
NextcloudPi version  v1.54.2
NextcloudPi image    NextCloudPi_RaspberryPi_v1.51.1.img
OS                   Debian GNU/Linux 12. 6.1.21-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /opt/ncdata/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      26G/59G
rootfs usage         26G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    29.0.3.4
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
Public IPv4          ***REMOVED SENSITIVE VALUE***
Public IPv6          ***REMOVED SENSITIVE VALUE***
Port 80              open
Port 443             open
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
Certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               2days

```
</details>

Jenfong, 2024-08-04:
the same after upgrade to bookworm ,cannot start server.
+ ncp-config
`$ sudo ncp-config 
Cannot load Zend OPcache - it was already loaded
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20210902/redis.so (/usr/lib/php/20210902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20210902/redis.so.so (/usr/lib/php/20210902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Cannot load Zend OPcache - it was already loaded
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20210902/redis.so (/usr/lib/php/20210902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20210902/redis.so.so (/usr/lib/php/20210902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
Running nc-fix-permissions
Cannot load Zend OPcache - it was already loaded
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20210902/redis.so (/usr/lib/php/20210902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20210902/redis.so.so (/usr/lib/php/20210902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
fixing permissions in /data/ncdata/data... done
Done. Press any key...
`
+ php-fpm
`$ php-fpm8.1 -v
Cannot load Zend OPcache - it was already loaded
[04-Aug-2024 08:46:44] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20210902/redis.so (/usr/lib/php/20210902/redis.so: undefined symbol: igbinary_serialize), /usr/lib/php/20210902/redis.so.so (/usr/lib/php/20210902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
PHP 8.1.29 (fpm-fcgi) (built: Jun  6 2024 16:31:57)
Copyright (c) The PHP Group
Zend Engine v4.1.29, Copyright (c) Zend Technologies
    with Zend OPcache v8.1.29, Copyright (c), by Zend Technologies
`

theCalcaholic, 2024-08-04:
That probably means, that the dist upgrade did, in fact, *not* complete successfully.
If everything works, it might have finished everything important, if not (as in @Jenfong's case, it is missing some relevant steps).

If everything works, you're probably good and can ignore the message (or delete it, I'll post instructions how to).

If your instance is broken you will need to reinstall and restore from backup or try to fix your system manually (some guidelines can be found [here]()).

closed 2024-08-04 by theCalcaholic

Jenfong, 2024-08-23:
> That probably means, that the dist upgrade did, in fact, _not_ complete successfully. If everything works, it might have finished everything important, if not (as in @Jenfong's case, it is missing some relevant steps).
> 
> @Andi-82 If everything works, you're probably good and can ignore the message - or delete it with the following commands:
> 
> ```shell
> sudo sed -i -e 's/bullseye/bookworm/' /usr/local/etc/ncp.cfg
> sudo rm -f /etc/update-motd.d/30-ncp-dist-upgrade 
> sudo rm -f /usr/local/etc/ncp-recommended.cfg
> ```
> 
> @Jenfong If your instance is broken you will need to reinstall and restore from backup or try to fix your system manually (some guidelines can be found [here](https://github.com/nextcloud/nextcloudpi/issues/1913)).

it's mean os upgrade sucess,but missing some relevant steps. 
actual nextcloud do not updagrade sucess. I upgrade nextcloud manually.  netxtcloud server  recoveryed!
```shell
root@nextcloudpi:/var/www/nextcloud/updater# pwd
/var/www/nextcloud/updater
root@nextcloudpi:/var/www/nextcloud/updater# php updater/updater.phar

```