nextcloud/nextcloudpi #1944: Logged out automatically - Session token problem

Ste-denko, 2024-07-04:
Hi All,

I am running NextCloudPi v1.54.2 on RaspberryPi5.

First of all a general question about the version:
Some time v1.54.2 was available, meanwhile latest version is again v1.54.0.
Which one is now latest and should be used?
Was v1.54.2 faulty?

Main Problem:

When using NCP I often get logged out automatically.
That can happen already 2 min after I just logged in.
Error in logs shows session token problem.

Error message from logs:

`{"reqId":"ZoZbcPQafwiSDmHqJ2C2WQAADRI","level":3,"time":"2024-07-04T08:21:04+00:00","remoteAddr":"x.x.x.x","user":"--","app":"core","method":"GET","url":"/ocs/v2.php/search/providers/comments/search?term=css&from=%2Fsettings%2Fapps","message":"Renewing session token failed: Token does not exist: df3ad82ad5c899448f18b6ebe05fb3042e00455dea37fc2cd10f898707148305b7ea3f5d92ebaf039ba0f75aabeb959ab6cdf556fe664dd7eed09f5cb72833a5","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:127.0) Gecko/20100101 Firefox/127.0","version":"28.0.5.1","exception":{"Exception":"OC\\Authentication\\Exceptions\\InvalidTokenException","Message":"Token does not exist: df3ad82ad5c899448f18b6ebe05fb3042e00455dea37fc2cd10f898707148305b7ea3f5d92ebaf039ba0f75aabeb959ab6cdf556fe664dd7eed09f5cb72833a5","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Authentication/Token/PublicKeyTokenProvider.php","line":167,"function":"getTokenFromCache","class":"OC\\Authentication\\Token\\PublicKeyTokenProvider","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Authentication/Token/PublicKeyTokenProvider.php","line":248,"function":"getToken","class":"OC\\Authentication\\Token\\PublicKeyTokenProvider","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/public/AppFramework/Db/TTransactional.php","line":63,"function":"OC\\Authentication\\Token\\{closure}","class":"OC\\Authentication\\Token\\PublicKeyTokenProvider","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Authentication/Token/PublicKeyTokenProvider.php","line":274,"function":"atomic","class":"OC\\Authentication\\Token\\PublicKeyTokenProvider","type":"->"},{"file":"/var/www/nextcloud/lib/private/Authentication/Token/Manager.php","line":172,"function":"renewSessionToken","class":"OC\\Authentication\\Token\\PublicKeyTokenProvider","type":"->"},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":929,"function":"renewSessionToken","class":"OC\\Authentication\\Token\\Manager","type":"->"},{"file":"/var/www/nextcloud/lib/base.php","line":1151,"function":"loginWithCookie","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/ocs/v1.php","line":62,"function":"handleLogin","class":"OC","type":"::"},{"file":"/var/www/nextcloud/ocs/v2.php","line":23,"args":["/var/www/nextcloud/ocs/v1.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Authentication/Token/PublicKeyTokenProvider.php","Line":197,"message":"Renewing session token failed: Token does not exist: df3ad82ad5c899448f18b6ebe05fb3042e00455dea37fc2cd10f898707148305b7ea3f5d92ebaf039ba0f75aabeb959ab6cdf556fe664dd7eed09f5cb72833a5","user":"ncp","exception":[],"CustomMessage":"Renewing session token failed: Token does not exist: df3ad82ad5c899448f18b6ebe05fb3042e00455dea37fc2cd10f898707148305b7ea3f5d92ebaf039ba0f75aabeb959ab6cdf556fe664dd7eed09f5cb72833a5"},"id":"66865c8c4be41"}`

Anyone else having this problem?
I would really appreciate any help or fix for this.
Thank you very much in advance.

Love NextcloudPi!
Cheers

MegaMarkey, 2024-07-04:
I am experiencing the same problem (NextCloudPi v1.54.2 and Nextcloud 28.0.5).

NickKopernikus, 2024-07-10:
As said before, this is not an issue of nextcloudpi. I had this issu, when I updated to 20.0.5 with nextcloudpi about a month ago.

After manual update to 28.0.6, which is working fine for about a mont right now with latest nextcloudpi, the problem is gone.

So my advice: full backup then manually update to 28.0.6 (on your own risk!).

Ste-denko, 2024-07-10:
> After manual update to 28.0.6, which is working fine for about a mont right now with latest nextcloudpi, the problem is gone.

Ok, so I hope that this bug will be gone with the update from 28.0.5 to 28.0.6.
Thank you very much for this information.
Did you have this problem for all NC users/accounts or only for the ncp/admin account?

Desislav-Arashev, 2024-07-11:
> Ok, so I hope that this bug will be gone with the update from 28.0.5 to 28.0.6. Thank you very much for this information. Did you have this problem for all NC users/accounts or only for the [...]

I have it will all accounts.

theCalcaholic, 2024-08-11:
Seems to be an upstream issue with Nextcloud <= 28.0.5

closed 2024-08-11 by theCalcaholic